MichaIng/DietPi #2600: Unable to complete installation of anything that requires PHP

akshay7394, 2019-02-28:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?
- `dietpi-bugreport` ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce

1. Manually upgrade to PHP 7.2 
2. Try to install LAMP stack or software that requires stack (like PiHole)

#### Expected behaviour

- Desired software (MariaDB stack with apache2) should get installed 

#### Actual behaviour

- The installation fails incessantly. Installation fails even if the installation is attempted after purging all instances of apache and php

#### Extra details

- Even after attempting to purge any and all packages relating to web server and php, this error still blocks installation. The actual failure happens when the system is trying to install PHP.

The error that is thrown is : 
```
[  OK  ] DietPi-Software | APT installation for: libapache2-mod-php7.0 php7.0-curl php7.0-gd php-apcu php7.0-mbstring ph
p7.0-opcache php7.0-zip php7.0-xml php7.0-mcrypt php7.0-mysql, please wait...
E: Unable to correct problems, you have held broken packages.

[FAILED] DietPi-Software | G_AGI: libapache2-mod-php7.0 php7.0-curl php7.0-gd php-apcu php7.0-mbstring php7.0-opcache ph
p7.0-zip php7.0-xml php7.0-mcrypt php7.0-mysql
```

MichaIng, 2019-02-28:
@akshay7394 
Many thanks for your report.

This is known and expected. DietPi-Software expect and installs PHP from the official APT repo, which is currently PHP7.0. If you manually install a more current PHP version from custom sources, you run into these issues.

But let me try to get this fixed for you, as we anyway plan to install PHP7.2 (or even 7.3) on Stretch as well, as some software titles are going to drop support for lower versions: https://github.com/MichaIng/DietPi/issues/2367

- I hope you installed PHP7.2 from Ondrejs repo: https://packages.sury.org/php/ ?
- I these two lines (197+198) repace `7.0` with `7.2`:
https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-software#L197-L198
- Remove or comment this line (3519):
https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-software#L3519

With Apache, nothing else should be required. Please report back if this works fine.

akshay7394, 2019-03-01:
@MichaIng Hi! No I didn't follow those instructions to set it up-- however, I will now.

Quick question though, if I Purged both apache2 and php7.2-related packages, why is the install still failing through dietpi-software? Aren't there no longer any packages for it to have problems with?

MichaIng, 2019-04-24:
@akshay7394 
Sorry for the forgotten reply.

Not sure why APT still fails. Most likely there is still some package installed that breaks earlier PHP and/or other software versions from official repo, perhaps some library. E.g. I found that when installing certain packages from Ondrejs PHP repo, the `libssl1.1` version he ships breaks Lighttpd on Stretch (and Jessie) systems. It is quite a hassle to downgrade those packages since they are dependencies for a whole bunch or other core packages. This happened to me when playing around with PHP7.3 on Stretch: https://github.com/MichaIng/DietPi/issues/2734

But in ~1 week we release DietPi v6.23 and migrate all systems to PHP7.3 (besides when ownCloud is installed, then PHP7.2): https://github.com/MichaIng/DietPi/issues/2367#issuecomment-485917360

From then on, Ondrejs repo will be the default and we will keep PHP versions up-to-date as fast as offered software titles support it.

closed 2024-11-20 by akshay7394