MichaIng/DietPi #2603: kernel hung at smb/cifs read ???

seniorgod, 2019-03-01:
#### Required Information
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=21
G_DIETPI_VERSION_RC=1
G_GITBRANCH=master
G_GITOWNER=Fourdee

stretch 

Linux DietPi 3.10.38 #22 SMP PREEMPT Fri Dec 22 10:45:41 IST 2017 armv7l GNU/Linux

Allo Sparky with usbridge 

- Power supply used | 2 x 5 V on Sparky and on usbridge 
- SDcard used | emmc 

Dear all,

i updated to 6.21.1 
i mounted my smb share including music under /mnt/dietpi_userdata/Music 
i'm able to walk through the files and directories on shell layer without any problem

I started mpd and run mpc update in order to rescan the music directory of mpd (/mnt/dietpi_userdata/Music ) to find my music files in mpd. 
But mpd stops after some files. 
In journalctl i found: 

 INFO: task kworker/0:0:5559 blocked for more than 120 seconds.
Mär 01 20:37:21 DietPi kernel: "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Mär 01 20:37:21 DietPi kernel: kworker/0:0     D c07bfe48     0  5559      2 0x00000000
Mär 01 20:37:21 DietPi kernel: Workqueue: cifsiod cifs_readv_complete
Mär 01 20:37:21 DietPi kernel: [<c07bfe48>] (__schedule+0x390/0x7cc) from [<c07c0e34>] (__down_write_nested+0x70/0xc4)
Mär 01 20:37:21 DietPi kernel: [<c07c0e34>] (__down_write_nested+0x70/0xc4) from [<c023769c>] (cifsFileInfo_put+0xf0/0x224)
Mär 01 20:37:21 DietPi kernel: [<c023769c>] (cifsFileInfo_put+0xf0/0x224) from [<c02377e8>] (cifs_readdata_release+0x18/0x24)
Mär 01 20:37:21 DietPi kernel: [<c02377e8>] (cifs_readdata_release+0x18/0x24) from [<c00470dc>] (process_one_work+0x134/0x414)
Mär 01 20:37:22 DietPi kernel: [<c00470dc>] (process_one_work+0x134/0x414) from [<c00477b8>] (worker_thread+0x138/0x37c)
Mär 01 20:37:22 DietPi kernel: [<c00477b8>] (worker_thread+0x138/0x37c) from [<c004d394>] (kthread+0xa4/0xb0)
Mär 01 20:37:22 DietPi kernel: [<c004d394>] (kthread+0xa4/0xb0) from [<c000f3d8>] (ret_from_fork+0x14/0x3c)
Mär 01 20:37:22 DietPi kernel: INFO: task update:5581 blocked for more than 120 seconds.
Mär 01 20:37:22 DietPi kernel: "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Mär 01 20:37:22 DietPi kernel: update          D c07bfe48     0  5581      1 0x00000800
Mär 01 20:37:22 DietPi kernel: [<c07bfe48>] (__schedule+0x390/0x7cc) from [<c07c0d8c>] (__down_read+0xac/0xe4)
Mär 01 20:37:22 DietPi kernel: [<c07c0d8c>] (__down_read+0xac/0xe4) from [<c023612c>] (cifs_reopen_file+0x1a4/0x37c)
Mär 01 20:37:22 DietPi kernel: [<c023612c>] (cifs_reopen_file+0x1a4/0x37c) from [<c0236334>] (cifs_retry_async_readv+0x30/0x58)
Mär 01 20:37:22 DietPi kernel: [<c0236334>] (cifs_retry_async_readv+0x30/0x58) from [<c0237b50>] (cifs_readpages+0x2ac/0x414)
Mär 01 20:37:22 DietPi kernel: [<c0237b50>] (cifs_readpages+0x2ac/0x414) from [<c00d5868>] (__do_page_cache_readahead+0x1a8/0x26c)
Mär 01 20:37:22 DietPi kernel: [<c00d5868>] (__do_page_cache_readahead+0x1a8/0x26c) from [<c00d5eb0>] (page_cache_sync_readahead+0x44/0x64)
Mär 01 20:37:22 DietPi kernel: [<c00d5eb0>] (page_cache_sync_readahead+0x44/0x64) from [<c00cc990>] (generic_file_aio_read+0x4d4/0x700)
Mär 01 20:37:22 DietPi kernel: [<c00cc990>] (generic_file_aio_read+0x4d4/0x700) from [<c023a9e0>] (cifs_strict_readv+0x160/0x184)
Mär 01 20:37:22 DietPi kernel: [<c023a9e0>] (cifs_strict_readv+0x160/0x184) from [<c0111c58>] (do_sync_read+0x78/0xa0)
Mär 01 20:37:22 DietPi kernel: [<c0111c58>] (do_sync_read+0x78/0xa0) from [<c01126dc>] (vfs_read+0xb8/0x1e0)
Mär 01 20:37:22 DietPi kernel: [<c01126dc>] (vfs_read+0xb8/0x1e0) from [<c0112938>] (SyS_read+0x3c/0x70)
Mär 01 20:37:22 DietPi kernel: [<c0112938>] (SyS_read+0x3c/0x70) from [<c000f4a8>] (__sys_trace_return+0x0/0x18)

what can i do?

MichaIng, 2019-03-02:
@seniorgod 
Thanks for reporting.

Did you recheck when applying the permissions fix from the other issue? https://github.com/MichaIng/DietPi/issues/2605#issuecomment-468944577

seniorgod, 2019-03-02:
after setting the user / group to mpd/audio it works or at last i didn't face the problem again.

closed 2019-03-02 by seniorgod