MichaIng/DietPi #2604: Dietpi-Drive_Manager makes /root read only, but it does not survive reboot

JBatSEEDLING, 2019-03-01:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
6
21
1
Master
Fourdee
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?
- `dietpi-bugreport` ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce

Fresh install, use the dietpi-drive_Manager
   select / (/dev/mmcblk0p2)
   remove swapfile
   set to readonly
  reboot

#### Expected behaviour

    / should be readonly

#### Actual behaviour

     / is readonly before the reboot
    / is no longer readonly after the reboot
    dietpi-drive_manager shows / as still containing the swap file

#### Extra details

    I did try to set the location of the swap file to dev/null in /Dietpi/dietpi.config as described in a previous bug, but that fails to survive reboot as well.

Fourdee, 2019-03-02:
@JBatSEEDLING 

Thanks for the report 👍 

Appears to be a bug our end, will take a look.

In regards to disabling the swapfile, simply set the value to ```0``` in ```dietpi-drive_manager```, avoid using ```/dev/null```

Fourdee, 2019-03-02:
🈯️ Resolved for v6.22. Applied to ```/etc/fstab``` before remounting drive as RO:
```
PARTUUID=27504eef-02 / auto defaults,noatime,ro 0 1
```

closed 2019-03-02 by Fourdee