MichaIng/DietPi #2607: hciuart.services failed, bluetoothctl no default controller

RynFlutsch, 2019-03-02:
[image]

MichaIng, 2019-03-02:
@RynFlutsch 
Thanks for your report. Did you check the suggested commands for further logs?
Please print the output of the following here:
`systemctl status hciuart`
`journalctl -t hciuart` or `journalctl -u hciuart`
`dmesg | tail`

RynFlutsch, 2019-03-03:
Hey MichaIng:
[image]
[image]
[image]
[image]
[image]

MichaIng, 2019-03-03:
@RynFlutsch 
Could you try:
`G_CONFIG_INJECT 'enable_uart=' 'enable_uart=1' /DietPi/config.txt`

reboot and:

`systemctl start hciuart`

RynFlutsch, 2019-03-04:
Yes it shows:
Enable_UART=1
got the same problems with a clean dietpi Version.

RynFlutsch, 2019-03-04:
Also cant setup wiringPi. But i already opened another issue.

MichaIng, 2019-03-04:
@RynFlutsch 
Is the content indeed with this capitalisation? AFAIK it needs to be all lower case, please try this as well.

Aside from that, is the bluetooth daemon in correct stage?
`journalctl -u bluetooth`

RynFlutsch, 2019-03-04:
No its all lower case.
I will check later

RynFlutsch, 2019-03-05:
root@Gara:~# journalctl -u bluetooth
-- Logs begin at Thu 2016-11-03 18:16:42 CET, end at Tue 2019-03-05 20:30:36 CET. --
Mär 05 20:30:00 Gara systemd[1]: Starting Bluetooth service...
Mär 05 20:30:00 Gara bluetoothd[1764]: Bluetooth daemon 5.43
Mär 05 20:30:00 Gara systemd[1]: Started Bluetooth service.
Mär 05 20:30:00 Gara bluetoothd[1764]: Starting SDP server
Mär 05 20:30:00 Gara bluetoothd[1764]: Bluetooth management interface 1.14 initialized

RynFlutsch, 2019-03-05:
root@Gara:~# bluetoothctl
[bluetooth]# scan on
No default controller available
[bluetooth]#

MichaIng, 2019-03-05:
@RynFlutsch 
Puhh, found some topic about which serial console bluetooth uses on which RPi version: https://wiki.fhem.de/wiki/Raspberry_Pi_3:_GPIO-Port_Module_und_Bluetooth
It on German language 😉.

Lets find out which console is used by what in your case. Perhaps there is some incompatibility by certain configs or settings applied during software installs. Also lets verify the required modules are enabled:
```
lsmod
cat /lib/systemd/system/hciuart.service
cat /boot/cmdline.txt
systemctl status serial-getty@ttyAMA0.service
systemctl status serial-getty@ttyS0.service
```

RynFlutsch, 2019-03-06:
Tablet: crap keyboard :D
[image]

MichaIng, 2019-03-06:
@RynFlutsch 
Okay lets first try to enable the kernel modules, which I couldn't find. Not sure which are present on RPi3, besides `btbcm`:
```
modprobe bluetooth
modprobe bnep
modprobe btbcm
modprobe hci_uart
```

Then lets disable all serial consoles, I hope your don't require them?
```
systemctl --now disable serial-getty@ttyAMA0
systemctl --now disable serial-getty@ttyS0
```

Then we also remove the serial console from cmdline.txt:
```
sed -i 's/console=serial0,115200 //' /boot/cmdline.txt
```
_____________
I actually think the issue is that we enable the serial console on `ttyAMA0` (aka UART0 aka „full UART“), which is default on  RPi1+2, but as well on `ttyS0` (aka UART1 aka „mini UART“), which is default on RPi3. In your case both TTY services are enabled, so blocked. On RPi3, `ttyAMA0` is by default used/intended for the bluetooth module. So the serial console should be active, if required, on `ttyS0` only to allow concurrent bluetooth.

Now I think `console=serial0,115200` in cmdline.txt enables/blocks the serial console as well, I am just not sure if `serial0` is `ttyAMA0` or `ttyS0`. https://raspberrypi.stackexchange.com/a/45571 states that it uses the first appropriate device, so might respect active bluetooth, but we can try this later, if required.

Bluetooth can be forced to use ttyS0 via `dtoverlay=pi3-miniuart-bt` in config.txt, however this seems to be only required if you use some special GPIO modules that require „full UART“ `ttyAMA0`, so bluetooth has to be happy with „mini UART“ `ttyS0`.

RynFlutsch, 2019-03-07:
Hey,
Serial Console is used for uart or/ and i2C Sensors?
I have executed your code and now i can start bluetoothctl and scan for devices.
I will check on another system.
Unfortunately my BT sensor is not working, but i think its another problem.
I have to fix something after `sed -i 's/console=serial0,115200 //' /boot/cmdline.txt`
or thats it?
I need bluetooth, i2c  and maybe uart connection.

Fourdee, 2019-03-07:
Had a similar issue, BT works on initial activation, however, after reboot BT fails in dmesg.

Enable:
```
root@DietPi:~# systemctl status hciuart
● hciuart.service - Configure Bluetooth Modems connected by UART
   Loaded: loaded (/lib/systemd/system/hciuart.service; enabled; vendor preset:
enabled)
   Active: active (running) since Thu 2019-03-07 13:21:01 GMT; 20s ag
o
  Process: 9191 ExecStart=/usr/bin/btuart (code=exited, status=0/SUCCESS)
 Main PID: 9214 (hciattach)
   CGroup: /system.slice/hciuart.service
           └─9214 /usr/bin/hciattach /dev/serial1 bcm43xx 921600 noflow - b8:27:
eb:24:f9:f1

Mar 07 13:21:00 DietPi systemd[1]: Starting Configure Bluetooth Modems connected
 by UART...
Mar 07 13:21:01 DietPi btuart[9191]: [40B blob data]
Mar 07 13:21:01 DietPi btuart[9191]: bcm43xx_init
Mar 07 13:21:01 DietPi btuart[9191]: Set BDADDR UART: b8:27:eb:24:f9:f1
Mar 07 13:21:01 DietPi btuart[9191]: Set Controller UART speed to 921600 bit/s
Mar 07 13:21:01 DietPi btuart[9191]: Device setup complete
Mar 07 13:21:01 DietPi systemd[1]: Started Configure Bluetooth Modems connected
by UART.
```

reboot:
```
dmesg
[   12.911622] Bluetooth: hci0 sending frame failed (-49)

root@DietPi:~# bluetoothctl
[bluetooth]# power on
No default controller available

```

Fourdee, 2019-03-07:
Test:
- Serial enabled
- Enable BT (disables serial) | verify BT working 🈯️ 
- Reboot and verify BT working 🈯️

Fourdee, 2019-03-07:
@RynFlutsch 

Fixed for v6.22, solution to apply now:

#### Step 1
```
/DietPi/dietpi/func/dietpi-set_hardware bluetooth disable
/DietPi/dietpi/func/dietpi-set_hardware serialconsole disable
reboot
```

#### Step 2
```
/DietPi/dietpi/func/dietpi-set_hardware bluetooth enable
```

MichaIng, 2019-03-07:
@Fourdee 
Many thanks for this. However I think we need to redo serial console handling anyway. The thing is that indeed it can be enabled concurrently with bluetooth:
- `serial-getty@ttyS0.service` and `console=ttyS0,115200` or `console=serial0,115200` should work fine on RPi3. We simply need to differentiate between the two serial devices (this and ttyAMA0).

The `cmdline.txt` entry I believe only lets the boot messages be printed to this serial console as well and most properly allows emergency login there in case of failure.
The serial-getty@ service might allow the same for for regular login.
So this is similar to `console=tty1` prints boot messages to tty1 but has nothing to do with getty@tty1.service which is responsible to allow final login on that TTY.

Anyhow, since on RPi3, bluetooth uses ttyAMA0, ttyS0 can be freely used for boot messages and/or as login console and/or for other I2C/SPI/I2S/UART devices.

`enable_uart=1` btw only enables ttyS0, while ttyAMA0 is enabled by default: https://www.raspberrypi.org/documentation/configuration/uart.md

So yeah, actually to have it all consistent on RPi3:
- Toggle serial console only toggles `enable_uart` setting, `serial-getty@ttyS0.service` and `console=ttyS0,115200`.
- So leave ttyAMA0 for bluetooth and other serial devices attached.
- Make clear that hardware full UART is always enabled and this setting indeed only toggles a boot/login console on software mini UART.
- Also this doc makes it clear, that the mini UART is intended to be used for console while the full UART is intended to be used for hardware devices: https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2835/BCM2835-ARM-Peripherals.pdf

Perhaps even a second option to toggle the ttyAMA0 would be great as well. However I think this is toggled automatically by the dtoverlays/params, kernel modules and such, when enabling bluetooth, I2C/...

MichaIng, 2019-03-07:
@RynFlutsch 
> I need bluetooth, i2c and maybe uart connection.

So yeah all of them use/block a serial "console/device" from which are as mention two available.
As bluetooth uses ttyAMA0, you can have other devices use ttyS0. `enable_uart=1` is what toggles this, and of course the related overlay, e.g. `dtoverlay=i2c_arm` for I²C.

`serial-getty@ttyS0.service` and `console=ttyS0,115200` should be only required, if you use an actual serial console to view log messages and login, e.g. for emergency access and/or debugging.

RynFlutsch, 2019-03-07:
> @RynFlutsch
> 
> Fixed for v6.22, solution to apply now:
> 
> #### Step 1
> ```
> /DietPi/dietpi/func/dietpi-set_hardware bluetooth disable
> /DietPi/dietpi/func/dietpi-set_hardware serialconsole disable
> reboot
> ```
> #### Step 2
> ```
> /DietPi/dietpi/func/dietpi-set_hardware bluetooth enable
> ```

Not working same error:
Job for hciuart.service failed because the control process exited with error code.

I will check michaing's solution or wait for 6.22. 
Thank you for Dietpi!

MichaIng, 2019-03-31:
PR opened which allows to toggle serial login console (+ boot message output) on individual existent serial devices: https://github.com/MichaIng/DietPi/pull/2678

When on RPi Bluetooth is enabled while serial console on ttyAMA0 is active, a warning prompts and allows to disable the serial console to continue or cancel.
The other way round when Bluetooth is active and a serial console on ttyAMA0 is about to be enabled, a warning prompts and allows to disable Bluetooth to continue or cancel.

As well on RPi the primary serial/UART device can be enabled/disabled completely but this has no effect on Bluetooth since:
- On all RPi with onboard WiFi+BT the primary UART is mini (software) UART on ttyS0 while ttyAMA0 cannot be disabled (AFAIK, at least not via enable_uart=0 only).
- On all RPi without onboard WiFi+BT the primary UART is indeed full (hardware) UART on ttyAMA0 while ttyS0 cannot be enabled without further custom hacks.

@RynFlutsch 
Did you further test?

If you need Bluetooth + I2C you need both UART devices, so `enable_uart=1` needs to be set in `/DietPi/config.txt`.
No serial console can be used, so you need to disable both:
```
systemctl disable serial-console@ttyS0
systemctl mask serial-console@ttyS0
systemctl disable serial-console@ttyAMA0
systemctl mask serial-console@ttyAMA0
```
And in `/boot/cmdline.txt` remove all `console=` entries besides `console=tty1` (and even this can be removed since tty1 is used as default if non is defined for boot messsages).

I hope this helps.

closed 2019-05-12 by MichaIng