MichaIng/DietPi #2608: DietPi Update G_AGUG on first login fails with hash mismatch

copain60, 2019-03-02:
#### Details:
- Date           | Sat  2 Mar 14:39:43 GMT 2019
- Bug report     | N/A
- DietPi version | v6.20.0 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Kernel version | #1159 SMP Sun Nov 4 17:50:20 GMT 2018
- Distro         | stretch (index=4)
- Command        | G_AGUG
- Exit code      | 100
- Software title | DietPi-Update

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
   Last modification reported: Sun, 10 Feb 2019 04:01:32 +0000
E: Failed to fetch http://mirror.iway.ch/raspbian/raspbian/pool/main/g/glibc/locales_2.24-11+deb9u4_all.deb  Hash Sum mismatch
   Hashes of expected file:
    - SHA256:7a57eeecdf1150b5daa7de0b8a2a27500a4ffb2716fca8544610015962822543
    - SHA1:fc7b36c1331fd7cfe7c254d64a5ed9fd9fbe7c91 [weak]
    - MD5Sum:4d9413dc5d9cee005cd06db4033f59f7 [weak]
    - Filesize:3288578 [weak]
   Hashes of received file:
    - SHA256:e3b0c44298fc1c149afbf4c8996fb92427ae41e4649b934ca495991b7852b855
    - SHA1:da39a3ee5e6b4b0d3255bfef95601890afd80709 [weak]
    - MD5Sum:d41d8cd98f00b204e9800998ecf8427e [weak]
    - Filesize:0 [weak]
   Last modification reported: Sun, 10 Feb 2019 04:01:32 +0000
E: Failed to fetch http://mirror.iway.ch/raspbian/raspbian/pool/main/c/curl/curl_7.52.1-5+deb9u9_armhf.deb  Hash Sum mismatch
   Hashes of expected file:
    - SHA256:ad5b29450f5af79a97e895996afec94be0bc66789535bba2e6a2092c90f318f2
    - SHA1:cc2a8333528a31a1206ea3143e71b228dc2e329e [weak]
    - MD5Sum:61defa5ff772f8b5762544ab4f7004d2 [weak]
    - Filesize:219988 [weak]
   Hashes of received file:
    - SHA256:e3b0c44298fc1c149afbf4c8996fb92427ae41e4649b934ca495991b7852b855
    - SHA1:da39a3ee5e6b4b0d3255bfef95601890afd80709 [weak]
    - MD5Sum:d41d8cd98f00b204e9800998ecf8427e [weak]
    - Filesize:0 [weak]
   Last modification reported: Thu, 07 Feb 2019 10:08:09 +0000
E: Failed to fetch http://mirror.iway.ch/raspbian/raspbian/pool/main/c/curl/libcurl3_7.52.1-5+deb9u9_armhf.deb  Hash Sum mismatch
   Hashes of expected file:
    - SHA256:2b0cdefba6b35b4cc7610f90e5115ad5ce10d5507af0af1c9d9ba1988bd58912
    - SHA1:2748e72f94e7185023aae0ee52be6a6cb9b61169 [weak]
    - MD5Sum:4f3c1f9494ce5e2e02368ffe19116d28 [weak]
    - Filesize:261506 [weak]
   Hashes of received file:
    - SHA256:e3b0c44298fc1c149afbf4c8996fb92427ae41e4649b934ca495991b7852b855
    - SHA1:da39a3ee5e6b4b0d3255bfef95601890afd80709 [weak]
    - MD5Sum:d41d8cd98f00b204e9800998ecf8427e [weak]
    - Filesize:0 [weak]
   Last modification reported: Thu, 07 Feb 2019 10:08:09 +0000
E: Failed to fetch http://mirror.iway.ch/raspbian/raspbian/pool/main/c/curl/libcurl3-gnutls_7.52.1-5+deb9u9_armhf.deb  Hash Sum mismatch
   Hashes of expected file:
    - SHA256:1de9457bf7f60d914598dab2f3785e08ff92577c42bf73f13fd32256edb6cffd
    - SHA1:8ecc0055c00331e923ea1e724fb379731b50e9e5 [weak]
    - MD5Sum:2302d16561b53f383958cf1277dced9f [weak]
    - Filesize:259736 [weak]
   Hashes of received file:
    - SHA256:e3b0c44298fc1c149afbf4c8996fb92427ae41e4649b934ca495991b7852b855
    - SHA1:da39a3ee5e6b4b0d3255bfef95601890afd80709 [weak]
    - MD5Sum:d41d8cd98f00b204e9800998ecf8427e [weak]
    - Filesize:0 [weak]
   Last modification reported: Thu, 07 Feb 2019 10:08:09 +0000
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

Is there any solution to this issue? Many thanks for your support.

tripflip, 2019-03-02:
same problem here, rpi3b, freshly burned img, and I'm getting the same errors

tripflip, 2019-03-02:
FYI, changing the apt mirror in dietpi-config fixed this for me

copain60, 2019-03-02:
Thanks tripflip, yes I changed the apt mirror to Germany and the update was successfully applied!
Nevertheless, I still leave the issue open. Hash mismatches probably need to be fixed somewhere.

MichaIng, 2019-03-02:
@copain60 @tripflip 
Thanks for reporting.

@copain60 
Which mirror did you use? The default one `raspbian.raspberrypi.org`?

Perhaps a temporary mirror sync mismatch that was fixed in some mirrors earlier then in others, depending on their sync schedule.

Would be good to know if this has been fixed within the repos meanwhile.

copain60, 2019-03-02:
@MichaIng 
Thanks for answering.

On first login of the freshly burned image, I used the default apt mirror.
Then I changed it to Germany and thereafter the update ran flawlessly.

MichaIng, 2019-03-02:
I also tested with the default mirror now, works flawlessly here, although Buster branch:
```
2019-03-02 19:44:53 root@micha:/tmp# apt update
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Hit:2 https://archive.raspberrypi.org/debian buster InRelease
Get:3 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.1 MB]
Fetched 13.1 MB in 18s (723 kB/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
4 packages can be upgraded. Run 'apt list --upgradable' to see them.
2019-03-02 19:45:22 root@micha:/tmp# apt list --upgradable
Listing... Done
nano/testing 3.2-2 armhf [upgradable from: 3.2-1]
python3-pkg-resources/testing 40.8.0-1 all [upgradable from: 40.7.1-1]
python3-requests/testing 2.21.0-1 all [upgradable from: 2.20.0-2]
python3-setuptools/testing 40.8.0-1 all [upgradable from: 40.7.1-1]
2019-03-02 19:45:30 root@micha:/tmp# apt full-upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  nano python3-pkg-resources python3-requests python3-setuptools
4 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 1,048 kB of archives.
After this operation, 13.3 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian buster/main armhf nano armhf 3.2-2 [523 kB]
Get:2 http://mirrors.dotsrc.org/raspbian/raspbian buster/main armhf python3-setuptools all 40.8.0-1 [306 kB]
Get:3 http://mirrors.dotsrc.org/raspbian/raspbian buster/main armhf python3-pkg-resources all 40.8.0-1 [153 kB]
Get:4 http://mirrors.dotsrc.org/raspbian/raspbian buster/main armhf python3-requests all 2.21.0-1 [66.9 kB]
Fetched 1,048 kB in 1s (1,214 kB/s)
(Reading database ... 23219 files and directories currently installed.)
Preparing to unpack .../archives/nano_3.2-2_armhf.deb ...
Unpacking nano (3.2-2) over (3.2-1) ...
Preparing to unpack .../python3-setuptools_40.8.0-1_all.deb ...
Unpacking python3-setuptools (40.8.0-1) over (40.7.1-1) ...
Preparing to unpack .../python3-pkg-resources_40.8.0-1_all.deb ...
Unpacking python3-pkg-resources (40.8.0-1) over (40.7.1-1) ...
Preparing to unpack .../python3-requests_2.21.0-1_all.deb ...
Unpacking python3-requests (2.21.0-1) over (2.20.0-2) ...
Setting up python3-pkg-resources (40.8.0-1) ...
Setting up python3-setuptools (40.8.0-1) ...
Setting up python3-requests (2.21.0-1) ...
Setting up nano (3.2-2) ...
2019-03-02 19:46:05 root@micha:/tmp#
```

To be recognized here, `raspbian.raspberrypi.org` is indeed a mirror director, as can be seen RWTH Aachen and dotsrc.org mirrors are used finally to download the packages, sadly not via HTTPS where available. So TO `http://mirror.iway.ch` was also chosen via mirror director then. And there seems to be something wrong with this mirror...
`https://mirror.iway.ch/raspbian/raspbian/dists/buster/InRelease` fails to open, when applying this mirror manually, I get:
```
2019-03-02 19:52:03 root@micha:/tmp# apt update
Hit:1 https://archive.raspberrypi.org/debian buster InRelease
Get:2 http://mirror.iway.ch/raspbian/raspbian buster InRelease [15.0 kB]
Get:2 http://mirror.iway.ch/raspbian/raspbian buster InRelease [15.0 kB]
Get:2 http://mirror.iway.ch/raspbian/raspbian buster InRelease [15.0 kB]
Get:2 http://mirror.iway.ch/raspbian/raspbian buster InRelease [15.0 kB]
Get:2 http://mirror.iway.ch/raspbian/raspbian buster InRelease [15.0 kB]
Get:2 http://mirror.iway.ch/raspbian/raspbian buster InRelease [15.0 kB]
Get:2 http://mirror.iway.ch/raspbian/raspbian buster InRelease [15.0 kB]
Get:2 http://mirror.iway.ch/raspbian/raspbian buster InRelease [15.0 kB]
Get:2 http://mirror.iway.ch/raspbian/raspbian buster InRelease [15.0 kB]
...
```
never ending.

To compare http://mirrors.dotsrc.org/raspbian/raspbian/dists/buster/InRelease opens well and contains the hashes.

MichaIng, 2019-03-03:
Okay I mark this as closed. This particular mirror is still faulty. Also it has no been synced since March 1. I hope they resolve it quickly or otherwise get removed from the mirror director...

But nothing reasonable we can do our side.

closed 2019-03-03 by MichaIng

copain60, 2019-03-05:
Hello MichaIng

Thanks for helping to resolve ticket #2608. Let's see what happens next.

I'm contacting you on this way because I stumbled over a problem 
installing pyephem on dietpi. And you are one of the experts for this 
raspi distro. Maybe you could help me sorting this out?

I am fairly new to raspi and dietpi. Unfortunately I could not find 
anything substancial on the web helping me to resolve the issue.

On my raspi I am running weewx, a software for weather station data 
logging. For it's almanach to work I need python 2.7 pyephem (I know 
that 2.7 has eol Jan-2020, but weewx does not work with python 3 yet),  
which I am trying to install. But the installation breaks with the 
following error:

sudo pip install pyephem

Quote:
Command "/usr/bin/python -u -c "import setuptools, 
tokenize;__file__='/tmp/pip-install-1m4B5N/pyephem/setup.py';f=getattr(tokenize, 
'open', open)(__file__);code=f.read().replace('\r\n', 
'\n');f.close();exec(compile(code, __file__, 'exec'))" install --record 
/tmp/pip-record-Bm7Y7Z/install-record.txt 
--single-version-externally-managed --compile" failed with error code 1 
in /tmp/pip-install-1m4B5N/pyephem/

Do you see a solution to resolve this? (Sorry, but I am totally lost 
with this message ...)

Before I installed python as follows:
- sudo apt-get install python-configobj
- sudo apt-get install python-cheetah
- sudo apt-get install python-imaging
- sudo apt-get install python-usb
- sudo apt-get install python-dev

- sudo apt-get install python-setuptools
- easy_install pip

Those I got working, although at first I had some trouble with 
setuptools and pip also.

Any reply would be highly appreciated. Thanks in advance.

Martin

BTW, do you speek German?

MichaIng, 2019-03-05:
@copain60 
Deutsch geht auch, but let's keep it in English, so other properly affected users can follow 😉.

Why did you install `pip` via `easy_install`? Usually I would do `sudo apt-get install python-pip`.
Not sure how and form what source `easy_install` installs pip from, but since it's pip that fails in your case, at least I would give it a try.

copain60, 2019-03-06:
@MichaIng

Many thanks, appreciate your support.

I removed python-dev, -setuptools, -pip and pyephem. Thereafter I reinstalled all 4 packages

sudo apt-get install python-dev
sudo apt-get install python-setuptools
sudo apt-get install python-pip
sudo pip install pyephem

and got at pyephem again the following error:

Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-sq60s8/pyephem/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-vKcYlO/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-install-sq60s8/pyephem/

Would you have any additional advice as to what I could do next?

Copain

MichaIng, 2019-03-06:
@copain60 
Can you try:
`pip install -U setuptools`
and
`apt install build-essential`

copain60, 2019-03-07:
@MichaIng
Solved! Many thanks for your support.

Is the 'correct' installation sequence as follows?
1) sudo apt-get install python-dev
2) sudo apt-get install python-setuptools
3) pip install -U setuptools
4) sudo apt-get install python-pip
5) apt install build-essential
6) sudo pip install pyephem

Please confirm. Thanks

MichaIng, 2019-03-07:
@copain60 
Jep that looks good. I am not sure if `pip install -U setuptools` and `apt install build-essential` are both required or only one of them. This was just what I found during web search. Will do a quick test.

If `pip install -U setuptools` was the required step, actually it is better to skip `apt-get install python-setuptools` since an APT upgrade would overwrite the pip update most likely. APT and pip do not know or respect each others version strings, so generally install modules only via one method.

copain60, 2019-03-12:
@MichaIng
I tested the the installation sequence as suggested above and got the following error:

Quote:
root@DietPi:~# sudo pip install pyephem
Collecting pyephem
  Downloading https://files.pythonhosted.org/packages/57/a8/70bb00ea0b71680afdc8                                                                             9779fbbb58d53c9a4da1818d5ef5359b179b46d1/pyephem-3.7.6.0.tar.gz (739kB)
    100% |████████████████████████████████| 747kB 288kB/s
Building wheels for collected packages: pyephem
  Running setup.py bdist_wheel for pyephem ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-JSg8e4/pyephem/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpS2flVPpip-wheel- --python-tag cp27:
 usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help

  error: invalid command 'bdist_wheel'

  ----------------------------------------
  Failed building wheel for pyephem
  Running setup.py clean for pyephem
Failed to build pyephem
Installing collected packages: pyephem
  Running setup.py install for pyephem ... done
Successfully installed pyephem-3.7.6.0

It seems as it still installed pyephem despite the error, which did not make much sense to me.

Finally, the only sequence working for me without any errors is as follows:

sudo apt-get install python-dev
sudo apt-get install python-setuptools
sudo easy_install pip
sudo apt install build-essential
sudo pip install pyephem

What are your conclusions? Is this of any help re dietpi development?

copain60, 2019-03-12:
@MichaIng 
Good to know, that the successful installation message prevails. That would most probably explain, why the weewx weather logging software ran flawlessly despite the wheels installation error.

I have some user experience of Linux Mint on PC, but I am new to RPi and dietpi. What I learned today is, that Linux offers several ways to install software packets. The only problem is, you need to know them in order to find the most suitable one ...

Thanks for all your help.

MichaIng, 2019-03-12:
@copain60 
Jep Linux itself is not bound to any software install method, nor is e.g. Windows, to be true, only Apple is most strict in this. Debian generally uses APT, but not all software is available via official APT repo (or even own/3rd party repo). Some Python modules are available on Debian APT but not all and for this Python has it's own module manager `pip`, Node.js has `npm` etc. etc. and of course one can always compile any software manually from source.

Linux/Debian tries to keep things a bid separated:
- APT packages install files to `/(usr/)(s)bin` and `/(usr/)(s)lib` while custom builds + `pip` + `npm` by default use `/usr/local/(s)bin` and `/usr/local/lib`. In case both binaries exist, according to default `$PATH` variable `/usr/local/(s)bin` will be preferred, however this clean separation so that APT + custom installs work well side-by-side does not always work 100% without manual steps, as seem above 😉.