MichaIng/DietPi #2609: Problem after update

g7kse, 2019-03-02:
#### Details:
- Date           | Sat  2 Mar 19:13:56 GMT 2019
- Bug report     | N/A
- DietPi version | v6.21.1 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | Unknown Device (armv7l) (index=0)
- Kernel version | #1200 SMP Tue Feb 12 20:27:48 GMT 2019
- Distro         | buster (index=5)
- Command        | G_AGI: libapache2-mod-php7.3 php7.3-curl php7.3-gd php-apcu php7.3-mbstring php7.3-opcache php7.3-zip php7.3-xml php7.3-mysql php7.3-sqlite* php-redis
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. Update via software updater and attempt to install additional software
2. ...

#### Expected behaviour:

- Software install as normal

#### Actual behaviour:

- Fails to install 

#### Extra details:

- Also noted the following:
W: GPG error: https://cdn-aws.deb.debian.org/debian sid InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 7638D0442B90D010 NO_PUBKEY 04EE7237B7D453EC
W: The repository 'https://deb.debian.org/debian sid InRelease' is not signed.

Also affects functionality of existing owncloud installation and ability of wordpress to connect to database. May be coincidence

#### Additional logs:
```
Log file contents:
E: Unable to correct problems, you have held broken packages.

MichaIng, 2019-03-02:
@g7kse 
You are on Raspberry Pi and installed WireGuard recently, right?

I have still no clue how/when this happens, but obviously even with APT preferences that should limit "sid" repo installs to only packages which are not available in other repos, some systems upgrade all a large bunch of packages to "sid". See: https://github.com/MichaIng/DietPi/issues/2581

The huge problem with this is that it is a Debian repo, not Raspbian, which means that the resulting system is kinda hybrid. Also "sid" repo is the experimental/unstable repo which contains untested package updates.

Just to verify that intended preferences are in place:
`cat /etc/apt/preferences.d/dietpi-wireguard*`
__________
Do you have any working backup from before installing WireGuard?

g7kse, 2019-03-03:
It's ok @MichaIng I can do a fresh install and sort it that way. I have a backup of everything I need so won't be an issue for me

MichaIng, 2019-03-03:
@g7kse 
Okay that is good. I just flashed a fresh DietPi RPi image to investigate how this can happen...

MichaIng, 2019-03-08:
Okay I mark this as closed. Issue has been fixed with v6.22 and steps to safely install WireGuard are available.

closed 2019-03-08 by MichaIng