MichaIng/DietPi #2610: error on usb audio

seniorgod, 2019-03-02:
### Creating a bug report/issue

#### Required Information
- DietPi version | `6.21.1 
- Distro version | stretch

i still test my  allo sparky with USBridge after upgrading to 6.21.1 
I have several audio software installed 

shairport-sync from dietpi installation 
spotifyd from GitHub (because i never get shairport-sync to work properly) 
mpd with upmpdcli as upnp frontend for mpd 

i control my music with my mobile phone, running spotify, apple music, mpdeluxe. 
spotiyfyd and shairport-sny use plough:1,0 as audio device. When i her a song from apple music and start playing music by spotify, the system tops smoothly the music from one source and go to the second music supplier (e.g. from apple music to spotify). It seems, that "plughw" stop the first stream and start the new stream. 
But i get from time to time the following error. Is it  error resulting from spotifyd/shairport-sync or is it  problem of the driver? 

Mär 02 23:55:12 DietPi spotifyd[1802]: 23:55:12 [INFO] Track "Berlin" loaded
Mär 02 23:55:12 DietPi spotifyd[1802]: 23:55:12 [ERROR] Alsa error PCM open -16
Mär 02 23:55:12 DietPi spotifyd[1802]: 23:55:12 [ERROR] Could not start audio: Alsa error: PCM open failed
Mär 02 23:55:12 DietPi spotifyd[1802]: 23:55:12 [ERROR] Alsa error PCM open -16
Mär 02 23:55:12 DietPi spotifyd[1802]: 23:55:12 [ERROR] Could not start audio: Alsa error: PCM open failed
Mär 02 23:55:13 DietPi kernel: timeout: still 1 active urbs on EP #1
Mär 02 23:55:14 DietPi kernel: timeout: still 1 active urbs on EP #1
Mär 02 23:55:14 DietPi kernel: ------------[ cut here ]------------
Mär 02 23:55:14 DietPi kernel: WARNING: at /imxhdd/opt/1301TAG/sparky_volumio/kernel/drivers/usb/core/urb.c:328 usb_submit_urb+0x3d4/0x404()
Mär 02 23:55:14 DietPi kernel: URB e1115000 submitted while active
Mär 02 23:55:14 DietPi kernel: Modules linked in: snd_usb_audio snd_hwdep snd_usbmidi_lib nls_cp437 ethernet spidev atc260x_cap_gauge atc260x_irkeypad autofs4
Mär 02 23:55:14 DietPi kernel: CPU: 1 PID: 2200 Comm: shairport-sync Tainted: G        W    3.10.38 #22
Mär 02 23:55:14 DietPi kernel: [<c0015b34>] (unwind_backtrace+0x0/0x138) from [<c001304c>] (show_stack+0x24/0x2c)
Mär 02 23:55:14 DietPi kernel: [<c001304c>] (show_stack+0x24/0x2c) from [<c002beb8>] (warn_slowpath_common+0x4c/0x6c)
Mär 02 23:55:14 DietPi kernel: [<c002beb8>] (warn_slowpath_common+0x4c/0x6c) from [<c002bf6c>] (warn_slowpath_fmt+0x30/0x40)
Mär 02 23:55:14 DietPi kernel: [<c002bf6c>] (warn_slowpath_fmt+0x30/0x40) from [<c03fe604>] (usb_submit_urb+0x3d4/0x404)
Mär 02 23:55:14 DietPi kernel: [<c03fe604>] (usb_submit_urb+0x3d4/0x404) from [<bf051518>] (snd_usb_endpoint_start+0x15c/0x348 [snd_usb_audio])
Mär 02 23:55:14 DietPi kernel: [<bf051518>] (snd_usb_endpoint_start+0x15c/0x348 [snd_usb_audio]) from [<bf0578f8>] (start_endpoints+0xc4/0x140 [snd_usb_audio])
Mär 02 23:55:14 DietPi kernel: [<bf0578f8>] (start_endpoints+0xc4/0x140 [snd_usb_audio]) from [<bf059394>] (snd_usb_pcm_prepare+0x138/0x3a8 [snd_usb_audio])
Mär 02 23:55:14 DietPi kernel: [<bf059394>] (snd_usb_pcm_prepare+0x138/0x3a8 [snd_usb_audio]) from [<c055384c>] (snd_pcm_do_prepare+0x14/0x2c)
Mär 02 23:55:14 DietPi kernel: [<c055384c>] (snd_pcm_do_prepare+0x14/0x2c) from [<c055322c>] (snd_pcm_action_single+0x38/0x78)
Mär 02 23:55:14 DietPi kernel: [<c055322c>] (snd_pcm_action_single+0x38/0x78) from [<c055354c>] (snd_pcm_action_nonatomic+0x54/0x60)
Mär 02 23:55:14 DietPi kernel: [<c055354c>] (snd_pcm_action_nonatomic+0x54/0x60) from [<c055482c>] (snd_pcm_common_ioctl1+0x828/0xd4c)
Mär 02 23:55:14 DietPi kernel: [<c055482c>] (snd_pcm_common_ioctl1+0x828/0xd4c) from [<c0555420>] (snd_pcm_playback_ioctl1+0x138/0x440)
Mär 02 23:55:14 DietPi kernel: [<c0555420>] (snd_pcm_playback_ioctl1+0x138/0x440) from [<c0121e38>] (do_vfs_ioctl+0x80/0x598)
Mär 02 23:55:14 DietPi kernel: [<c0121e38>] (do_vfs_ioctl+0x80/0x598) from [<c01223bc>] (SyS_ioctl+0x6c/0x7c)
Mär 02 23:55:14 DietPi kernel: [<c01223bc>] (SyS_ioctl+0x6c/0x7c) from [<c000f340>] (ret_fast_syscall+0x0/0x30)
Mär 02 23:55:14 DietPi kernel: ---[ end trace d41fcab1096be704 ]---
Mär 02 23:55:14 DietPi kernel: cannot submit urb 6, error -16: unknown error
Mär 02 23:55:14 DietPi kernel: cannot submit urb 6, error -16: unknown error
Mär 02 23:55:14 DietPi shairport-sync[1580]: ALSA lib pcm.c:8309:(snd_pcm_recover) cannot recovery from underrun, prepare failed: Broken pipe
Mär 02 23:55:14 DietPi shairport-sync[1580]: ALSA lib pcm.c:8309:(snd_pcm_recover) cannot recovery from underrun, prepare failed: Broken pipe
Mär 02 23:55:14 DietPi kernel: cannot submit urb 6, error -16: unknown error
Mär 02 23:55:14 DietPi kernel: cannot submit urb 6, error -16: unknown error
Mär 02 23:55:14 DietPi kernel: cannot submit urb 6, error -16: unknown error
Mär 02 23:55:14 DietPi kernel: aotg_ring_enqueue_isoc_td:927, dbg!
Mär 02 23:55:14 DietPi kernel: cannot submit urb 5, error -1: unknown error
Mär 02 23:55:14 DietPi kernel: cannot submit urb 6, error -16: unknown error
Mär 02 23:55:14 DietPi kernel: aotg_hub_urb_dequeue, retval:-43, urb not submitted or unlinked , type = 0 
Mär 02 23:55:14 DietPi kernel: aotg_hub_urb_dequeue, retval:-43, urb not submitted or unlinked , type = 0 
Mär 02 23:55:14 DietPi kernel: aotg_hub_urb_dequeue, retval:-43, urb not submitted or unlinked , type = 0 
Mär 02 23:55:14 DietPi kernel: aotg_hub_urb_dequeue, retval:-43, urb not submitted or unlinked , type = 0 
Mär 02 23:55:14 DietPi kernel: aotg_hub_urb_dequeue, retval:-43, urb not submitted or unlinked , type = 0 
Mär 02 23:55:14 DietPi shairport-sync[1580]: ALSA lib pcm.c:8309:(snd_pcm_recover) cannot recovery from underrun, prepare failed: Broken pipe
Mär 02 23:55:15 DietPi shairport-sync[1580]: ALSA lib pcm.c:8309:(snd_pcm_recover) cannot recovery from underrun, prepare failed: Broken pipe
Mär 02 23:55:15 DietPi shairport-sync[1580]: ALSA lib pcm.c:8309:(snd_pcm_recover) cannot recovery from underrun, prepare failed: Broken pipe
Mär 02 23:55:15 DietPi shairport-sync[1580]: Very large sync error: 140086 frames, with delay: 228286, td_in_frames: 19129, rt: 4750335090, nt: 4750125933, current_delay: 0, seqno: 20351, given timestamp: 455158637.
Mär 02 23:55:15 DietPi kernel: timeout: still 5 active urbs on EP #1
Mär 02 23:55:15 DietPi kernel: cannot submit urb 0, error -16: unknown error
Mär 02 23:55:15 DietPi kernel: cannot submit urb 0, error -16: unknown error
Mär 02 23:55:15 DietPi kernel: cannot submit urb 0, error -16: unknown error
Mär 02 23:55:15 DietPi kernel: cannot submit urb 0, error -16: unknown error

MichaIng, 2019-03-02:
@seniorgod 
Thanks for your report.

Will investigate, need to have a look into the sound card selection code.

As you use MPD, please first apply this fix that causes permission issues that might or might not be related to your issue: https://github.com/MichaIng/DietPi/issues/2522#issuecomment-466820376

MichaIng, 2019-03-03:
Okay, did some search for the actual error messages and indeed it seems to be a kernel bug that is fixed with kernel 3.8: https://github.com/torvalds/linux/commit/ae5943de8c8c4438cbac5cda599ff0b88c224468

Strange is that on our current Sparky SBC image kernel is 3.10.
Can you paste: `uname -a`

seniorgod, 2019-03-03:
To do  new set up with the sparky the Friday, i started with an image i downloaded last December. I think it was Dietpi 6.14. I called dietpi-config to update all to the new version. 

The language of the system is german. sorry. :-) 
So you se some german strings....: 

root@DietPi:~# uname -a
Linux DietPi 3.10.38 #22 SMP PREEMPT Fri Dec 22 10:45:41 IST 2017 armv7l GNU/Linux
root@DietPi:~# cat /etc/asound.conf
pcm.!default {
    type hw
    card 1
    device 0
}
ctl.!default {
    type hw
    card 1
}
root@DietPi:~# aplay -l
**** Liste der Hardware-Geräte (PLAYBACK) ****
Karte 0: atm7059link [atm7059_link], Gerät 0: ATC2603C PCM atc2603c-dai-0 []
  Sub-Geräte: 1/1
  Sub-Gerät #0: subdevice #0
Karte 0: atm7059link [atm7059_link], Gerät 1: HDMI PCM atm7059-hdmi-dai-1 []
  Sub-Geräte: 1/1
  Sub-Gerät #0: subdevice #0
Karte 0: atm7059link [atm7059_link], Gerät 2: SPDIF PCM atm7059-spdif-dai-2 []
  Sub-Geräte: 1/1
  Sub-Gerät #0: subdevice #0
Karte 1: P20 [Project RS USB Audio 2.0], Gerät 0: USB Audio [USB Audio]
  Sub-Geräte: 0/1
  Sub-Gerät #0: subdevice #0
root@DietPi:~#  

mpd is working fine

seniorgod, 2019-03-03:
one question regarding #2522 
I would prefer to change simply mpd.conf and mpd.service by hand
Is that ok? 
What's the advantage by doing it like described with sed and  G_CONFIG?

MichaIng, 2019-03-03:
@seniorgod 
> What's the advantage by doing it like described with sed and G_CONFIG?

No advantage as long as you add them correctly to the right line.
- `sed -Ei '/^(user|group)[[:blank:]]/d' /etc/mpd.conf` will simply remove the `user` and `group` setting line from mpd.conf.
- The `G_CONFIG_INJECT` commands add the user instead to the MPD systemd unit (service file) and make resulting limited permissions only being applied to the final daemon execution. `User=mpd` and `PermissionsStartOnly=true` need to be in the `[Service]` section, possibly existent `Group=` setting is not required and can/should be removed.

The reason why this caused issues was:
- In `mpd.conf` we set `user mpd` and `group dietpi` to run MPD with the desired limited/safe permissions and allow it via group to ready from `dietpi` owned media directories.
- The `mpd` is also inside the `audio` group which allows him to access the audio device interfaces.
- But the group setting in `mpd.conf` does not "add" this group to the users permissions but overrides all user groups instead. So with `group dietpi` the `audio` group permission were lost, leading to access errors.
- To solve this, we now do not touch user and group via `mpd.conf` but instead do with within the systemd unit that inits the MPD daemon. User and group settings there do not override any system-wise applied permission, do it is possible then to have MPD access as group `audio` AND `dietpi`.

If MPD already works fine for you, you possible already changed something about that (e.g. setting `user root`) or recovered your previous `mpd.conf` after update. However the above method is safer then running MPD with root permissions.

seniorgod, 2019-03-03:
thank you, most things are clear. 
My question was more dedicated to the usage of  the tools G_CONFIG_INCJECT. 
When i use that tool to do the changes, will may changes survive the rollout of an new mpd.conf oder mpd.service? If not, why using that tool instead of doing the changes in the files mentioned above manually?

MichaIng, 2019-03-03:
@seniorgod 
`G_CONFIG_INCJECT` is just a function we created to easily add/change settings to config files in a failsafe way: https://github.com/MichaIng/DietPi/blob/dev/dietpi/func/dietpi-globals
- This assures there are no multiple entries of the same setting.
- If the setting is already present but commented, it will use the comment line.
- Via options it is possible to preserve the existing values, add the setting after a specific line and create a backup of the config file before touching.
- And it includes all required escaping of special characters and prints any result and in case errors to terminal.

Mostly we use this internally, but it is available from the command line by default on DietPi systems.

seniorgod, 2019-03-03:
understand. fine. :-)

What we do with the kernel problem?
Does it make sense to do a new compile of sharepoint-sync?

MichaIng, 2019-03-03:
@seniorgod 
Not sure about the kernel issue currently, if it is indeed related to the audio source transition and thus software related or a kernel bug indeed.

**Ah can you verify these errors (timestamps) show up when switching the audio source and/or track and not randomly in the middle of playing any source?**

The first error messages come from `spotifyd`, so I assume, if software related, then it is more likely a `spotifyd` issue then `sharepoint-sync`.
Also our `shairport-sync` packages are already the most current upstream version. But indeed there are several build options, where I am not 100% sure if anything might be related: https://github.com/mikebrady/shairport-sync#building-and-installing

seniorgod, 2019-03-03:
hm,

i loaded spotifyd because spotify-connect-web don’t work for me. (and i have no idea why) 
spotifyd seems to be newer, and as far as i understand it is a new implementation with rust, with uses the new spotify web api.
But i never get a own compile to work. Something in rust was outdated and i get now answer from the spotifyd group. 
As far as i understand the future of the „old“ spotify lib and app keys, there will come the day when we all have to move from the old spotifylib world to a new one…… 
Do you plan to move from spotify-connect-web to spotifyd in future?

MichaIng, 2019-03-03:
@seniorgod 
I think we should add https://github.com/Spotifyd/spotifyd as new install option by times and drop https://github.com/Fornoth/spotify-connect-web when it does not work anymore. Spotifyd is much more actively developed, so seems to be indeed the way to go.

Would be great if you could open a new feature request about this and possible as well add it to our FeatHub page: https://feathub.com/MichaIng/DietPi/

MichaIng, 2019-03-04:
@seniorgod 
Ah just found another open issue. Please try this: https://github.com/MichaIng/DietPi/issues/2427#issuecomment-456205689

In `/DietPi/uEnv.txt`
change
`aotg.urb_fix = 1`
to
`aotg.urb_fix = 0`

Reboot and check if it makes a difference.

seniorgod, 2019-03-05:
@MichaIng 

nice try:-) 

#2427 was created by me. 
aotg.urb_fix = 0 is already set. 

To remember: the behavior if atop_urb_fix != 0 is that sparky hang, after shutting of the DAC while music is playing. 

what i wonder: 
even when spotifyd and shairport have a "problem" working together. 
This should not lead to an kernel trace. Isn't it?

MichaIng, 2019-03-05:
@seniorgod 
Ah lol recognized that this is your topic as well just now 🤣.

User programs generally can trigger kernel error when using kernel features a wrong way, accessing resources they have no permissions for or in case of general I/O issues due to corruption and stuff. But yeah to be true I am not sure how exactly the audio device is accessed here. Of course ALSA is always middleware and also always shows up in your error log, but possible it is accessed/used wrong by the players... But as well possible that it is indeed a kernel issue Sparky-side: `/imxhdd/opt/1301TAG/sparky_volumio/kernel/drivers/usb/core/urb.c`

Actually would be great to test the exact same setup on another SBC. Hmm actually I have an USB DAC here as well, perhaps I find some time to test it on notebook. Hmm have no spotify or something, have to digging a bid into this to have two sources to switch between. The setup will be totally different, but at least we can sort out a general issue with Debian + ALSA + switch audio source.

But I mark this as low priority now, since functionality is fine and it is a visual issue only.

seniorgod, 2019-03-16:
Hi MichaIng,

i have a new error trace. But today only **spotifyd** was running. No other music client was playing music (but all run as daemons):

`Mär 16 11:13:31 DietPi kernel: aotg_hub_endpoint_disable
Mär 16 11:13:31 DietPi kernel: aotg_hcd aotg_hcd.1: <EP DISABLE> ep1 index 1 from ep [out]
Mär 16 11:13:31 DietPi kernel: aotg_hub_endpoint_disable
Mär 16 11:13:31 DietPi kernel: aotg_hcd aotg_hcd.1: <EP DISABLE> ep1 index 3 from ep [in]
Mär 16 11:13:31 DietPi spotifyd[27338]: 11:13:31 [INFO] Loading track "Halfway Up The Hindu Kush" with Spotify URI "spotify:track:2ibjom0ewZZYf9heGgXIhc"
Mär 16 11:13:32 DietPi spotifyd[27338]: 11:13:32 [INFO] Track "Halfway Up The Hindu Kush" loaded
Mär 16 11:13:32 DietPi kernel: snd_pcm_mmap,default
Mär 16 11:13:32 DietPi kernel: iso_packets:0, bInterval:1, urb_interval:1, reg_con:0x4
Mär 16 11:13:32 DietPi kernel: iso_packets:0, bInterval:4, urb_interval:8, reg_con:0x4
Mär 16 11:16:38 DietPi kernel: noticd:aotg_ring_dequeue_td, IN1IRQ:0x0; OUT1IRQ:0x2
Mär 16 11:16:38 DietPi kernel: aotg_hub_endpoint_disable
Mär 16 11:16:38 DietPi kernel: aotg_hcd aotg_hcd.1: <EP DISABLE> ep1 index 1 from ep [out]
Mär 16 11:16:38 DietPi kernel: aotg_hub_endpoint_disable
Mär 16 11:16:38 DietPi kernel: aotg_hcd aotg_hcd.1: <EP DISABLE> ep1 index 3 from ep [in]
Mär 16 11:16:38 DietPi spotifyd[27338]: 11:16:38 [INFO] Loading track "Pumped Up Kicks" with Spotify URI "spotify:track:7w87IxuO7BDcJ3YUqCyMTT"
Mär 16 11:16:39 DietPi spotifyd[27338]: 11:16:39 [INFO] Track "Pumped Up Kicks" loaded
Mär 16 11:16:39 DietPi kernel: snd_pcm_mmap,default
Mär 16 11:16:39 DietPi kernel: iso_packets:0, bInterval:1, urb_interval:1, reg_con:0x4
Mär 16 11:16:39 DietPi kernel: iso_packets:0, bInterval:4, urb_interval:8, reg_con:0x4
Mär 16 11:16:44 DietPi kernel: TX ERROR status: 0x00760000
Mär 16 11:17:01 DietPi CRON[27354]: pam_unix(cron:session): session opened for user root by (uid=0)
Mär 16 11:17:01 DietPi CRON[27355]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Mär 16 11:17:03 DietPi CRON[27354]: pam_unix(cron:session): session closed for user root
Mär 16 11:20:38 DietPi kernel: aotg_hub_endpoint_disable
Mär 16 11:20:38 DietPi kernel: aotg_hcd aotg_hcd.1: <EP DISABLE> ep1 index 1 from ep [out]
Mär 16 11:20:38 DietPi kernel: aotg_hub_endpoint_disable
Mär 16 11:20:38 DietPi kernel: aotg_hcd aotg_hcd.1: <EP DISABLE> ep1 index 3 from ep [in]
Mär 16 11:20:38 DietPi spotifyd[27338]: 11:20:38 [INFO] Loading track "Crystal Frontier - Original Version" with Spotify URI "spotify:track:4osMHG9HvVIEqTciAPtZKK"
Mär 16 11:20:38 DietPi spotifyd[27338]: 11:20:38 [INFO] Track "Crystal Frontier" loaded
Mär 16 11:20:38 DietPi kernel: snd_pcm_mmap,default
Mär 16 11:20:38 DietPi kernel: iso_packets:0, bInterval:1, urb_interval:1, reg_con:0x4
Mär 16 11:20:38 DietPi kernel: iso_packets:0, bInterval:4, urb_interval:8, reg_con:0x4
Mär 16 11:20:39 DietPi kernel: ------------[ cut here ]------------
Mär 16 11:20:39 DietPi kernel: WARNING: at /imxhdd/opt/1301TAG/sparky_volumio/kernel/net/sched/sch_generic.c:255 dev_watchdog+0x24c/0x26c()
Mär 16 11:20:39 DietPi kernel: NETDEV WATCHDOG: eth0 (owl-ethernet): transmit queue 0 timed out
Mär 16 11:20:39 DietPi kernel: Modules linked in: snd_usb_audio snd_hwdep snd_usbmidi_lib nls_cp437 ethernet spidev atc260x_cap_gauge atc260x_irkeypad autofs4
Mär 16 11:20:39 DietPi kernel: CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W    3.10.38 #22
Mär 16 11:20:39 DietPi kernel: [<c0015b34>] (unwind_backtrace+0x0/0x138) from [<c001304c>] (show_stack+0x24/0x2c)
Mär 16 11:20:39 DietPi kernel: [<c001304c>] (show_stack+0x24/0x2c) from [<c002beb8>] (warn_slowpath_common+0x4c/0x6c)
Mär 16 11:20:39 DietPi kernel: [<c002beb8>] (warn_slowpath_common+0x4c/0x6c) from [<c002bf6c>] (warn_slowpath_fmt+0x30/0x40)
Mär 16 11:20:39 DietPi kernel: [<c002bf6c>] (warn_slowpath_fmt+0x30/0x40) from [<c05ea148>] (dev_watchdog+0x24c/0x26c)
Mär 16 11:20:39 DietPi kernel: [<c05ea148>] (dev_watchdog+0x24c/0x26c) from [<c003a2d8>] (call_timer_fn+0x3c/0x154)
Mär 16 11:20:39 DietPi kernel: [<c003a2d8>] (call_timer_fn+0x3c/0x154) from [<c003ac90>] (run_timer_softirq+0x1c0/0x2c4)
Mär 16 11:20:39 DietPi kernel: [<c003ac90>] (run_timer_softirq+0x1c0/0x2c4) from [<c00339e0>] (__do_softirq+0xf4/0x2a0)
Mär 16 11:20:39 DietPi kernel: [<c00339e0>] (__do_softirq+0xf4/0x2a0) from [<c0033c1c>] (do_softirq+0x4c/0x58)
Mär 16 11:20:39 DietPi kernel: [<c0033c1c>] (do_softirq+0x4c/0x58) from [<c0033e90>] (irq_exit+0x90/0xc8)
Mär 16 11:20:39 DietPi kernel: [<c0033e90>] (irq_exit+0x90/0xc8) from [<c000fbf8>] (handle_IRQ+0x3c/0x94)
Mär 16 11:20:39 DietPi kernel: [<c000fbf8>] (handle_IRQ+0x3c/0x94) from [<c00085e0>] (gic_handle_irq+0x28/0x5c)
Mär 16 11:20:39 DietPi kernel: [<c00085e0>] (gic_handle_irq+0x28/0x5c) from [<c000ef40>] (__irq_svc+0x40/0x70)
Mär 16 11:20:39 DietPi kernel: Exception stack(0xc0c77f68 to 0xc0c77fb0)
Mär 16 11:20:39 DietPi kernel: 7f60:                   ffffffed 00a37000 c0c8b6e4 00000000 c0c76000 c0c76000
Mär 16 11:20:39 DietPi kernel: 7f80: c0c76000 c0d29dec c0c89ed4 414fc091 c07c6920 c0d295bd 00000000 c0c77fb0
Mär 16 11:20:39 DietPi kernel: 7fa0: c0010050 c0010048 60010013 ffffffff
Mär 16 11:20:39 DietPi kernel: [<c000ef40>] (__irq_svc+0x40/0x70) from [<c0010048>] (arch_cpu_idle+0x28/0x38)
Mär 16 11:20:39 DietPi kernel: [<c0010048>] (arch_cpu_idle+0x28/0x38) from [<c00736b8>] (cpu_startup_entry+0x68/0x24c)
Mär 16 11:20:39 DietPi kernel: [<c00736b8>] (cpu_startup_entry+0x68/0x24c) from [<c0c00a34>] (start_kernel+0x2c4/0x320)
Mär 16 11:20:39 DietPi kernel: ---[ end trace 715bce8250835476 ]---
Mär 16 11:20:40 DietPi kernel: TX ERROR status: 0x00760000
Mär 16 11:37:33 DietPi sshd[27487]: Accepted password for root from 192.168.178.149 port 58873 ssh2
Mär 16 11:37:33 DietPi sshd[27487]: pam_unix(sshd:session): session opened for user root by (uid=0)`

MichaIng, 2019-03-16:
@seniorgod 
If you find some time, one thing you could try is using an alternative `uImage` I just found on the Sparky GitHub page while seeking for `uEnv.txt` documentation: https://github.com/sparkysbc/sparky_linux_images/blob/master/usbdac_mod_uImage.tar.bz2

I cannot say reliably if it is newer or older than the one from our image, since ours has overwritten timestamps from image creation date. There is no documentation about it so I have no idea what this uImage does differently than the default one, but according to the same it has something to do with USB DACs.

So if you find some time you could:
- At best create a full SDcard clone
- Download this uImage and place it instead of the old:
```
mv /boot/uImage /boot/uImage.bak
wget https://github.com/sparkysbc/sparky_linux_images/raw/master/usbdac_mod_uImage.tar.bz2
tar -xf usbdac_mod_uImage.tar.bz2
rm usbdac_mod_uImage.tar.bz2
mv uImage /boot/uImage
```
- Hope that it is otherwise compatible and the system boots. Perhaps it has some positive influence on the USB DAC.
- If the SBC does not boot or you face other issues, simply recovering `/boot/uImage.bak` should do the job. If some more damage happened to the system (very unlikely, as it is only the bootloader), recover the SDcard clone.

Fourdee, 2019-03-22:
@MichaIng 

> I cannot say reliably if it is newer or older than the one from our image

All sparky kernel updates are patched during our updates. Allo provides these to me directly. Therefor once DietPi is updated, users are running the latest kernel, as provided by Allo.

MichaIng, 2019-03-22:
@Fourdee 
Good to know. Do you know something about this USB DAC image: https://github.com/sparkysbc/sparky_linux_images/blob/master/usbdac_mod_uImage.tar.bz2

seniorgod, 2019-03-22:
Dear all,

i get an answer in spotifyd project: 
https://github.com/Spotifyd/spotifyd/issues/218#issuecomment-473569353

They wonder if the error is caused by insufficient data supply via network. I have no idea. But even if the date stream is broken, should that result in a kernel trace? 

By the way: i compiled spotifyd from scratch and will monitor the situation.

MichaIng, 2019-03-22:
@seniorgod 
Worth to try. Hmm I would have expected different behaviour as well if the bandwidth is too low, e.g. simply audio lag or hang and good.
However my knowledge is too low the be sure what CAN lead to kernel error traces and what not 😉.

SirWindfield, 2019-03-22:
Just switching in from the other thread. Insufficient data supply was my first guess as I new that the underlying library behind `spotifyd` does not reconnect when it loses internet connection. The session becomes invalid then and playback isn't possible without restarting `spotifyd`. 

And seeing that it had to do with an ethernet error I just assumed that it might be that problem. But reading through this thread I might be wrong altogether.

MichaIng, 2019-03-22:
@SirWindfield @seniorgod 
Indeed this would be an interesting test to use two (or one) other audio source to place/switch between and see if same/similar error traces appear. I think you did not yet sort out that/if it is a `spotifyd-only` issue?

seniorgod, 2019-03-24:
The spotify's client i used to use was a binary download. To be sure, that the binary fit to my sparky/DietPi i decided to compile spotifyd from scratch. To be able to do so, i needed to install rustc and cargo from debian/testing..... If you interested in the procedure, i can send the steps....
Since then i had no new problem, until today. I faced the problem that i already posted on spotifyd page. But this is obviously a problem, which seems to be started by low network traffic. That means, the error described above and the error i posted at spotifyd  Spotifyd/spotifyd#218 (comment) are different problems. Here is the stack trace from today: 

Mär 24 12:43:50 DietPi spotifyd[16557]: 12:43:50 [INFO] Track "Little Child Runnin' Wild (2014 Remaster)" loaded
Mär 24 12:43:50 DietPi kernel: snd_pcm_mmap,default
Mär 24 12:43:50 DietPi kernel: iso_packets:0, bInterval:1, urb_interval:1, reg_con:0x4
Mär 24 12:43:50 DietPi kernel: iso_packets:0, bInterval:4, urb_interval:8, reg_con:0x4
Mär 24 12:43:50 DietPi kernel: ------------[ cut here ]------------
Mär 24 12:43:50 DietPi kernel: WARNING: at /imxhdd/opt/1301TAG/sparky_volumio/kernel/net/sched/sch_generic.c:255 dev_watchdog+0x24c/0x26c()
Mär 24 12:43:50 DietPi kernel: NETDEV WATCHDOG: eth0 (owl-ethernet): transmit queue 0 timed out
Mär 24 12:43:50 DietPi kernel: Modules linked in: nls_cp437 snd_usb_audio snd_hwdep snd_usbmidi_lib ethernet spidev atc260x_cap_gauge atc260x_irkeypad autofs4
Mär 24 12:43:50 DietPi kernel: CPU: 2 PID: 0 Comm: swapper/2 Tainted: G        W    3.10.38 #22
Mär 24 12:43:50 DietPi kernel: [<c0015b34>] (unwind_backtrace+0x0/0x138) from [<c001304c>] (show_stack+0x24/0x2c)
Mär 24 12:43:50 DietPi kernel: [<c001304c>] (show_stack+0x24/0x2c) from [<c002beb8>] (warn_slowpath_common+0x4c/0x6c)
Mär 24 12:43:50 DietPi kernel: [<c002beb8>] (warn_slowpath_common+0x4c/0x6c) from [<c002bf6c>] (warn_slowpath_fmt+0x30/0x40)
Mär 24 12:43:50 DietPi kernel: [<c002bf6c>] (warn_slowpath_fmt+0x30/0x40) from [<c05ea148>] (dev_watchdog+0x24c/0x26c)
Mär 24 12:43:50 DietPi kernel: [<c05ea148>] (dev_watchdog+0x24c/0x26c) from [<c003a2d8>] (call_timer_fn+0x3c/0x154)
Mär 24 12:43:50 DietPi kernel: [<c003a2d8>] (call_timer_fn+0x3c/0x154) from [<c003ac90>] (run_timer_softirq+0x1c0/0x2c4)
Mär 24 12:43:50 DietPi kernel: [<c003ac90>] (run_timer_softirq+0x1c0/0x2c4) from [<c00339e0>] (__do_softirq+0xf4/0x2a0)
Mär 24 12:43:50 DietPi kernel: [<c00339e0>] (__do_softirq+0xf4/0x2a0) from [<c0033c1c>] (do_softirq+0x4c/0x58)
Mär 24 12:43:50 DietPi kernel: [<c0033c1c>] (do_softirq+0x4c/0x58) from [<c0033e90>] (irq_exit+0x90/0xc8)
Mär 24 12:43:50 DietPi kernel: [<c0033e90>] (irq_exit+0x90/0xc8) from [<c000fbf8>] (handle_IRQ+0x3c/0x94)
Mär 24 12:43:50 DietPi kernel: [<c000fbf8>] (handle_IRQ+0x3c/0x94) from [<c00085e0>] (gic_handle_irq+0x28/0x5c)
Mär 24 12:43:51 DietPi kernel: [<c00085e0>] (gic_handle_irq+0x28/0x5c) from [<c000ef40>] (__irq_svc+0x40/0x70)
Mär 24 12:43:51 DietPi kernel: Exception stack(0xe2a13f98 to 0xe2a13fe0)
Mär 24 12:43:51 DietPi kernel: 3f80:                                                       ffffffed 00a49000
Mär 24 12:43:51 DietPi kernel: 3fa0: c0c8b6e4 00000000 e2a12000 e2a12000 e2a12000 c0d29dec c0c89ed4 414fc091
Mär 24 12:43:51 DietPi kernel: 3fc0: c07c6920 c0d295bd 00000000 e2a13fe0 c0010050 c0010048 600f0013 ffffffff
Mär 24 12:43:51 DietPi kernel: [<c000ef40>] (__irq_svc+0x40/0x70) from [<c0010048>] (arch_cpu_idle+0x28/0x38)
Mär 24 12:43:51 DietPi kernel: [<c0010048>] (arch_cpu_idle+0x28/0x38) from [<c00736b8>] (cpu_startup_entry+0x68/0x24c)
Mär 24 12:43:51 DietPi kernel: [<c00736b8>] (cpu_startup_entry+0x68/0x24c) from [<0000821c>] (0x821c)
Mär 24 12:43:51 DietPi kernel: ---[ end trace 56440f5c44b4fd9a ]---
Mär 24 12:43:55 DietPi kernel: TX ERROR status: 0x007e0000
Mär 24 12:49:14 DietPi kernel: noticd:aotg_ring_dequeue_td, IN1IRQ:0x0; OUT1IRQ:0x2
Mär 24 12:49:14 DietPi kernel: aotg_hub_endpoint_disable
Mär 24 12:49:14 DietPi kernel: aotg_hcd aotg_hcd.1: <EP DISABLE> ep1 index 1 from ep [out]
Mär 24 12:49:14 DietPi kernel: aotg_hub_endpoint_disable
Mär 24 12:49:14 DietPi kernel: aotg_hcd aotg_hcd.1: <EP DISABLE> ep1 index 3 from ep [in]

yesterday i got a "problem" with airplay (shairport-sync). I was  not at home, my daughter call me because airplay stops playing. My daughter "boot" the sparky by disconnecting/reconnecting the power cables. So i have no loggings about that fail. As a result and also why i sometimes face the error, that shairport-sync stops playing after some seconds i recompiled shairport-sync following the description from GitHub page. Until now there is no new problem. 

@MichaIng I see that there is a new version of DietPi. How could i update Dietpi without overwriting my new compiled shairport-sync? Next question: To be able to compile shairport-sync and spotifyd a had to install some libraries. During that i ran apt-get dist-upgrade. Is that ok or will it create incompatibility together with diet-update?

MichaIng, 2019-03-24:
@seniorgod 
> To be sure, that the binary fit to my sparky/DietPi i decided to compile spotifyd from scratch.

Note that if you are able to start the binary, it is correct. E.g. ARM devices will instantly fail to start x86 binaries and the other way round. But I guess your build from current source includes some fixes.

> I recompiled shairport-sync following the description from GitHub page. Until now there is no new problem.

Ah yeah there was a new release 20 days ago with seems to be a large one, especially what looks like connected to your issue:

> - Greatly improved stability when playing is interrupted by another play request, by the abrupt disappearance of the audio source or by severe network problems.

Ah although this is still a pre-release. Did you install this anyway or the current stable release, which would match our binary: https://github.com/mikebrady/shairport-sync/releases/tag/3.3RC1

> How could i update Dietpi without overwriting my new compiled shairport-sync?

Your shairport-sync install is not touched by the update. See here the software titles where a reinstall is forced: https://github.com/MichaIng/DietPi/blob/master/dietpi/patch_file#L1661-L1664

MichaIng, 2019-05-23:
@seniorgod 
Has been a while. Sparky provides updated USB audio driver files, which could be tested. They also provide native DSD to a large number of additional USB DACs.

To update the drivers, simply do:
```
wget https://raw.githubusercontent.com/sparky-sbc/sparky-test/master/dsd-marantz/snd-usb-audio.ko -O /lib/modules/3.10.38/kernel/sound/usb/snd-usb-audio.ko
wget https://raw.githubusercontent.com/sparky-sbc/sparky-test/master/dsd-marantz/snd-usbmidi-lib.ko -O /lib/modules/3.10.38/kernel/sound/usb/snd-usbmidi-lib.ko
sync
```

Then reboot to have the active. Perhaps it helps with your kernel errors as well, otherwise it should not hurt.

M1m1s, 2019-05-25:
Apologies if I shouldn't post here, but I seem to have a similar issue (not a developer).
I have also sent a bug report, reference code: 95673cb8-1092-4fda-a268-dc7ed4921cab

Issue: USB DAC can't be selected in dietpi-config. I have set it to other options and back to USB-DAC but did not help.
Hardware is USBridge. DAC is Benchmark DAC3.

I have checked that it runs in USB2 mode and restarted it.

Worked perfectly before. Issue appeared after disconnecting USBridge from power.

I clean-installed latest dietpi on USBridge EMMC card:

DietPi v6.24.1 : 09:49 - Sat 25/05/19
 ─────────────────────────────────────────────────────
 - Device model : Sparky SBC (armv7l)

aplay -l results:

**** List of PLAYBACK Hardware Devices ****
card 0: atm7059link [atm7059_link], device 0: ATC2603C PCM atc2603c-dai-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: atm7059link [atm7059_link], device 1: HDMI PCM atm7059-hdmi-dai-1 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: atm7059link [atm7059_link], device 2: SPDIF PCM atm7059-spdif-dai-2 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

Also followed suggestions on https://dietpi.com/phpbb/viewtopic.php?t=2483. Received error:

/DietPi/dietpi/func/dietpi-set_hardware soundcard usb-dac
[  OK  ] DietPi-Set_hardware | Root access verified.
[  OK  ] DietPi-Set_hardware | RootFS R/W access verified.

 DietPi-Set_hardware
─────────────────────────────────────────────────────
 Mode: soundcard (usb-dac)

[ INFO ] DietPi-Set_hardware | Checking for pre-req APT packages: alsa-utils
[ INFO ] DietPi-Set_hardware | Pre-req APT packages were already installed
Simple mixer control 'audio output mode switch',0
  Capabilities: enum
  Items: 'i2s' 'hdmi' 'spdif'
  Item0: 'hdmi'
[  OK  ] DietPi-Set_hardware | Setting in /DietPi/dietpi.txt adjusted: CONFIG_SOUNDCARD=usb-dac
[FAILED] DietPi-Set_hardware | Unable to find a USB-DAC on system.
[  OK  ] DietPi-Set_hardware | Setting in /DietPi/dietpi.txt adjusted: CONFIG_SOUNDCARD=none
[FAILED] soundcard usb-dac | An issue has occurred

Have also updated USB DAC drivers according to post above.

Thanks for any help. Again, if I should post this elsewhere, please let me know.

Fourdee, 2019-05-25:
@M1m1s 

Hi,

Thanks for the report.

Very strange that no USB DAC is being detected, could indicate an issue at a hardware level, especially considering you've reinstalled.

Please try the following:
- Plug the USB DAC into a USB socket on the Sparky (not the USB bridge), then run:
```
lsusb
```
Please paste results
- If you have any other USB DAC available for testing, please do the same as above. Also plug into USB Bridge port, and run ```lsusb``` again.

---

Notes:
```
               ========usb_detect_plugout_event===start det!!========
[    1.903940] 
               /imxhdd/opt/1301TAG/sparky_volumio/kernel/drivers/usb/monitor/umonitor_core.c--497, SYS_MSG_USB_B_OUT
[    1.903961] 
               ----udc_set_plugstate--PLUGSTATE_OUT--
[    1.903983] [PowerGate] name: 'usb3', on: 0, before SPS_PG_CTL: 0x717761
[    1.903995] [PowerGate] name: 'usb3', on: 0, after SPS_PG_CTL: 0x717361
[    1.904005] 
               ------------dwc3_clk_exit----------
[    1.953386] ++++++++cpufreq ready++++++++
[    1.953422] ++++++++pdata->freq_tab[0].freq_clip_max:504000++++++++
[    1.953446] ++++++++pdata->freq_tab[1].freq_clip_max:504000++++++++
[    1.958102] ------------[ cut here ]------------
[    1.958138] WARNING: at /imxhdd/opt/1301TAG/sparky_volumio/kernel/drivers/gpio/gpiolib.c:160 gpio_ensure_requested+0x58/0xb4()
[    1.958149] autorequest GPIO-114
[    1.958158] Modules linked in:
[    1.958178] CPU: 0 PID: 4 Comm: kworker/0:0 Not tainted 3.10.38 #22
[    1.958205] Workqueue: events wait_cpufreq_ready
[    1.958244] [<c0015b34>] (unwind_backtrace+0x0/0x138) from [<c001304c>] (show_stack+0x24/0x2c)
[    1.967142] [<c001304c>] (show_stack+0x24/0x2c) from [<c002beb8>] (warn_slowpath_common+0x4c/0x6c)
[    1.976369] [<c002beb8>] (warn_slowpath_common+0x4c/0x6c) from [<c002bf6c>] (warn_slowpath_fmt+0x30/0x40)
[    1.986218] [<c002bf6c>] (warn_slowpath_fmt+0x30/0x40) from [<c0314f68>] (gpio_ensure_requested+0x58/0xb4)
[    1.996153] [<c0314f68>] (gpio_ensure_requested+0x58/0xb4) from [<c0315ed0>] (gpiod_direction_output+0x60/0x2d4)
[    2.006625] [<c0315ed0>] (gpiod_direction_output+0x60/0x2d4) from [<c04c4b58>] (owl_tmu_read+0x15c/0x1bc)
[    2.016474] [<c04c4b58>] (owl_tmu_read+0x15c/0x1bc) from [<c04c43bc>] (owl_get_temp+0x28/0x58)
[    2.025338] [<c04c43bc>] (owl_get_temp+0x28/0x58) from [<c04c0e9c>] (thermal_zone_get_temp+0x54/0x108)
[    2.034917] [<c04c0e9c>] (thermal_zone_get_temp+0x54/0x108) from [<c04c1e04>] (thermal_zone_device_update+0x24/0xd4)
[    2.045746] [<c04c1e04>] (thermal_zone_device_update+0x24/0xd4) from [<c04c2504>] (thermal_zone_device_register+0x650/0x76c)
[    2.057287] [<c04c2504>] (thermal_zone_device_register+0x650/0x76c) from [<c04c4cac>] (wait_cpufreq_ready+0xf4/0x1b8)
[    2.068204] [<c04c4cac>] (wait_cpufreq_ready+0xf4/0x1b8) from [<c00470dc>] (process_one_work+0x134/0x414)
[    2.078049] [<c00470dc>] (process_one_work+0x134/0x414) from [<c00477b8>] (worker_thread+0x138/0x37c)
[    2.087539] [<c00477b8>] (worker_thread+0x138/0x37c) from [<c004d394>] (kthread+0xa4/0xb0)
[    2.096050] [<c004d394>] (kthread+0xa4/0xb0) from [<c000f3d8>] (ret_from_fork+0x14/0x3c)
[    2.104367] ---[ end trace 83f77b3363f3c293 ]---
[    2.105111] OWL: Kernel Thermal management registered
[    2.475575] systemd-udevd[531]: starting version 212
[    2.533336] 
               /imxhdd/opt/1301TAG/sparky_volumio/kernel/drivers/usb/monitor/umonitor_core.c--746, SYS_MSG_USB_A_IN
[    2.533358] /imxhdd/opt/1301TAG/sparky_volumio/kernel/drivers/usb/monitor/umonitor_api.c--1212, wake_lock !!! 
[    2.533369] 
               ----udc_set_plugstate--PLUGSTATE_IN-2--
[    2.533377] 
               ------------dwc3_clk_init-----ic=1-----
```

M1m1s, 2019-05-25:
Thanks for your fast response.

The issue happened before I reinstalled. I reinstalled in an effort to resolve it.

lsusb with DAC connected to Sparky (it is detected in Roon):

Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 004: ID 1781:0e4f Multiple Vendors 
Bus 003 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

lsusb with DAC connected to USBridge (not detected in Roon):

Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

No other DAC available, I'm afraid.

Thanks again!

Fourdee, 2019-05-25:
@M1m1s 

Appears the USB bridge is faulty. However, I am running a test now on mine to see if we can replicate.

Will update shortly.

Also, did you try the USB1.1 option when selecting sound card with USB bridge?

M1m1s, 2019-05-25:
Yeah I tried that too. Just goes back to “none “.
Of course: USBridge is 20 days out of warranty...

MichaIng, 2019-05-25:
@M1m1s 
I have zero experience with the USBridge but just as a general debug idea, before giving up about the hardware:
```
dmesg
journalctl
```
- Check `dmesg` for red lines or anything voltage related. Who knows it not the power supply has an issue so that the USBridge doesn't run as expected or such.
- The same for `journalctl`, check the first ~3 pages (boot) for any suspicious lines about USB detection/failures.

Is there another chance to test the USBridge general functionality with something else than a DAC?

> Of course: USBridge is 20 days out of warranty...

😞 However always worth to still give it a try and contact Allo regardless. Perhaps they will still help 🙂.

M1m1s, 2019-05-25:
@MichaIng Thanks for chiming in, in addition to @Fourdee!

These lines from dmesg may be relevant:

usb 4-1: device not accepting address 4, error -110
[   48.875340] aotg_hcd aotg_hcd.1: <EP DISABLE> ep0 index 0 from ep [in]
[   48.877044] set port reset!80
[   48.930379] USBRESET IRQ
[   48.930398] aotg_hub_irq: USB device is 2  
[   48.995293] usb 4-1: new full-speed USB device number 5 using aotg_hcd
[   48.995311] think_time:1
[   48.995321] think_time:0x14
[   48.995338] aotg_hcd aotg_hcd.1: device addr : 0x00000000
[   48.995428] irqvector:16, 0x10
[   48.995441] hcd ep err ep_num:0, is_in:0
[   48.995449] err_type :4
[   48.995465] aotg_hcd aotg_hcd.1: HC OUT ep 0 error [0x13] error type [0x04], reset it...
[   54.415285] usb 4-1: device not accepting address 5, error -110
[   54.415334] aotg_hcd aotg_hcd.1: <EP DISABLE> ep0 index 0 from ep [in]
[   54.415360] hub 4-0:1.0: unable to enumerate USB device on port 1
[   60.765504] nf_conntrack: automatic helper assignment is deprecated and it will be removed soon. Use the iptables CT target to attach helpers instead.
[  610.735611] ethernet link state:disconnected

Also:

[ 8029.636205] usb 3-1.2: Product: Benchmark DAC2 USB Audio 2.0
[ 8029.636216] usb 3-1.2: Manufacturer: Benchmark DAC2 
[ 8029.636225] usb 3-1.2: SerialNumber: 0000
[ 8279.593910] irqvector:66, 0x42
[ 8279.593925] hcd ep err ep_num:2, is_in:1
[ 8279.593934] err_type :4
[ 8279.593942] td->err_count:0
[ 8279.626012] usb 3-1.2: USB disconnect, device number 4
[ 8279.627309] aotg_hcd aotg_hcd.0: <EP DISABLE> ep0 index 0 from ep [in]
[ 8279.627330] aotg_hub_endpoint_disable
[ 8279.627341] aotg_hub_endpoint_disable, ep2 dma buf free
[ 8279.627363] aotg_hcd aotg_hcd.0: <EP DISABLE> ep4 index 2 from ep [in]

Other red lines:

[    2.894904] ----monitor_wake_lock for 10s 
[    2.968722] read_mi_head, line 429, magic 0x0, length 0, item num 0
[    2.968754] read Head null
[    2.968776] OWL_HDMI_ERR: failed to read hdcp key from secure storage
[    5.832066] FAT-fs (mmcblk0p1): codepage cp437 not found
[    5.833812] FAT-fs (mmcblk0p1): codepage cp437 not found
[    5.834952] [EXFAT] ===== ffsMountVol =====
[    5.836452] EXFAT: Codepage cp437 not found
[    7.485329] EXT3-fs (mmcblk0p2): error: couldn't mount because of unsupported optional features (240)
[    7.485904] EXT2-fs (mmcblk0p2): error: couldn't mount because of unsupported optional features (240)

[    9.599801] found sparky chip c014639ca06931b provisioned
[    9.599938] idx 1030 newmac 17f7010407

[    9.745836] ATC260X_GAUGE: [get_stored_soc] not store soc

[   11.747015] set port reset!80

[   17.415456] usb 4-1: device descriptor read/64, error -110

No mention of USB in journalctl.

Thank you!

MichaIng, 2019-05-25:
@M1m1s 
Hmm, looks like the USBridge is port 4.1 while onboard USB is port 3.1 where the DAC was recognised correctly.
> [   17.415456] usb 4-1: device descriptor read/64, error -110
[   54.415285] usb 4-1: device not accepting address 5, error -110
[   54.415360] hub 4-0:1.0: unable to enumerate USB device on port 1

- Applies best, states insufficient USB power: https://askubuntu.com/a/644014
- Old but states to check cable (does not apply in your case) or USB speed: https://bugzilla.kernel.org/show_bug.cgi?id=6078
- States hw/firmware/module error: https://superuser.com/questions/742340/error-on-unchanged-system-hub-4-01-0-unable-to-enumerate-usb-device-on-port-1

It might not exactly apply in your case, but with some luck some recent change in the USB audio driver magically fixes the issue. At least worth to give the new testing drivers a try: https://github.com/MichaIng/DietPi/issues/2610#issuecomment-495206219

Also:
- You do not have any other USB device attached, do you? At least nothing visible in logs and lsusb output. In case try to detach all other USB devices.
- Do you have another power supply to test with, just to sort that out (even that there are no under-voltage errors)?

M1m1s, 2019-05-27:
@MichaIng @Fourdee Thank you so much for your very helpful and extensive troubleshooting help for the USBridge (over the weekend, too!). I have performed all your suggestions, as well as verified that the DAC and USB cable work properly. I'm in contact with Allo support. All your help is very appreciated and hopefully will help anyone else who might experience similar issues.

seniorgod, 2019-06-10:
Dear all,

sorry for late reply, but i was absent and afterwards "under heavy workload". 
@M1m1s : Did All support reply. Do you know, if the usbridge was defect? 

@MichaIng: Thank you for noticing the new driver. I installed it last week. 
Because i use airplay currently very rarely, i have no problems at switching the audio source. 
But i still get kernel traces when using spotifyd. I send it to the spotifyd project. One member answered and point to a network problem. It seems, when spotifyd is not getting enough data from network it get into problems. I wonder if insufficient data stream cause kernel traces. But ti be clear: I have no idea how alsa work and if insufficient pcm streams can cause kernel traces. Her is an example from yesterday. The second line after "cut here" is an warning from the network layer:

    Jun 09 17:07:14 DietPi spotifyd[1343]: 17:07:14 [INFO] Track "So soll es sein" loaded
    Jun 09 17:07:14 DietPi kernel: snd_pcm_mmap,default
    Jun 09 17:07:14 DietPi kernel: iso_packets:0, bInterval:1, urb_interval:1, reg_con:0x4
    Jun 09 17:07:14 DietPi kernel: iso_packets:0, bInterval:4, urb_interval:8, reg_con:0x4
    Jun 09 17:10:16 DietPi kernel: noticd:aotg_ring_dequeue_td, IN1IRQ:0x0; OUT1IRQ:0x2
    Jun 09 17:10:16 DietPi kernel: aotg_hub_endpoint_disable
    Jun 09 17:10:16 DietPi kernel: aotg_hcd aotg_hcd.1: <EP DISABLE> ep1 index 1 from ep [out]
    Jun 09 17:10:16 DietPi kernel: aotg_hub_endpoint_disable
    Jun 09 17:10:16 DietPi kernel: aotg_hcd aotg_hcd.1: <EP DISABLE> ep1 index 3 from ep [in]
    Jun 09 17:10:16 DietPi spotifyd[1343]: 17:10:16 [INFO] Loading track "Mindestens in 1000 Jahren"     with Spotify URI "spotify:track:1oKpxxe1L6YdriO8SnV6I1"
    Jun 09 17:10:16 DietPi spotifyd[1343]: 17:10:16 [INFO] Track "Mindestens in 1000 Jahren" loaded
    Jun 09 17:10:16 DietPi kernel: snd_pcm_mmap,default
    Jun 09 17:10:16 DietPi kernel: iso_packets:0, bInterval:1, urb_interval:1, reg_con:0x4
    Jun 09 17:10:16 DietPi kernel: iso_packets:0, bInterval:4, urb_interval:8, reg_con:0x4
    Jun 09 17:10:21 DietPi kernel: ------------[ cut here ]------------
    Jun 09 17:10:21 DietPi kernel: WARNING:at/imxhdd/opt/1301TAG/sparky_volumio/kernel/net/sched/sch_generic.c:255 dev_watchdog+0x24c/0x26c()
    Jun 09 17:10:21 DietPi kernel: NETDEV WATCHDOG: eth0 (owl-ethernet): transmit queue 0 timed out
    Jun 09 17:10:21 DietPi kernel: ethernet link state:linked
    Jun 09 17:10:21 DietPi kernel: Modules linked in:
    Jun 09 17:10:21 DietPi kernel:  snd_usb_audio snd_hwdep snd_usbmidi_lib nls_cp437 ethernet spidev atc260x_cap_gauge atc260x_irkeypad autofs4
    Jun 09 17:10:21 DietPi kernel: CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W    3.10.38 #22
    Jun 09 17:10:21 DietPi kernel: [<c0015b34>] (unwind_backtrace+0x0/0x138) from [<c001304c>] (show_stack+0x24/0x2c)
    Jun 09 17:10:21 DietPi kernel: [<c001304c>] (show_stack+0x24/0x2c) from [<c002beb8>] (warn_slowpath_common+0x4c/0x6c)
    Jun 09 17:10:21 DietPi kernel: [<c002beb8>] (warn_slowpath_common+0x4c/0x6c) from [<c002bf6c>] (warn_slowpath_fmt+0x30/0x40)
    Jun 09 17:10:21 DietPi kernel: [<c002bf6c>] (warn_slowpath_fmt+0x30/0x40) from [<c05ea148>] (dev_watchdog+0x24c/0x26c)
    Jun 09 17:10:21 DietPi kernel: [<c05ea148>] (dev_watchdog+0x24c/0x26c) from [<c003a2d8>] (call_timer_fn+0x3c/0x154)
    Jun 09 17:10:21 DietPi kernel: [<c003a2d8>] (call_timer_fn+0x3c/0x154) from [<c003ac90>] (run_timer_softirq+0x1c0/0x2c4)
    Jun 09 17:10:21 DietPi kernel: [<c003ac90>] (run_timer_softirq+0x1c0/0x2c4) from [<c00339e0>] (__do_softirq+0xf4/0x2a0)
    Jun 09 17:10:21 DietPi kernel: [<c00339e0>] (__do_softirq+0xf4/0x2a0) from [<c0033c1c>] (do_softirq+0x4c/0x58)
    Jun 09 17:10:21 DietPi kernel: [<c0033c1c>] (do_softirq+0x4c/0x58) from [<c0033e90>] (irq_exit+0x90/0xc8)
    Jun 09 17:10:21 DietPi kernel: [<c0033e90>] (irq_exit+0x90/0xc8) from [<c000fbf8>] (handle_IRQ+0x3c/0x94)
    Jun 09 17:10:21 DietPi kernel: [<c000fbf8>] (handle_IRQ+0x3c/0x94) from [<c00085e0>] (gic_handle_irq+0x28/0x5c)
    Jun 09 17:10:21 DietPi kernel: [<c00085e0>] (gic_handle_irq+0x28/0x5c) from [<c000ef40>] (__irq_svc+0x40/0x70)
    Jun 09 17:10:21 DietPi kernel: Exception stack(0xc0c77f68 to 0xc0c77fb0)
    Jun 09 17:10:21 DietPi kernel: 7f60: ffffffed 00a37000 c0c8b6e4 00000000 c0c76000 c0c76000
    Jun 09 17:10:21 DietPi kernel: 7f80: c0c76000 c0d29dec c0c89ed4 414fc091 c07c6920 c0d295bd 00000000 c0c77fb0
    Jun 09 17:10:21 DietPi kernel: 7fa0: c0010050 c0010048 60010013 ffffffff
    Jun 09 17:10:21 DietPi kernel: [<c000ef40>] (__irq_svc+0x40/0x70) from [<c0010048>] (arch_cpu_idle+0x28/0x38)
    Jun 09 17:10:21 DietPi kernel: [<c0010048>] (arch_cpu_idle+0x28/0x38) from [<c00736b8>] (cpu_startup_entry+0x68/0x24c)
    Jun 09 17:10:21 DietPi kernel: [<c00736b8>] (cpu_startup_entry+0x68/0x24c) from [<c0c00a34>] (start_kernel+0x2c4/0x320)
    Jun 09 17:10:21 DietPi kernel: ---[ end trace 548b4a4b1a772f82 ]---
    Jun 09 17:10:23 DietPi kernel: TX ERROR status: 0x00760000

M1m1s, 2019-06-11:
@seniorgod @MichaIng @Fourdee 
Allo support's reply was: "usbridge board some of the LDOs failed , so the chip not detecting."

MichaIng, 2019-06-11:
@M1m1s
LDOs are the linear dropout regulators. So this indeed sounds like a deductive board. Perhaps a warranty case?

MichaIng, 2019-10-05:
@seniorgod 
Just to not forget about this, did you try to build the new shairport sync release: https://github.com/mikebrady/shairport-sync/releases
As mentioned above, the initial 3.3 release notes sound very promising:
> Greatly improved stability when playing is interrupted by another play request, by the abrupt disappearance of the audio source or by severe network problems.

MichaIng, 2020-06-24:
I mark this as closed as of new Sparky SBC image and new Shairport-Sync build. Feel free to reopen if required.

closed 2020-06-24 by MichaIng