MichaIng/DietPi #2612: Install WiringPi cert error

RynFlutsch, 2019-03-03:
Installation of additional software

[image]

MichaIng, 2019-03-04:
Hmm, dietpi.com certificate is not trusted? Accessed it now several times directly via browser and also during software installs and no issues. It's a Let's Encrypt certificate, respectively applied through Cloudflare.

Please try again manually the connection test, if it was just a temporary issue:
`G_CHECK_URL dietpi.com`

If it still fails, try to reinstall/refresh the ca-cerficates packages, then retry the above:
`G_AGI --reinstall ca-certificates`

MichaIng, 2019-03-16:
I will mark as closed as above. Feel free to reopen if error persists after updating the system clock (as above).

closed 2019-03-16 by MichaIng