MichaIng/DietPi #2613: RPi | Fresh image install and WireGuard test

MichaIng, 2019-03-03:
Found an old SDcard. Due to some reports I'm testing our image and especially the WireGuard install to find out how it can happen that several system core packages are upgraded to `Debian Sid`.

First run setup:
- 🈯️ Reach 1. login
- 🈯️ First run update after login until reboot
- 🈯️ Reach 2. login
- 🈯️ First run installs (no additional software selected) after login until reboot
- 🈺 Reach 3. login
  - Debian login prompt hangs a bid, after entering user name, the same with password prompt. Took some seconds before both prompts appear immediately. `DietPi-PostBoot` had finished, so there seem to be some other processes/tasks interfering. However worked well finally.

🈴 Plymouth service active after final login. Interfering with active console leading to annoying visual quirks.
- ctrl+alt+rm reboot hangs on "Starting Hold until boot process finishes up", which looks like as well related to plymouth.
- Next ctrl+alt+rm reboot shows plymouth service stops, no active plymouth service and no quirks.

Adding `logo.nologo` to `cmdline.txt`. No change, not the reason!

Checking installed packages:
- `plymouth` indeed installed, but not marked as manual. `initramfs-tools` installed as well, even that we explicitly do not mark/install it or RPi in PREP.
```
root@DietPi:~# apt purge plymouth
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages were automatically installed and are no longer required:
  cpio initramfs-tools initramfs-tools-core klibc-utils libdrm2 libklibc libnih-dbus1 libnih1 libplymouth4 libpng16-16 linux-base makedev
Use 'apt autoremove' to remove them.
The following packages will be REMOVED:
  mountall* plymouth*
0 upgraded, 0 newly installed, 2 to remove and 0 not upgraded.
After this operation, 584 kB disk space will be freed.
Do you want to continue? [Y/n]
```
- Reason is `mountall` is marked as "required" package, depending on `plymouth` which again depends on `initramfs-tools`.
- [ ] We need to purge `mountall` on RPi during PREP to have all other packages autoremoved:
```
root@DietPi:~# G_AGP mountall
[  OK  ] Root access verified.
[ INFO ] Install verified: mountall
[  OK  ] APT removal for: mountall , please wait...
Reading package lists...
Building dependency tree...
Reading state information...
The following packages were automatically installed and are no longer required:
  cpio initramfs-tools initramfs-tools-core klibc-utils libdrm2 libklibc
  libnih-dbus1 libnih1 libplymouth4 libpng16-16 linux-base makedev plymouth
Use 'apt autoremove' to remove them.
The following packages will be REMOVED:
  mountall*
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
                                                              After this operation, 178 kB disk space will be freed.
(Reading database ... 17040 files and directories currently installed.)                                             (Reading database ...
Removing mountall (2.54) ...
(Reading database ... 17019 files and directories currently installed.)
Purging configuration files for mountall (2.54) ...

[  OK  ] G_AGP: mountall
root@DietPi:~# G_AGA
[  OK  ] APT autoremove + purge, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
The following packages will be REMOVED:
  cpio* initramfs-tools* initramfs-tools-core* klibc-utils* libdrm2* libklibc*
  libnih-dbus1* libnih1* libplymouth4* libpng16-16* linux-base* makedev*
  plymouth*
0 upgraded, 0 newly installed, 13 to remove and 0 not upgraded.
                                                               After this operation, 3,195 kB disk space will be freed.
(Reading database ... 16997 files and directories currently installed.)                                                (Reading database ...
Removing plymouth (0.9.2-4+rpi1) ...
update-initramfs: deferring update (trigger activated)
Removing initramfs-tools (0.130) ...
Removing initramfs-tools-core (0.130) ...
Removing cpio (2.11+dfsg-6) ...
Removing klibc-utils (2.0.4-9+rpi1) ...
Removing 'diversion of /usr/share/initramfs-tools/hooks/klibc to /usr/share/initramfs-tools/hooks/klibc^i-t by klibc-utils'
Removing libdrm2:armhf (2.4.74-1) ...
Removing libklibc (2.0.4-9+rpi1) ...
Removing libnih-dbus1 (1.0.3-8) ...
Removing libnih1 (1.0.3-8) ...
Removing libplymouth4:armhf (0.9.2-4+rpi1) ...
Removing libpng16-16:armhf (1.6.28-1) ...
Removing linux-base (4.5) ...
Removing makedev (2.3.1-93) ...
Processing triggers for libc-bin (2.24-11+deb9u4) ...
(Reading database ... 16707 files and directories currently installed.)
Purging configuration files for klibc-utils (2.0.4-9+rpi1) ...
Purging configuration files for initramfs-tools-core (0.130) ...
Purging configuration files for linux-base (4.5) ...
Purging configuration files for initramfs-tools (0.130) ...
Purging configuration files for plymouth (0.9.2-4+rpi1) ...
Processing triggers for systemd (232-25+deb9u9) ...

[  OK  ] G_AGA
```

MichaIng, 2019-03-03:
Next issue is the ordering cycle getty@tty1.service, rc-local.service and dietpi-postboot, leading to any of those is not started. This is already fixed with v6.22.

MichaIng, 2019-03-03:
### WireGuard install test:
```
root@DietPi:~# echo 'deb https://deb.debian.org/debian/ sid main' > /etc/apt/sources.list.d/dietpi-wireguard.list
root@DietPi:~# echo -e 'Package: *\nPin: release n=sid\nPin-Priority: 99' > /etc/apt/preferences.d/dietpi-wireguard
root@DietPi:~# G_AGUP
[  OK  ] Root access verified.
[  OK  ] APT update, please wait...
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:2 https://archive.raspberrypi.org/debian stretch InRelease
Get:3 https://cdn-aws.deb.debian.org/debian sid InRelease [242 kB]
Ign:3 https://cdn-aws.deb.debian.org/debian sid InRelease
Get:4 https://cdn-aws.deb.debian.org/debian sid/main armhf Packages [8,094 kB]
Get:5 https://cdn-aws.deb.debian.org/debian sid/main Translation-en [6,320 kB]
Fetched 14.7 MB in 25s (570 kB/s)
Reading package lists...
W: GPG error: https://cdn-aws.deb.debian.org/debian sid InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 7638D0442B90D010 NO_PUBKEY 04EE7237B7D453EC
W: The repository 'https://deb.debian.org/debian sid InRelease' is not signed.

[  OK  ] G_AGUP
root@DietPi:~# G_AGDUG
[  OK  ] APT dist-upgrade, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

[  OK  ] G_AGDUG
```
- 🈯️ Generally the preferences do their job
```
root@DietPi:~# apt install wireguard iptables qrencode
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  binutils cpp cpp-6 dkms gcc gcc-6 libasan3 libatomic1 libcc1-0 libgcc-6-dev libgomp1 libip6tc0 libiptc0 libisl15 libmpc3 libmpfr4 libnetfilter-conntrack3
  libnfnetlink0 libpng12-0 libqrencode3 libubsan0 libxtables12 make patch wireguard-dkms wireguard-tools
Suggested packages:
  binutils-doc cpp-doc gcc-6-locales python3-apport menu gcc-multilib manpages-dev autoconf automake libtool flex bison gdb gcc-doc gcc-6-doc libgcc1-dbg libgomp1-dbg
  libitm1-dbg libatomic1-dbg libasan3-dbg liblsan0-dbg libtsan0-dbg libubsan0-dbg libcilkrts5-dbg libmpx2-dbg libquadmath0-dbg make-doc ed diffutils-doc
Recommended packages:
  fakeroot lsb-release libc6-dev | libc-dev libc6-dev
The following NEW packages will be installed:
  binutils cpp cpp-6 dkms gcc gcc-6 iptables libasan3 libatomic1 libcc1-0 libgcc-6-dev libgomp1 libip6tc0 libiptc0 libisl15 libmpc3 libmpfr4 libnetfilter-conntrack3
  libnfnetlink0 libpng12-0 libqrencode3 libubsan0 libxtables12 make patch qrencode wireguard wireguard-dkms wireguard-tools
0 upgraded, 29 newly installed, 0 to remove and 0 not upgraded.
Need to get 17.7 MB of archives.
After this operation, 70.8 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
WARNING: The following packages cannot be authenticated!
  wireguard-dkms wireguard-tools wireguard
Install these packages without verification? [y/N] y
Get:1 http://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian stretch/main armhf libmpfr4 armhf 3.1.5-1 [530 kB]
Get:2 http://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian stretch/main armhf libmpc3 armhf 1.0.3-1 [31.1 kB]
Get:3 http://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian stretch/main armhf libnfnetlink0 armhf 1.0.1-3 [12.1 kB]
Get:4 http://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian stretch/main armhf libip6tc0 armhf 1.6.0+snapshot20161117-6 [66.5 kB]
Get:5 http://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian stretch/main armhf libiptc0 armhf 1.6.0+snapshot20161117-6 [57.7 kB]
Get:7 http://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian stretch/main armhf libxtables12 armhf 1.6.0+snapshot20161117-6 [74.0 kB]
Get:10 http://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian stretch/main armhf libnetfilter-conntrack3 armhf 1.0.6-2 [31.4 kB]
Get:11 http://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian stretch/main armhf iptables armhf 1.6.0+snapshot20161117-6 [259 kB]
Get:12 http://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian stretch/main armhf patch armhf 2.7.5-1+deb9u1 [99.4 kB]
Get:13 http://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian stretch/main armhf binutils armhf 2.28-5 [3,613 kB]
Get:6 https://cdn-aws.deb.debian.org/debian sid/main armhf wireguard-dkms all 0.0.20190227-1 [263 kB]
Get:8 https://cdn-aws.deb.debian.org/debian sid/main armhf wireguard-tools armhf 0.0.20190227-1 [87.9 kB]
Get:14 http://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian stretch/main armhf libisl15 armhf 0.18-1 [426 kB]
Get:9 https://cdn-aws.deb.debian.org/debian sid/main armhf wireguard all 0.0.20190227-1 [21.4 kB]
Get:15 http://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian stretch/main armhf cpp-6 armhf 6.3.0-18+rpi1+deb9u1 [5,140 kB]
Get:16 http://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian stretch/main armhf cpp armhf 4:6.3.0-4 [18.7 kB]
Get:17 http://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian stretch/main armhf libcc1-0 armhf 6.3.0-18+rpi1+deb9u1 [24.9 kB]
Get:18 http://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian stretch/main armhf libgomp1 armhf 6.3.0-18+rpi1+deb9u1 [63.9 kB]
Get:19 http://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian stretch/main armhf libatomic1 armhf 6.3.0-18+rpi1+deb9u1 [6,856 B]
Get:20 http://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian stretch/main armhf libasan3 armhf 6.3.0-18+rpi1+deb9u1 [268 kB]
Get:21 http://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian stretch/main armhf libubsan0 armhf 6.3.0-18+rpi1+deb9u1 [91.7 kB]
[... cut ...]
Get:26 http://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian stretch/main armhf dkms all 2.3-2 [74.8 kB]
Get:27 http://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian stretch/main armhf libpng12-0 armhf 1.2.54-6 [160 kB]
Get:28 http://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian stretch/main armhf libqrencode3 armhf 3.4.4-1 [29.4 kB]
Get:29 http://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian stretch/main armhf qrencode armhf 3.4.4-1 [30.0 kB]
Fetched 17.7 MB in 4s (4,262 kB/s)
```
- 🈯️ As desired only WireGuard packages are pulled from sid repo.

Joulinar, 2019-03-04:
Hi, if you need some testing on RB3+ let me know.

MichaIng, 2019-03-04:
@Joulinar 
Many thanks. Jep I'm currently creating a new RPi image with the above fixes applied. Would be great if you could test this on RPi3 since it is created on RPi2.

I will link it here, when done.

Joulinar, 2019-03-04:
@MichaIng  ok cool, will do once ready

MichaIng, 2019-03-04:
@Joulinar 
Okay new images uploaded to testing folder: https://dietpi.com/downloads/testing/DietPi_RPi-ARMv6-Stretch.7z
Works fine here. Hope the same on RPi3.

Joulinar, 2019-03-04:
Ok will test it tomorrow. Anything special to test next to a clean install + WireGuard?

Joulinar, 2019-03-05:
@MichaIng So, I was running two cycle to test the installation. So far, I could not find any issues. Basic installation + Wireguard are running fine. Except one small think I spotted right after update to 6.21.1

`[  OK  ] DietPi-Survey | Successfully sent survey data

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Completed

[ INFO ] DietPi-Update | Current version : v6.21.1
[ INFO ] DietPi-Update | Latest version  : v6.21.1
/DietPi/dietpi/dietpi-update: line 528: local: can only be used in a function
[  OK  ] DietPi-Update | systemctl restart dietpi-ramdisk`

MichaIng, 2019-03-05:
@Joulinar 
Ah you tested WireGuard as well. It is great that obviously as well in your case the system didn't upgrade to sid. Actually I didn't apply some fix about this to the image, so be careful if you continue to use this image and if any other package from the sid repo is installed by APT, please report back.

> /DietPi/dietpi/dietpi-update: line 528: local: can only be used in a function

Jep this is already fixed. Has no effect since non existing variable has the same affect as the value that was intended to be applied with this line.

Joulinar, 2019-03-05:
@MichaIng well I just did, Are there logs in the PI that you like to have? Just for checking?

MichaIng, 2019-03-05:
@Joulinar 
Boot up and first run setup was the most important thing. I just wanted to be sure that a RPi2 created image runs without issues on RPi3, but actually this is the well known advantage of RPi Raspbian images: They are compatible with all RPi models, you can even simply plug around the same SDcard.

But as you are already asking, let's have a look into the boot log:
`journalctl`

Copy&paste the ~first 20 seconds, so the boot process.

Joulinar, 2019-03-05:
[pi_start.txt](https://github.com/MichaIng/DietPi/files/2931569/pi_start.txt)

There you go

MichaIng, 2019-03-05:
@Joulinar 
Good we did this:
```
Nov 03 17:16:42 DietPi systemd[1]: plymouth-quit.service: Found ordering cycle on plymouth-quit.service/start
Nov 03 17:16:42 DietPi systemd[1]: plymouth-quit.service: Found dependency on rc-local.service/start
Nov 03 17:16:42 DietPi systemd[1]: plymouth-quit.service: Found dependency on dietpi-postboot.service/start
...
Mar 05 12:55:59 DietPi systemd[1]: Starting Show Plymouth Boot Screen...
...
Mar 05 12:56:02 DietPi systemd[1]: Started Terminate Plymouth Boot Screen.
```
This was really from the image I linked above?

Because I actively removed `plymouth` from this with the PREP run, which only brings trouble, installs/requires the generic `initramfs-tools`, while RPi has an own implementation of this and messes with the boot process.

And I actively removed the boot order issue from `dietpi-postboot.service`, just rechecked the image and indeed it is not present there.

Please verify that you did not just install the image from https://dietpi.com/downloads/images/ but the linked above.
- If so, was some dietpi-update running or did you do this manually? Actually since the image version string is `6.21.1`, no update should be done, nor is it required. Also was not done during my install tests.
- If an update is forced, e.g. `dietpi-update -1`, the `dietpi-postboot.service`/`rc-local.service` ordering issue would be reapplied 😉.

Joulinar, 2019-03-05:
@MichaIng hmm strange, Usually I downloaded the image you linked above. Strange one. As well I was not doing any manual update. Just the one that was automatically started during 1st boot. Maybe I stored the image in wrong folder on my computer and used old image to build up SD. Let me try again and use fresh image/ download.

MichaIng, 2019-03-05:
@Joulinar 
Would be great. I hope there is no other package that forcefully pulls plymouth.

And as I already have my RPi2 back on Buster production. Might you check that
`apt search netdata`
indeed gives no result on Raspbian Stretch?

I want to use the APT package from now on, but on Debian it is only available in backports, which does not exist in Raspbian: https://github.com/MichaIng/DietPi/issues/2446

Joulinar, 2019-03-05:
ok the issue was in front of computer as allways. Sorry for that. Now the log with the correct image 

`journalctl`  
[pi_start.txt](https://github.com/MichaIng/DietPi/files/2931910/pi_start.txt)

```
root@DietPi:~# apt search netdata
Sorting... Done
Full Text Search... Done
root@DietPi:~#
```

MichaIng, 2019-03-05:
@Joulinar 
Jep nice, now these quirks do not show up anymore, as expected. Thanks for testing.

One thing I recognized:
```
Mar 05 16:49:09 DietPi wpa_supplicant[264]: dbus: Could not acquire the system bus: org.freedesktop.DBus.Error.FileNotFound - Failed to connect to so
cket /var/run/dbus/system_bus_socket: No such file or directory
Mar 05 16:49:09 DietPi wpa_supplicant[264]: Failed to initialize wpa_supplicant
Mar 05 16:49:09 DietPi systemd[1]: wpa_supplicant.service: Main process exited, code=exited, status=255/n/a
Mar 05 16:49:09 DietPi systemd[1]: Failed to start WPA supplicant.
Mar 05 16:49:09 DietPi systemd[1]: wpa_supplicant.service: Unit entered failed state.
Mar 05 16:49:09 DietPi systemd[1]: wpa_supplicant.service: Failed with result 'exit-code'.
```
- Since when wpasupplicant requires DBus?? Definitely new, since my noteboot just few days ago successfully connected without dbus. Will verify.

Joulinar, 2019-03-05:
During setup, there was a "new" question regarding serialconsole. I selected to disable.

`serialconsole (disable)`

MichaIng, 2019-03-05:
Jep, by default we have serial console enable for first boot, just in case some issue occurs, so one can use it for debugging. But since most users do not require it or even know what it is and how to use, we ask if it should be disabled 😉.

MichaIng, 2019-03-05:
About wpasupplicant and dbus:
```
root@VM-Buster:~# cat /lib/systemd/system/wpa_supplicant.service
[Unit]
Description=WPA supplicant
Before=network.target
After=dbus.service
Wants=network.target

[Service]
Type=dbus
BusName=fi.w1.wpa_supplicant1
ExecStart=/sbin/wpa_supplicant -u -s -O /run/wpa_supplicant

[Install]
WantedBy=multi-user.target
Alias=dbus-fi.w1.wpa_supplicant1.service
root@VM-Buster:~# cat /lib/systemd/system/wpa_supplicant@.service
[Unit]
Description=WPA supplicant daemon (interface-specific version)
Requires=sys-subsystem-net-devices-%i.device
After=sys-subsystem-net-devices-%i.device
Before=network.target
Wants=network.target

# NetworkManager users will probably want the dbus version instead.

[Service]
Type=simple
ExecStart=/sbin/wpa_supplicant -c/etc/wpa_supplicant/wpa_supplicant-%I.conf -Dnl80211,wext -i%I

[Install]
Alias=multi-user.target.wants/wpa_supplicant@%i.service
```
- There are two versions, the one that requires dbus is e.g. used by NetworkManager, while the one we require is the second interface specific one, that is started automatically with the WiFi interface.
- So what we should do on PREP, is disabling `wpa_supplicant.service` which we don't require. I will do a quick test on notebook.

Joulinar, 2019-03-09:
Hi, 

anything special you would like to be tested?

MichaIng, 2019-03-09:
@Joulinar 
Most importantly general flash and first run setup + boot in general most importantly, since I created a non-virtual image the first time, including shrink file system + partition + truncate final image file.

If you could test some video and audio features (e.g. desktop and playing some video and/or Kodi), this would be perfect, just to assure everything hardware related is working as desired on RPi3 as well.

Joulinar, 2019-03-09:
ok will do this evening.

Joulinar, 2019-03-09:
@MichaIng I did another test and first installation is running smooth. Well done

Afterwards I was running Desktop installation of Mate. I takes a while, but it finished without issues and I was able to launch it. Was testing a little bit with Firefox and Chrome on YouTube. Well its running, but not fun. It brings the Pi to his limit. But that’s how it is on such small hardware :)

MichaIng, 2019-03-09:
@Joulinar 
Many thanks for testing.
- On RPi2 as well I remember browser being extremely slow. Was hoping this had been enhanced with RPi3, but yeah most likely and obviously not much. I remember NetSurf was pre-installed on earlier Raspbian versions, which was much faster than Iceweasel (Firefox, that time). Perhaps not at videos but general navigation. But then I read some versions ago they switched to Chromium as default.
- And yes WireGuard takes a long time. The RPi firmware (which you pasted above) is already a long taking step, although it takes not more than 10 minutes on my RPi2. You see all the single firmware files being copied in place... WireGuard module build afterwards might even take longer. Actually was not really paying attention during my test above, ran several other tests and image creation steps meanwhile on notebook, perhaps taking half an hour, before I had the next look onto the RPi screen 😄. But on much faster VM it also takes several minutes. So yeah all together for sure > 10 minutes, both includes CPU-consuming compilation steps...

Okay I will move the image in place 😃!

closed 2019-03-09 by MichaIng