MichaIng/DietPi #2615: Installed RetroPie but do not know how to access again the setup script and proper way to install Kodi

garret, 2019-03-04:
I have installed Retropie on a Raspberry Pi 3B+ following the guide already posted (https://github.com/MichaIng/DietPi/issues/1653). Everything works flawless but now I was wondering how I can re-access the same setup script used initially when installing the basic installation of Retropie. 

I am asking since I observed that the setup script allowed to extend the functionalities of Retropie by adding eventually some more packages/applications. I am mostly new to RetroPie installed on other distros. Could it be that these addions can also be installed directly from the menu of EmulationStation? 

Furthermore, I know that with the RetroPie distro you can install Kodi from their cconsolemenu and then have a voice on EmulationStation. 

On DietPi what would be the best way to install Kodi? Through EmulationStation (still have to figure out how if it is possible to do it like on the RetroPie distro through ththeironsole script) or from the dietpi software optimized? I was thinking that installing it through the software optimized can have as a consequence that is not recognized in EmulationStation, right?

MichaIng, 2019-03-05:
@garret 
You can try to redo the steps, but I am not sure what the setup does to an existing install. At best do a backup of your current RetroPie data dir:
`cp -a /mnt/dietpi_userdata/RetroPie /mnt/dietpi_userdata/RetroPie.bak`

The setup dir should be still there:
`find / -name retropie_setup.sh`
`cd` into the dir and do `git pull` to update to current version, then `./retropie_setup.sh`.

If indeed the setup dir does not exist anymore, you can download it freshly:
```
git clone --depth=1 https://github.com/RetroPie/RetroPie-Setup.git
cd RetroPie-Setup
./retropie_setup.sh
```

MichaIng, 2019-03-16:
I mark as closed, as of answer above. Feel free to reopen if it does not work as intended.

closed 2019-03-16 by MichaIng