MichaIng/DietPi #2616: chmod cannot access '/etc/profile.d/dietpi-*.sh'

RynFlutsch, 2019-03-04:
Tried to upgrade vie dietpi-update
[image]

RynFlutsch, 2019-03-04:
I retried, with success. Update finished..
?

MichaIng, 2019-03-05:
@RynFlutsch 
Jep known issue if an update attempt to v6.20 failed. Indeed rerunning the script works, since the new version has already been downloaded when the error prompts.

Just as there were some other related issue:
`cat /DietPi/dietpi/.install_stage`
should be `2`, otherwise set it:
`echo 2 > /DietPi/dietpi/.install_stage`

But if you already rebooted and did not run into a repeated first run setup, then this is already correct for sure 😉.

MichaIng, 2019-03-12:
I mark as closed since solved. Please feel free to reopen if required.

closed 2019-03-12 by MichaIng