MichaIng/DietPi #2617: Upgrade failed

Phonascus, 2019-03-05:
#### Details:
- Date           | Mon  4 Mar 21:13:44 CST 2019
- Bug report     | N/A
- DietPi version | v6.7.0 (Fourdee/master)
- Img creator    | n/a
- Pre-image      | n/a
- SBC device     | RPi B (ECN0001) (armv6l) (index=0)
- Kernel version | #1110 Mon Apr 16 14:51:42 BST 2018
- Distro         | stretch (index=4)
- Command        | G_AGUG
- Exit code      | 100
- Software title | DietPi-Update

#### Steps to reproduce:

1. dietpi-upgrade

#### Expected behaviour:

- an upgrade

#### Actual behaviour:

- This was reapeated over and over...

Use of uninitialized value $item in hash element at /usr/share/perl5/Debconf/DbDriver/File.pm line 85, <__ANONIO__> chunk 43370.
Use of uninitialized value $value in substitution (s///) at /usr/share/perl5/Debconf/Format/822.pm line 65, <__ANONIO__> line 43386.
Use of uninitialized value $value in substitution (s///) at /usr/share/perl5/Debconf/Format/822.pm line 65, <__ANONIO__> line 43387.
Use of uninitialized value $value in substitution (s///) at /usr/share/perl5/Debconf/Format/822.pm line 65, <__ANONIO__> line 43388.
Use of uninitialized value $value in substitution (s///) at /usr/share/perl5/Debconf/Format/822.pm line 65, <__ANONIO__> line 43389.
Use of uninitialized value $value in substitution (s///) at /usr/share/perl5/Debconf/Format/822.pm line 65, <__ANONIO__> line 43390.
Use of uninitialized value $value in substitution (s///) at /usr/share/perl5/Debconf/Format/822.pm line 65, <__ANONIO__> line 43391.
Use of uninitialized value $value in substitution (s///) at /usr/share/perl5/Debconf/Format/822.pm line 65, <__ANONIO__> line 43392.
Use of uninitialized value $item in hash element at /usr/share/perl5/Debconf/DbDriver/File.pm line 85, <__ANONIO__> chunk 43393.
Use of uninitialized value $value in substitution (s///) at /usr/share/perl5/Debconf/Format/822.pm line 65, <__ANONIO__> line 43410.
Use of uninitialized value $value in substitution (s///) at /usr/share/perl5/Debconf/Format/822.pm line 65, <__ANONIO__> line 43411.
Use of uninitialized value $value in substitution (s///) at /usr/share/perl5/Debconf/Format/822.pm line 65, <__ANONIO__> line 43412.

etc...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
Get:24 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf libcurl3-gnutls armhf 7.52.1-5+deb9u9 [260 kB]
Get:25 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf git armhf 1:2.11.0-3+deb9u4 [3,390 kB]
Get:27 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf multiarch-support armhf 2.24-11+deb9u4 [201 kB]
Get:28 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf dirmngr armhf 2.1.18-8~deb9u4 [547 kB]
Get:29 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf gnupg-agent armhf 2.1.18-8~deb9u4 [499 kB]
Get:30 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf gnupg armhf 2.1.18-8~deb9u4 [1,023 kB]
Get:31 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf tzdata all 2018i-0+deb9u1 [273 kB]
Get:40 https://archive.raspberrypi.org/debian stretch/main armhf firmware-misc-nonfree all 1:20161130-3+rpt4 [2,321 kB]
Get:32 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf libssl1.0.2 armhf 1.0.2r-1~deb9u1 [899 kB]
Get:33 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf libssl1.1 armhf 1.1.0j-1~deb9u1 [1,115 kB]
Get:34 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf libc-l10n all 2.24-11+deb9u4 [820 kB]
Get:35 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf locales all 2.24-11+deb9u4 [3,289 kB]
Get:36 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf openssh-sftp-server armhf 1:7.4p1-10+deb9u6 [32.9 kB]
Get:37 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf openssh-server armhf 1:7.4p1-10+deb9u6 [301 kB]
Get:49 https://archive.raspberrypi.org/debian stretch/main armhf firmware-realtek all 1:20161130-3+rpt4 [342 kB]
Get:38 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf openssh-client armhf 1:7.4p1-10+deb9u6 [699 kB]
Get:39 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf apt-transport-https armhf 1.4.9 [166 kB]
Get:52 https://archive.raspberrypi.org/debian stretch/main armhf raspberrypi-sys-mods armhf 20181127 [10.2 kB]
Get:41 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf curl armhf 7.52.1-5+deb9u9 [220 kB]
Get:42 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf libcurl3 armhf 7.52.1-5+deb9u9 [262 kB]
Get:53 https://archive.raspberrypi.org/debian stretch/main armhf fake-hwclock all 0.11+rpt1 [7,126 B]
Get:43 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf fuse armhf 2.9.7-1+deb9u2 [69.1 kB]
Get:44 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf libfuse2 armhf 2.9.7-1+deb9u2 [115 kB]
Get:45 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf hdparm armhf 9.51+ds-1+deb9u1 [105 kB]
Get:46 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf libtiff5 armhf 4.0.8-2+deb9u4 [220 kB]
Get:47 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf libx11-data all 2:1.6.4-3+deb9u1 [287 kB]
Get:48 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf libx11-6 armhf 2:1.6.4-3+deb9u1 [682 kB]
Get:50 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf libgd3 armhf 2.2.4-2+deb9u4 [114 kB]
Get:51 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf openssl armhf 1.1.0j-1~deb9u1 [712 kB]
Get:54 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf php7.0-readline armhf 7.0.33-0+deb9u1 [10.8 kB]
Get:55 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf php7.0-xml armhf 7.0.33-0+deb9u1 [87.0 kB]
Get:56 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf php7.0-json armhf 7.0.33-0+deb9u1 [15.2 kB]
Get:57 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf php7.0-sqlite3 armhf 7.0.33-0+deb9u1 [20.4 kB]
Get:58 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf php7.0-zip armhf 7.0.33-0+deb9u1 [17.3 kB]
Get:59 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf php7.0-opcache armhf 7.0.33-0+deb9u1 [64.8 kB]
Get:60 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf php7.0-fpm armhf 7.0.33-0+deb9u1 [1,060 kB]
Get:61 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf php7.0-curl armhf 7.0.33-0+deb9u1 [24.3 kB]
Get:62 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf php7.0-mcrypt armhf 7.0.33-0+deb9u1 [12.7 kB]
Get:63 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf php7.0-gd armhf 7.0.33-0+deb9u1 [21.9 kB]
Get:64 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf php7.0-mbstring armhf 7.0.33-0+deb9u1 [415 kB]
Get:65 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf php7.0-cli armhf 7.0.33-0+deb9u1 [1,052 kB]
Get:66 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf php7.0-cgi armhf 7.0.33-0+deb9u1 [1,019 kB]
Get:67 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf php7.0-common armhf 7.0.33-0+deb9u1 [485 kB]
Get:68 http://raspbian.mirrors.lucidnetworks.net/raspbian stretch/main armhf wpasupplicant armhf 2:2.4-1+deb9u2 [793 kB]
Extracting templates from packages: 100%
Preconfiguring packages ...
Fetched 50.3 MB in 21s (2,326 kB/s)
dpkg: unrecoverable fatal error, aborting:
 unable to open files list file for package 'libgdbm3:armhf': No such device or address
E: Sub-process /usr/bin/dpkg returned an error code (2)
```

MichaIng, 2019-03-05:
@Phonascus 
Try to solve the APT error first:
```
apt clean
apt update
apt -f install
```

Now let's upgrade the packages in waves, please break and report back if any of those fails:
```
apt install git tzdata
apt install openssh-server locales
apt install apt-transport-https hdparm fake-hwclock
apt install wpasupplicant curl
apt install gnupg dirmngr
apt install raspberrypi-sys-mods firmware-misc-nonfree firmware-realtek
```

These are all packages we anyway want to have marked as manual installed. They will be updated and force an update of their dependencies as well. The remaining we again try via:
`apt upgrade`

Finally:
`apt full-upgrade`

MichaIng, 2019-03-16:
I will mark as closed. Feel free to reopen if the above step-by-step solution to fix/identify the faulty APT install does not work.

closed 2019-03-16 by MichaIng