MichaIng/DietPi #2618: can't install or remove software because OpenBSD fails

maiker86, 2019-03-05:
#### Details:
- Date           | Tue  5 Mar 07:33:32 GMT 2019
- Bug report     | N/A
- DietPi version | v6.21.1 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B+ (armv7l) (index=3)
- Kernel version | #1200 SMP Tue Feb 12 20:27:48 GMT 2019
- Distro         | stretch (index=4)
- Command        | G_AGUG
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. Select software to install.
2. Install.
3. wait to fail. 

#### Expected behaviour:
1. Select software to install.
2. Install.
3. Successfull installation

#### Actual behaviour:
1. Select software to install.
2. Install.
3. Fails due to OpenBSD i think

#### Additional logs:
```
Log file contents:
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
The following packages were automatically installed and are no longer required:
  libminiupnpc10 libnatpmp1 transmission-common
Use 'apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up openssh-server (1:7.4p1-10+deb9u6) ...
Job for ssh.service failed because the control process exited with error code.
See "systemctl status ssh.service" and "journalctl -xe" for details.
invoke-rc.d: initscript ssh, action "restart" failed.

MichaIng, 2019-03-05:
@maiker86 
Thanks for your report.

You meant OpenSSH server (not OpenBSD), right? OpenBSD is an own OS while OpenSSH server allows you to connect to your system remotely via SSH protocol.

Some questions:
- Is this a fresh install?
- Was the OpenSSH server already installed and in use before, or,
- was Dropbear (our default SSH server) used at first and now you attempt to install OpenSSH server over it?

Anyhow let's check ssh.service logs:
`journalctl -u ssh`

maiker86, 2019-03-06:
I would assume that you are correct, since SSH was still working fine, but i'm pretty sure the error message said OpenBSD. 

To answer your questions:
- It's quite fresh as it's about a week old, however i've installed and unistalled a bunch of stuff just to test things out.
-I've been using dropbear exclusively.
-I have been using dropbear yes and was and is still working. I tried to install transmission, and webmin which failed, which lead me to try to install openSSH which also failed.

This doesn't seem to be an issue specific to OpenSSH, I was messing around with file permission and probably messed something up.

I'm thinking of just reinstalling which wouldn't be so bad, unless you still want the service logs for some other reason?

MichaIng, 2019-03-06:
@maiker86 
Generally the initial reason why Transmission and/or Webmin failed to install would be interesting. However this is not hard to replicate due to other changes you did.

Why did you think that installing OpenSSH could fix the Transmission/Webmin install?
Generally it is important that Dropbear is uninstalled before installing any other SSH server. DietPi-Software should do that automatically.

Do you have local console access? You could try to assure that dropbear is uninstalled:
`G_AGP dropbear*`

Then try to restart OpenSSH:
`systemctl start ssh`

If that fails, show it's error messages:
`journalctl -u ssh`

Or you indeed go with a fresh install, if you feel that it's a rats tail.

maiker86, 2019-03-07:
I tried to replicate the error by trying to installing webmin, interesting that the error is slightly different now, however i've seen this error about binding to port 22 in the forum somewhere so i'll have a look:

```
Setting up openssh-server (1:7.4p1-10+deb9u6) ...
Job for ssh.service failed because the control process exited with error code.
See "systemctl status ssh.service" and "journalctl -xe" for details.
invoke-rc.d: initscript ssh, action "restart" failed.
● ssh.service - OpenBSD Secure Shell server
   Loaded: loaded (/lib/systemd/system/ssh.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2019-03-07 04:46:16 GMT; 30ms ago
  Process: 17819 ExecStart=/usr/sbin/sshd -D $SSHD_OPTS (code=exited, status=255)
  Process: 17816 ExecStartPre=/usr/sbin/sshd -t (code=exited, status=0/SUCCESS)
 Main PID: 17819 (code=exited, status=255)

Mar 07 04:46:16 DietPi systemd[1]: Starting OpenBSD Secure Shell server...
Mar 07 04:46:16 DietPi sshd[17819]: error: Bind to port 22 on 0.0.0.0 faile…use.
Mar 07 04:46:16 DietPi sshd[17819]: error: Bind to port 22 on :: failed: Ad…use.
Mar 07 04:46:16 DietPi sshd[17819]: fatal: Cannot bind any address.
Mar 07 04:46:16 DietPi systemd[1]: ssh.service: Main process exited, code=e…/n/a
Mar 07 04:46:16 DietPi systemd[1]: Failed to start OpenBSD Secure Shell server.
Mar 07 04:46:16 DietPi systemd[1]: ssh.service: Unit entered failed state.
Mar 07 04:46:16 DietPi systemd[1]: ssh.service: Failed with result 'exit-code'.
Hint: Some lines were ellipsized, use -l to show in full.
dpkg: error processing package openssh-server (--configure):
 subprocess installed post-installation script returned error exit status 1
Errors were encountered while processing:
 openssh-server
E: Sub-process /usr/bin/dpkg returned an error code (1)

[FAILED] DietPi-Software | G_AGUG
```

If I Run `journalctl -u ssh`

```
root@DietPi:~# journalctl -u ssh
-- Logs begin at Thu 2019-03-07 03:00:52 GMT, end at Thu 2019-03-07 04:46:16 GMT
. --
Mar 07 03:00:56 DietPi systemd[1]: Starting OpenBSD Secure Shell server...
Mar 07 03:00:56 DietPi systemd[1]: ssh.service: Main process exited, cod
e=exited, status=255/n/a
Mar 07 03:00:56 DietPi systemd[1]: Failed to start OpenBSD Secure Shell
server.
Mar 07 03:00:56 DietPi systemd[1]: ssh.service: Unit entered failed stat
e.
Mar 07 03:00:56 DietPi systemd[1]: ssh.service: Failed with result 'exit
-code'.
Mar 07 04:42:11 DietPi systemd[1]: Starting OpenBSD Secure Shell server...
Mar 07 04:42:11 DietPi sshd[8109]: error: Bind to port 22 on 0.0.0.0 fai
led: Address already in use.
Mar 07 04:42:11 DietPi sshd[8109]: error: Bind to port 22 on :: failed:
Address already in use.
Mar 07 04:42:11 DietPi sshd[8109]: fatal: Cannot bind any address.
Mar 07 04:42:11 DietPi systemd[1]: ssh.service: Main process exited, cod
e=exited, status=255/n/a
Mar 07 04:42:11 DietPi systemd[1]: Failed to start OpenBSD Secure Shell
server.
Mar 07 04:42:11 DietPi systemd[1]: ssh.service: Unit entered failed stat
e.
Mar 07 04:42:11 DietPi systemd[1]: ssh.service: Failed with result 'exit
-code'.-
Mar 07 04:46:16 DietPi systemd[1]: Starting OpenBSD Secure Shell server...
Mar 07 04:46:16 DietPi sshd[17819]: error: Bind to port 22 on 0.0.0.0 fa
iled: Address already in use.
Mar 07 04:46:16 DietPi sshd[17819]: error: Bind to port 22 on :: failed:
 Address already in use.
Mar 07 04:46:16 DietPi sshd[17819]: fatal: Cannot bind any address.
Mar 07 04:46:16 DietPi systemd[1]: ssh.service: Main process exited, cod
e=exited, status=255/n/a
Mar 07 04:46:16 DietPi systemd[1]: Failed to start OpenBSD Secure Shell
server.-
Mar 07 04:46:16 DietPi systemd[1]: ssh.service: Unit entered failed stat
e.
Mar 07 04:46:16 DietPi systemd[1]: ssh.service: Failed with result 'exit
-code'.-
```

I can't recall why I wanted to install OpenSSH, but come to think of it Dropbear is working fine so I think I'll stick with that.
I don't want to waste your time on a problem which would be fixed if I just re-install, it should be quite easy??

MichaIng, 2019-03-07:
@maiker86 
The issue is still and only OpenSSH, which want's to finish it's install when running APT and by this blocks all other installs.
Btw. now I see the `OpenBSD Secure Shell server`, which seems to be the full name, didn't know that 😉.

Did you run `G_AGP dropbear*`?
Without that it is expected to fail, since Dropbear blocks port 22.

However yep, if Dropbear works fine, no need to switch to heavy weight OpenSSH. Instead then run `G_AGP openssh-* && G_AGA` to re-allow using APT for other things.

maiker86, 2019-03-07:
This worked.

Thanks!

MichaIng, 2019-03-07:
@maiker86
Just a quick question:
Did you install OpenSSH initially via dietpi-software or manually via APT install?

In case we need to tweak the install script when switching the SSH server to assure the existing one is uninstalled first and the other one installed afterwards. Actually during tests we did the current SSH session is allowed to stay active, but if that is not true in some cases we need to either allow the switch only from local console and kill any active session. But this then would not work on headless devices. Not sure how to handle it best then, doing the switch automatically on reboot instead? :thinking:

maiker86, 2019-03-07:
I did it initially via the dietpi-software.

but I could always still connect via SSH, just once I killed it for some reason i read somewhere and forgot I wasn't at home...

MichaIng, 2019-03-07:
Hmm, actually the installer already checks for active SSH server and in case stops it and marks it for uninstall. This should assure that port 22 is unbound and the new SSH server install succeeds. The active SSH session stays active but should not block the port.

I will run some more tests about this. Hopefully I can find a reason why this still failed on your case and how to resolve without losing current SSH session.

MichaIng, 2019-03-08:
Okay as your issue is solved and above test result I mark this as closed.
We do all we can to handle port 22 binds on SSH server switch. Killing active sessions kills attached processes as well, leaving a system with most likely broken SSH connectivity. So we must never do that.

Better to resolve these rare bind issues manually then and perhaps one time we will find the reason why systemctl stop dropbear/ssh is not enough in some cases and how to resolve it without killing the active SSH session.

closed 2019-03-08 by MichaIng