MichaIng/DietPi #2619: Software Request | Hiawatha Webserver

Krawei, 2019-03-05:
Hello dietpi-devs,

it would be great to have the hiawatha webserver in the dietpi-software archive. It is more lightwheigh than lighttpd and it offers several built-in protection mechanism against threats like SQL injections, XSS and CSRF attacks and exploit attempts.

So in my opinion hiawatha is a lightwheight and secure basis for a decent nextcloud server.

MichaIng, 2019-03-05:
@Krawei 
Thanks for your request. Looks quite interesting, please add this to our FeatHub page as well and link it here: https://feathub.com/MichaIng/DietPi/

Krawei, 2019-03-05:
Addition to FeatureHub - done

Krawei, 2019-10-01:
Hello @MichaIng : Is there any progress here vor something I can do further?

Kind Regards,
K.

MichaIng, 2019-10-01:
@Krawei 
Since there seem to be not much interest from other users (as well only 1 vote on FeatHub), priority is very low, and especially currently there is a lot of other work to do, as well on already implemented software titles/installers, so generally implementing new software titles is delayed.

If someone does a coding start, I will however quickly assist and implement.

What I am also concerned about is that development on Hiawatha seems to be on hold: https://www.hiawatha-webserver.org/support
I see one minor release from two weeks ago (https://www.hiawatha-webserver.org/changelog) but no related code release on the GitLab repo: https://gitlab.com/hsleisink/hiawatha
However when reading the reasons, probably we can even breathe new life into Hiawatha when adding and promoting it within DietPi 🤔.

MichaIng, 2021-09-24:
We decided to cleanup open issues a bit and close those which will obviously not addressed soon. The FeatHub request stays however open to allow votes and if anyone wants to start working on implementing this software, please feel free to to a start, and we'll help where we can.

Here you find instructions about how to implement a new software title to `dietpi-software`: https://github.com/MichaIng/DietPi/wiki/How-to-add-a-new-software-title

closed 2021-09-24 by MichaIng