MichaIng/DietPi #2620: Gitea custom folder no images/pages

WisseHesNL, 2019-03-05:
Hello everyone, 
I have installed gitea on my Raspberry Pi running DietPi, but in the custom folder (`/mnt/dietpi_userdata/gitea/custom`) is only a `conf` folder and no templates or images. In which folder are those files?

Thank you

MichaIng, 2019-03-05:
@WisseHesNL 
Can you please link some info/docs about which content you are expecting in this dir? I never used Gitea actively so not sure what should be inside and in case why it's not in our install. Aside the install works fine here and daemon running without errors.

WisseHesNL, 2019-03-05:
I used the docs at [docs.gitea.io](//docs.gitea.io) and then [Customizing Gitea](https://docs.gitea.io/en-us/customizing-gitea/)

MichaIng, 2019-03-05:
@WisseHesNL 
Okay it looks like you can/need to simply create those dirs manually:
```
mkdir -p /mnt/dietpi_userdata/gitea/custom/templates/custom/ # For custom links and tabs
mkdir -p /mnt/dietpi_userdata/gitea/custom/public/ # For custom served files/web pages files
```
etc...

After you added your customizations, make everything owned by the `dietpi` run user:
`chown -R dietpi:dietpi /mnt/dietpi_userdata/gitea`

Simply create any of the dirs and files. Looks like those are not created by default on install, which makes sense to keep it clean and slim.

WisseHesNL, 2019-03-05:
Thanks alot @MichaIng !

MichaIng, 2019-03-05:
@WisseHesNL 
Did it work well then?

I mark this issue as closed, but feel free to reuse/reopen if any issue left.

closed 2019-03-05 by MichaIng