MichaIng/DietPi #2622: Dietpi-Software | Dietpi style automatic apt Updates

WilburWalsh, 2019-03-06:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- not a problem, rather an enhancement :-)

#### Describe the solution you'd like:

- I would love some kind of dietpi-update style solution for apt package updates. Considering that you also run G_AGUP and G_AGUG before installing new software, with the added feature of stopping and restarting the Dietpi controlled services, it would be nice if there was an automated solution like dietpi-update. So, either automatically install all updates, or give a notification on log in, if the User desires the Check/Update. Of course it would probably be necessary to warn users, that if enabled, this might interrupt some services during daily/weekly cron run times.

#### Describe alternatives you've considered:

- This was inspired by the suggestion of using unattended-upgrades, but I think Dietpi could even enhance the experience with the points listed above. 

####

MichaIng, 2019-03-07:
@WilburWalsh 
Many thanks for your request.

By chance I just had another look on how we handle the apt-daily timer/services and then another look onto unattended-upgrades.

We disable/mask apt-daily timer/services on DietPi, since while running they block other APT tasks, so dietpi-software/update executions fail randomly. Actually we could re-enable them and simply prompt a waiting-for message, when `/usr/lib/apt/apt.systemd.daily` process is found.

What I do not understand is why the by default implemented version indeed just offers to update the package list, clean/autoclean and download-only packages, but to indeed upgrading the packages the `unattended-upgrades` addition is required, even that this is what most users require.
I mean if you install new packages, you anyway do apt update first. Where is the advantage of having the list updated and packages downloaded but not installed, when you anyway want to do that in case of installs/upgrades manually. Also I don't see any kind of automated mail system or something, so user is informed as fast as new updates are available.

The default implementation actually only makes sense, if you want to use the system as kind of local APT server/cache for a bunch of local clients. So it keeps the package list up-to-date and locally required packages downloaded in the cache and other local clients can check and grab packages there instead of downloading all separately from www and eating up bandwidth.
___________________
So yeah I think we could implement our own simpler solution. Easiest of course is to add `G_AGUP && G_AGUG` into the existing daily cron job to run based on dietpi.txt setting, but perhaps it makes sense to implement it into `dietpi-update`?

So if you run `dietpi-update` and no update was found, you are offered to at least upgrade APT packages.

`dietpi.txt` would then allow to choose between four options:
- 0: Do nothing
- 1: Check for updates only
  - Shall this check for APT package upgrades as well and add a list to dietpi-banner (login prompt) similar to what is not shown if a DietPi update is available?
- 2: Check for updates and auto-upgrade APT packages
- 3: Auto-update DietPi

WilburWalsh, 2019-03-07:
Hi @MichaIng ,
thanks for your quick reply.
I think the idea with implementing it in dietpi-update is great. All the update options in one place 👍 

> `dietpi.txt` would then allow to choose between four options:
>     * 0: Do nothing
>     * 1: Check for updates only
>     * 2: Check for updates and auto-upgrade APT packages
>     * 3: Auto-update DietPi

Seems to be a pretty neat solution.

>     * 1: Check for updates only
>       * Shall this check for APT package upgrades as well and add a list to dietpi-banner (login prompt) similar to what is not shown if a DietPi update is available?

I would say yes. Not sure if a list is too much information, as sometimes a whole bunch of packages is updated and might be confusing. I guess a simple messages that updates are available (maybe number of upgradable packages) with an advice to check/install the upgrades would be sufficient.

MichaIng, 2019-03-07:
@WilburWalsh 
> I guess a simple messages that updates are available (maybe number of upgradable packages) with an advice to check/install the upgrades would be sufficient.

I think you are right. Simply leave the info to run `apt list --upgradeable` to review updates and/or `apt upgrade`.

notDavid, 2019-11-27:
Just wondering... please consider if it would be a good idea to add a 5th option to only install APT security-upgrades (like unattended-upgrades can do with `label=Debian-Security`), for minimal impact/breakage.

MichaIng, 2019-11-27:
@notDavid 
Yeah, probably, but actually I am not big fans of this. Which is security-relevant _marked_ and which not, relies on the package maintainer and not on the facts. And which of these marks reach Raspbian repo, is again another question, 3rd party repos will never of course. Stable Debian versions never do major upgrades, hence the breakage risk is already minimal, and every minor version update has to be considered as either stability, security or performance-relevant without major feature changes and by guarantee backwards compatible, hence I would always apply all of them.

Side note:
apt full-upgrade / apt-get dist-upgrade is a slightly other case, since upgrades are applied as well if dependencies changes, which can lead to additional packages being installed, other packages being marked ready for auto-removal or even conflicting packages being purged. This is the reason why never do this automated on dietpi-update/dietpi-software calls.

closed 2020-12-20 by MichaIng

ravenclaw900, 2021-11-09:
Hi,
There is no way to automatically update DietPi. See https://github.com/MichaIng/DietPi/pull/3899#issuecomment-725630693 for context.