MichaIng/DietPi #2623: Image | Odroid N2

MichaIng, 2019-03-08:
### Creating an image request

#### Formal device information
- Device name | Odroid N2
- Official product URL | https://www.hardkernel.com/blog-2/odroid-n2/
- Image download URL | https://forum.odroid.com/viewtopic.php?f=179&t=33865

#### Is the SBC officially supported by the Debian installer?
- Nope

#### If not, is a reliable 3rd party Debian image available for this SBC?
- Meveric is working on it, currently no graphics driver yet: https://forum.odroid.com/viewtopic.php?f=179&t=33865

Millichrome, 2019-04-05:
Odroid N2 support would be awesome! I've been reading up on it and it seems to be the best option if you want an SBC with USB3 support.

richardhughes260, 2019-04-05:
I pre ordered it!

rox-wolf, 2019-04-11:
Today I received the N2, if any testing will be required, I can help.

Fourdee, 2019-04-11:
Fastest ARM based SBC on the market!
```
DietPi-Benchmark | https://dietpi.com/survey#benchmark :               │   
   │  - CPU Performance   : Duration = 6.75 seconds (lower is faster)       │   
   │  - CPU Temp          : Idle = 26'c | Full load = 32'c                  │   
   │  - RootFS            : Write = 39 MB/s | Read = 66 MB/s                │   
   │  - RAM               : Write = 164 MB/s | Read = 262 MB/s  
```

@rox-wolf 

Thanks :+1: Aiming to have test image available 2-5 days.

Fourdee, 2019-04-12:
@rox-wolf 

Beta image ready for testing:
- https://dietpi.com/downloads/testing/DietPi_OdroidN2-ARMv8.7z

Notes:
- No GPU drivers/support yet (waiting for Meveric to implement)
- I've only run basic testing, not tried audio etc.

rox-wolf, 2019-04-14:
On it, will install my typical things and poke around. Maybe you have some things to test that I should try too?

rox-wolf, 2019-04-14:
@Fourdee 
* Connected via HDMI to my 1080p TV
* DietPi recognized TV's resolution correctly. Odroid's Ubuntu and even Andoid are failing to recognize it and displays distorted video so I need to edit boot.ini file on those distros. DietPi on the other hand works without any issues :) 
* Launched DietPi Benchmark
* Set up sound from dietpi-config, DietPi installed alsa components. Default 0,0 device is HDMI, sound is working. Then I've set up headphones output 0,1, sound is working too.
* Successfully changed hostname.
* Installed OpenSSH - OK
* Installed Docker from Software Optimized - docker info shows these warnings:

```
WARNING: No swap limit support
WARNING: No cpu cfs quota support
WARNING: No cpu cfs period support
```

* On Docker installed Portainer, Grafana, MariaDB - OK
* Installed InfluxDB from Software Optimized - OK
* Tried to install Wireguard from Software Optimized - There's no Wireguard in VPN section
* Installed Pihole from Software Optimized - OK
* Installed Samba from Software Optimized - OK
* Installed rTorrent from Software Optimized - Almost OK
Global password which I've set didn't work, had to login with root/dietpi. Also ruTorrent web interface threw these warnings and an error regarding CfScrape. Maybe it's good to install all the media dependencies (mediainfo, sox, ffmpeg) when installing the full package. Or disabling ruTorrent plugins which are using those media programs.

[14.04.2019 20:54:18] mediainfo: Plugin will not work. rTorrent user can't access external program (mediainfo).
[14.04.2019 20:54:18] spectrogram: Plugin will not work. rTorrent user can't access external program (sox).
[14.04.2019 20:54:18] screenshots: Plugin will not work. rTorrent user can't access external program (ffmpeg).
[14.04.2019 20:54:19] _cloudflare plugin: CfScrape module can't be loaded in Python

* Installed Pydio from Software Optimized - Almost OK
I've followed Pydio installation guide from https://dietpi.com/phpbb/viewtopic.php?p=1064#p1064  and the guide states that MySQL username is pydio and password is dietpi. But this information is wrong and I was getting access denied for pydio@localhost. Correct password is your global password, which is set during DietPi's install.

* Installed Owncloud from Software Optimized - OK
In the guide https://dietpi.com/phpbb/viewtopic.php?f=8&t=5#p47 admin password is written as dietpi, but actually it's a global password.

* Installed Nextcloud from Software Optimized - OK
Just the global password is not mentioned in the guide on the forum.

* Installed Chromium from Software Optimized - OK
I have set Chromium as autostart in dietpi-config and set SOFTWARE_CHROMIUM_RES to 1920x1080 in the /DietPi/dietpi.txt Chromium is starting on boot and in fullscreen. Tested Youtube, sound is working. Video is very laggy though, but if video acceleration is still not done, that's ok (Youtube works well on Odroid's Ubuntu).

* Installed LXDE from Software Optimized - OK
Firefox can play Youtube videos much better, especially with h264ify plugin. During installation I'm getting few dialogs with OK button, which are not really needed as you can't select much there. For example it says [INFO] Updating file: /etc/xdg/autostart/xcompmgr.desktop and OK button.

Fourdee, 2019-04-15:
@rox-wolf 

Many thanks for the in-depth testing, really appreciate it :+1: 

> WARNING: No swap limit support
WARNING: No cpu cfs quota support
WARNING: No cpu cfs period support

This appears to be missing CFS kernel features in current kernel. I'll talk to Meveric and see if we can get them enabled.

> Correct password is your global password, which is set during DietPi's install.

Ah yes, all our docs use the default password of ```dietpi```. Will see if we can mention that in docs.

>  Quite powerful board, fun to see tiny resource usage and free RAM on 4GB version :)

Hehe, yep, more room to run the stuff you need :+1: 

Considering this board is cheaper $79.99 and much faster than the RK3399. Really impressed with it so far.
Heatsink design is simply excellent. Odroid really learnt a lesson and innovated from the XU4 2 seconds at full load to reach 95'c. Very few companies will focus on this area. Very impressed.
My only gripe, it lacks the SATA ports of the N1.

Jlkam, 2019-04-25:
I got my N2 now :D (4gb using emmc card), but the setup will fail when checking connection to 
https://deb.debian.org/debian.
If i change the APT mirror to http://ftp.debian.org/debian it will go on.

I have test some sites with the DietPi-Config network test tool and it seem that https doesn't work at this point. 

(more tests later)

MichaIng, 2019-04-25:
@Jlkam 
That is strange. `deb.debian.org` is definitely reachable via HTTPS. apt-transport-https is installed? `G_AGI apt-transport-https`

Possibly the timeout of our check is simply to short. We have another report where the DNS name resolving takes unexpected long, although this has nothing to do with HTTPS (different protocol) but most properly that one of the three redirects that `deb.debian.org` contains takes too long to resolve. Note that `deb.debian.org` is a CDN (mirror director) that automatically redirects you to the closest mirror, so is a good default for images (and the default for fresh Debian images as well). `ftp.debian.org` is a fixed server (mirror).

At least this is how it should be. Strange is that no matter if I check the redirections from my home server  (northern Germany) or from UK server, it always redirects me finally to `http://ftp.debian.org/debian/`
You can check via: `curl -LI https://deb.debian.org/debian`
But perhaps it is also only done correctly when connecting via APT which then possibly contains some special request content that is handled differently. Not sure.

For reference:
- CDN: https://deb.debian.org/
- General: https://wiki.debian.org/DebianRepository#Using_a_repository
- And finally to find a fixed mirror nearby your location: https://www.debian.org/mirror/list

Jlkam, 2019-04-25:
I have cancel the update process because too slow and it tries to reinstall the same packages over and over again.

I tried the original emmc linux (ubuntu mate) and it doesn't have any issues to reach a https or even the https://deb.debian.org/debian/ address. 
Tested with the same IP settings as before, and curl -LI https://deb.debian.org/debian and firefox.
(curl running into a time out under DietPi)

Switching back to the N2 DietPi image and it told me:
Log file contents:
Spider mode enabled. Check if remote file exists.
--2019-04-12 15:25:39--  https://deb.debian.org/debian/
Resolving deb.debian.org  (deb.debian.org)... 130.89.148.14, 149.20.4.15, 128.31.0.62
...
Connecting to deb.debian.org (deb.debian.org)|130.89.148.14|:443... connected.
Unable to establish SSL connection.

Note the wrong time stamp.

Sorry but currently i can't bring this to life.

MichaIng, 2019-04-26:
@Jlkam 
> Note the wrong time stamp.

Ah, so it is indeed to wrong time stamp that causes SSL connection issues. Please try: `/DietPi/dietpi/func/run_ntpd`

Wixely, 2019-04-26:
Tested using the image provided. I couldn't get Kodi to run, I'm guessing because no GPU. I was able to get Chromium to autorun on startup but it didnt scale to the full screen, likely GPU again since it's X11.

Jlkam, 2019-04-26:
@MichaIng
Now the time is right but the error stays the same. :(

Jlkam, 2019-04-28:
@MichaIng
If i ping some addresses internet or even local ones then most of the packets are lost, changing the cable brings no improvement.

After i go back to the original ubuntu mate image with the same cable and configurations all ping or curl issues are gone.

So i think the dietpi network driver is the issue.

Jlkam, 2019-04-29:
I found the solution just [update u-boot](https://forum.odroid.com/viewtopic.php?f=179&p=254002#p254025).

MichaIng, 2019-04-29:
@Jlkam 
Whatever the u-boot version has to do with network capability 🤔. However I am glad it works now.
Is there as well an update available in the firmware APT repo? `apt dist-upgrade` does it offer an update as well?

@Fourdee 
Perhaps a final image should contain the updated u-boot then? Not sure though if this would be included automatically via APT update on DietPi-PREP.

Jlkam, 2019-04-29:
> @MichaIng
U-boot also includes a base network driver. It seem that there is an update in the repo.
But a fixed u-boot has to be include in a new image because you can't do apt dist-upgrade with a broken network .

MichaIng, 2019-04-29:
@Jlkam 
But the u-boot network driver is only for u-boot internal use, isn't it? I though as fast as systemd stage is reached the system network driver is in use.

However jep new image makes sense, at best after some more testing/fine tuning has been done. Generally since the device is pretty new, including Meverics image we use + his kernel/firmware repository, some childhood sicknesses are expected 😉.

For reference to track possible kernel/firmware related issues: https://forum.odroid.com/viewtopic.php?f=179&t=33865

Jlkam, 2019-04-29:
@MichaIng
Yes your right but there must something that cause these behavior.
That is the forum thread were i found the solution please read the last posts.

carlosedp, 2019-04-30:
Following the issue. I'm getting a N2 as well and hope to contribute.

adamhf, 2019-05-01:
Likewise, received mine yesterday.  Let me know if theres anything you would like testing.

rox-wolf, 2019-05-01:
@Wixely 
> I was able to get Chromium to autorun on startup but it didnt scale to the full screen, likely GPU again since it's X11.

To make Chromium scale, edit `/DietPi/dietpi.txt`  and set `SOFTWARE_CHROMIUM_RES` parameter to 1920x1080 or other resolution.

rox-wolf, 2019-05-02:
Today I've did a fresh install and the MOTD is messed up:

[image]

MichaIng, 2019-05-02:
Ah sorry the N2 image was already created with v6.23 code and I made a change to the MOTD system that it is a script that allows targeted messages based on system/device/version or such. I suggest you update it to the current beta:
`G_DEV_BRANCH beta`

adamhf, 2019-05-02:
Thats fixed the MOTD issue for me

carlosedp, 2019-05-03:
Got the board. I already built the newer Kernel for it (4.9.170) with additional cgroups like reported by @rox-wolf (from Docker).

I can upload it and the modules if needed. Will write a review soon.

adamhf, 2019-05-03:
Have you manage to get a woking Kernel with CONFIG_CFS_BANDWIDTH enabled? I've not managed to get a booting kernel with it enabled, it just boot loops.

carlosedp, 2019-05-03:
@adamhf yes:
```
carlosedp@DietPi:~ $ zcat /proc/config.gz |grep CONFIG_CFS_BANDWIDTH
CONFIG_CFS_BANDWIDTH=y
carlosedp@DietPi:~ $
carlosedp@DietPi:~ $ uname -a
Linux DietPi 4.9.170+ #1 SMP PREEMPT Thu May 2 16:23:59 -03 2019 aarch64 GNU/Linux
```
Grab it here:

https://gist.github.com/carlosedp/704c1cb05a4d19ec044e684a5895edca

adamhf, 2019-05-03:
Thank you, I'll give it a go over the weekend. I already raised https://github.com/hardkernel/linux/pull/374 for some other modules.  You may want to make a similar request.

carlosedp, 2019-05-04:
Nice! The one I built already have those too. Let me know if your tests worked!

rox-wolf, 2019-05-10:
I also had to do uboot update as my N2 stopped getting IP address on to the network when I got new router and static IP also didn't want to work. Downloaded the uboot image from https://github.com/hardkernel/u-boot/releases and ran the included script: `./sd_fusing.sh /dev/sdb` where sdb was my sd card name on Linux.

ODINITE, 2019-05-13:
I just received my ODROID N2 4GB in the mail with a 128GB eMMC.

I'd like to help with testing, but am not sure how to help. Any pointers would be helpful.

Fourdee, 2019-05-14:
@ODINITE 

Hi, thanks for chipping in. Most general testing has been already completed in this ticket. We are waiting for GPU drivers before we can test Kodi etc, until then, simply use the device as a server and report any issues you experience (if any, yet to find any issues on mine).

rox-wolf, 2019-05-14:
@Fourdee Is there any info regarding the GPU drivers? If I understand correctly, they are meant for kernel v5 and are not working well with currently used v4 kernel by DietPi.

ODINITE, 2019-05-14:
@Fourdee 
Thanks for the input!

Where is the most up-to-date image for the N2? Is it still this one:

https://dietpi.com/downloads/testing/DietPi_OdroidN2-ARMv8.7z

Thank you!

rox-wolf, 2019-05-15:
@ODINITE I'm using the same image, but I've set the branch to beta. Simply edit `dietpi.txt` file on your boot partition and set this variable from `DEV_GITBRANCH=master` to `DEV_GITBRANCH=beta`

benjaminpreiss, 2019-06-03:
I'm also using the same image as @ODINITE on my N2. I am trying to set up a nextcloud + mail + bitwarden server with a 500Gb SSD connected via USB.

When trying to select my SSD as my user data directory it successfully moves my data but afterwards the path that i can see is still the SD-card. should it be like that?

MichaIng, 2019-06-03:
@benjaminpreiss 
> afterwards the path that i can see is still the SD-card

Where (in which program/menu) do you mean? Note that **/mnt/dietpi_userdata** will always stay. After moving the userdata, it is a symlink that points to the actual location. But in DietPi scripts/menus the real location should be shown, so if this is not the case, I will check and fix.

You can assure the correctly used location: `ls -l /mnt/dietpi_userdata`

benjaminpreiss, 2019-06-06:
sorry for the delay... after restarting the device it was all displayed correctly. i meant the initial configuration menu that one gets when installing dietpi the first time.

everything is working just fine now.

MichaIng, 2019-06-25:
@Fourdee 
GPU support seems to be there: https://forum.odroid.com/viewtopic.php?f=179&t=33865
Lets implement based on setup-odroid method: http://fuzon.co.uk/meveric/pool/main/s/setup-odroid/
- Will check this.

closed 2019-07-03 by MichaIng