MichaIng/DietPi #2624: fd

msongz, 2019-03-09:
### Creating a software request

#### Formal software information
- Software title | fd
- Short description | A simple, fast and user-friendly alternative to 'find'
- Official URL (if available) | 
- GitHub URL (if available) | https://github.com/sharkdp/fd
- (Official) install documentation | 
https://github.com/sharkdp/fd#from-binaries
#### Are there similar/alternative software titles available with DietPi-Software?

- ...

#### What makes your requested software better than the above solutions, if available?

- good performance, simple to use

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

- ...

#### Can you provide the installation steps that you would suggest DietPi-Software to do?

1. ...

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.

- ...

MichaIng, 2019-03-09:
@msongz 
Many thanks for your request.

Wow that is actually cool, faster and simpler syntax.
- Available on Debian Buster repo: https://packages.debian.org/de/buster/fd-find
- Pre-compiled binaries available as well: https://github.com/sharkdp/fd/releases
  - Only one ARM binary available, requires check on ARMv6/7/8.

However this does not add any new features and most users are simply used to use `find`. Since `find` is an "essential" package there is not chance in replacing it, so there is one more package/binary installed. But yeah why not promoting and testing this a bid by making it available as optional install. At best we increase the chance that earlier or later it replaces `find` as core/essential package in Distros.

msongz, 2019-03-10:
it will be an "essential" package in ubuntu 19.04, seems much potential.

MichaIng, 2019-03-10:
@msongz 
> it will be an "essential" package in ubuntu 19.04, seems much potential.

Really? I saw that it will be available in Ubuntu repo then, but not that it is an "essential" package there, if this means the same then on Debian, that it is expected as system core package that will never be autoremoved and cannot be removed without special APT option or entering "Yes, I know what I do!"?

msongz, 2019-03-10:
https://github.com/sharkdp/fd#installation

`Note: fd will be officially available in Ubuntu Disco Dingo (19.04).`

hope I do not misunderstand this sentence.

MichaIng, 2019-03-10:
@msongz 
Jep that was exactly what I found as well. It just means that you can install it from the main repo, similar to Debian Buster. But it is not pre-installed nor "essential".

msongz, 2019-03-10:
oh~easy to understand from your mouth.😃

MichaIng, 2021-09-24:
We decided to cleanup open issues a bit and close those which will obviously not addressed soon. The FeatHub request stays however open to allow votes and if anyone wants to start working on implementing this software, please feel free to to a start, and we'll help where we can.

Here you find instructions about how to implement a new software title to `dietpi-software`: https://github.com/MichaIng/DietPi/wiki/How-to-add-a-new-software-title

closed 2021-09-24 by MichaIng