MichaIng/DietPi #2627: storage left info

msongz, 2019-03-10:
[image]

show storage left space info in banner using code:

```df -h|grep '/dev/root'|awk '{ print $4 }' ```

when using sd card,especially small one, space left is always important info we need.

MichaIng, 2019-03-10:
@msongz 
Many thanks for your request.

This fits to this idea: https://dietpi.com/phpbb/viewtopic.php?f=15&t=5257

Since the need/taste about the banner is different, I will turn this request into a DietPi-Banner customisation request to allow overriding each banner slot with a custom one, adding or hiding them.

msongz, 2019-03-10:
customize banner would be great!

MichaIng, 2019-03-12:
Just for reference: https://github.com/MichaIng/DietPi/commit/419657182bbf61e96bcaad77606262ceb922286c

Not space left but CPU temps instead 😄. However the code cleanup made it easier to implement custom banner infos.

MichaIng, 2019-03-17:
Another idea/request is to show the external IP/domain to easier keep apart DietPi systems on different locations connected to via SSH: https://dietpi.com/phpbb/viewtopic.php?f=9&t=5708

Fourdee, 2019-03-17:
@MichaIng 

> external IP/domain

Was thinking about this in the past. We could use a simple html page on dietpi.com to display connected external IP and pull that back to users system.

The only downside is bandwidth usage (would need to check server once implimented), and, required delay waiting for this info, before banner displayed.

MajorWolf1, 2019-03-17:
Hey, so the post on the forums was actually mine, I've already implemented it into the ssh banner. If its not connected to the outside world then it just states "N/A". I was using ifconfig.me to retrieve the external IP, however as you have now setup a php script on your own server you can easily change it to use that instead. I could share my 'new' banner if you wish.

MichaIng, 2019-03-17:
@Fourdee @MajorWolf1 
Ah the myip.php is a very good idea, since /etc/hosts /etc/hostname and $(hostname) all "should" match the external domain but "can" be freely chosen.
Although a domain name (if available) would be even better, but I see no chance to derive it without manual user input.

> The only downside is bandwidth usage (would need to check server once implimented), and, required delay waiting for this info, before banner displayed.

I think this info should in case only be retrieved via hourly cron job and e.g. stored to `/DietPi/dietpi/.network`. In case of static IP, it even only needs to be retrieved once, in case of available domain, the domain would be better for all the other cases (software installs) where we need external connection info.

So I think when using this, we should add an option to dietpi-config to allow adding a domain name, if available, otherwise static IP if available, otherwise dynamic IP.
Only in case of dynamic IP and if printing it in the banner is enabled (or statically added by us), a cron job will update this info. Otherwise the domain/static IP entered by the user will be shown.

The first time we require external domain/IP info (e.g. Nextcloud Talk + WireGuard install, or, then Banner when implemented) we prompt to user to choose which one as above is available and store that info to /DietPi/dietpi/.network. Afterwards we never need to ask again. To change it we could add an option to dietpi-config > Security Options. Usually hostname should match external domain, but e.g. if on local network multiple servers are running that listen to different ports, that would not work, so then the external IP/domain of all of them is the same of course, but the local hostname needs to be different 🤔.

MajorWolf1, 2019-03-17:
Using a simple curl command is the way that I'm retrieving it for the banner, (that then goes into an if statement) so i don't see why we couldn't add a cron task to curl it and store it in /DietPi/dietpi/.network on the next line down then whenever it is needed then just sed -* to that file, as you said just run it once an hour or so, this would reduce the amount of requests to the server.
I don't see why we couldn't ask the user if they would like to display hostname OR external IP on the banner, as some people only run this locally. This could probably be done with a simple If statement. As the hostname is already stored in that file.

MichaIng, 2019-03-17:
@MajorWolf1 
`curl` will succeed in any case. The outputs of `hostname` and `cat /etc/hostname` will work on local networks to access the device, especially from the device itself. `/etc/hosts` as well contain the loopback IP and other aliases that will as well always work on the device itself. But as said we don't know if there also allow access externally.

Just a little example:
```
root@VM-Stretch:~# hostname
VM-Stretch

root@VM-Stretch:~# cat /etc/hostname
VM-Stretch

root@VM-Stretch:~# cat /etc/hosts
127.0.0.1 localhost
127.0.1.1 VM-Stretch
::1 localhost ip6-localhost ip6-loopback
ff02::1 ip6-allnodes
ff02::2 ip6-allrouters

root@VM-Stretch:~# cat /var/www/html/index.lighttpd.html
success

root@VM-Stretch:~# curl VM-Stretch
success
root@VM-Stretch:~# curl localhost
success
root@VM-Stretch:~# curl 127.0.0.1
success
root@VM-Stretch:~# curl 127.0.1.1
success
```
- `curl` succeeds on all of them, but non of them are valid to access the VM externally of course, not even from local network.
- The local IP can be easily gathered:
```
root@VM-Stretch:~# ip a s eth0
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 08:00:27:9e:d0:6f brd ff:ff:ff:ff:ff:ff
    inet 192.168.178.29/24 brd 192.168.178.255 scope global eth0
       valid_lft forever preferred_lft forever
```
- But the external IP not without anyone externally that you connect to tells you with which IP you connect.
- And in case you have a domain, there is absolutely no way AFAIK to find this out automated.

MajorWolf1, 2019-03-17:
Ahh I see, im kinda new to this so ngl its a little trial and error. 

We already have the internal IP on the banner, so i just added the external IP. if it cant connect to the internet then it states "N/A". Could have the option to choose if you want the hostname or the external ip depending if its offline or not. And in my case was just to make sure ive connected to the correct pi. ofc this isnt gonna be the case for everyone. was just a though on something you could add at a later date.

I dunno about retrieving a domain name automatically, but I'm in the process of adding the option into the dietpi-config file under security options. so it will append the domain name onto the .network then we could use that to display the domain name. (If this isn't the best place to append it then please to tell me. dont wanna break it XD)

Fourdee, 2019-03-18:
@MichaIng 

Marking a start on this, had some ideas 👍 
[image]

[image]

msongz, 2019-03-18:
@Fourdee looking so great !

MichaIng, 2019-03-18:
@Fourdee 
Indeed looking really great! My idea was to divide the upper banner info into segments like in a table:

DietPi | Time
-- | --
RPi 3 | `<CPU temp>`
`<LAN IP>` | `<WAN IP>`

Rows separated by green lines, columns separated by `|`. And then allow to configure each field as desired.

But however your solution looks way simpler to configure and faster to run and beautiful, so I like it more 😃.

Just one thing. To be fair to contributes, supply- and code-wise, I would not allow to hide credits? But hiding the DietPi commands for experienced users of course totally makes sense.

MajorWolf1, 2019-03-18:
Hey, so ive been working on something similar that you guys may want to impliment. Kinda similar but a little different.
[image]
[image]
[image]
[image]
[image]

Fourdee, 2019-03-19:
@MichaIng 

> To be fair to contributes, supply- and code-wise, I would not allow to hide credits? But hiding the DietPi commands for experienced users of course totally makes sense.

Yep, good call, will make the change 👍 

I also need to:
- dietpi-globals add ```dietpi-banner``` alias
- temp array on G_WHIP_CHECKLIST

> My idea was to divide the upper banner info into segments like in a table:

Yep, had similar thoughts aswell. The only issue is alignment of the contents in the cells. Unless we check for each string length + cut excess and/or new line it (more processing), table wouldn't line up correctly.

Fourdee, 2019-03-19:
@MajorWolf1 

Added host name and domain name to available options, if you can think of any more, please let us know 👍 
[image]

MichaIng, 2019-03-19:
@Fourdee 
What do you think about merging Hostname and Domain name into FQDN? The short form of the hostname is already shown in the command prompt (PS1), so doubles and in the end both belong together for the full hostname.

Fourdee, 2019-03-19:
@MichaIng 

> What do you think about merging Hostname and Domain name into FQDN? The short form of the hostname is already shown in the command prompt (PS1), so doubles and in the end both belong together for the full hostname.

Yep can do, ```"$(hostname) | $(domainname)"```?

Fourdee, 2019-03-19:
o0o0, weather option:
```
curl wttr.in/London?format=4
```

msongz, 2019-03-19:
haha, the year progress. such as https://www.yearprogressbar.com/ https://mobile.twitter.com/year_progress/status/1107673014634725378

MajorWolf1, 2019-03-19:
We could always change the "Dietpi" in the top left of the banner into the hostname. Saves you adding another option that could be made simpler

MichaIng, 2019-03-19:
@MajorWolf1 
DietPi in the top left should stay, since it shows the version of DietPi regardless of the hostname 😉. It is the default that hostname == DietPi on fresh installs, but for a public server (or multiple DietPi servers in local network) this should be changed anyway.

The reason I would show only FQDN is due to the convention that `/etc/hostname` content (and `hostname`) output anyway defaults (or should default) to this. E.g. tested on my VM:
```
root@VM-Stretch:~# cat /etc/hostname
VM-Stretch.my.domain.org

root@VM-Stretch:~# cat /etc/hosts
127.0.0.1 localhost
127.0.1.1 VM-Stretch.my.domain.org VM-Stretch
::1 localhost ip6-localhost ip6-loopback
ff02::1 ip6-allnodes
ff02::2 ip6-allrouters

root@VM-Stretch:~# hostname
VM-Stretch.my.domain.org

root@VM-Stretch:~# hostname -f
VM-Stretch.my.domain.org

root@VM-Stretch:~# hostname -d
my.domain.org

root@VM-Stretch:~# hostname -s
VM-Stretch
```
When `/etc/hostname` only contains the short hostname, the command correctly takes the FQDN from `/etc/hosts`:
```
root@VM-Stretch:~# cat /etc/hostname
VM-Stretch

root@VM-Stretch:~# hostname
VM-Stretch

root@VM-Stretch:~# hostname -d
micha.gnoedi.org

root@VM-Stretch:~# hostname -f
VM-Stretch.micha.gnoedi.org

root@VM-Stretch:~# hostname -s
VM-Stretch
```

So from my point of view just show:
`Hostname | $(hostname -f)`
or to be more precise but less known wording:
`FQDN | $(hostname -f)`

MichaIng, 2019-03-20:
@Fourdee @MajorWolf1 
What do you think about a "custom" entry?

When selecting/enabling the "Custom" an input menu would allow you to add a custom bash command. This is then saved to a file and simply sourced when the banner is shown, if option enabled and file exists. It would be shown as last part of the upper banner. Only question is how to deal with the description.
We could allow two inputs: The description first, shown on the left side as of the other lines, and the command that provides the value separately to be shown on the right side.

MajorWolf1, 2019-03-20:
Hmmm, not a bad idea actually @MichaIng would be useful for people that know the script they wish to impliment but dont understand how the config files etc work. (Took me a little time for it to click). Would be useful for a lot of people tbf to make their banner as suitable and personal to them.

MichaIng, 2019-03-20:
@MajorWolf1 
Jep, and besides making it easier it also allows the custom command to be persistent over DietPi updates. In your case the update to v6.22 will overwrite your customizations and you would need to re-add them to the new script that looks now quite different. A hassle.
Also now that we support all your ideas natively, I guess you don't need to do custom changes anymore 😉.

MajorWolf1, 2019-03-20:
Would make it better for everyone tbf, meaning it a little more personal. Simplicity will mean more people can have a little more personal version. Persistence is probably the main thing people would be bothered by as they don't have to make a restore backups. Also thanks for implementing my idea. First thing I've actually contributed to and I'm glad it was with you guys.

MichaIng, 2019-03-20:
@MajorWolf1 
Custom entry added: https://github.com/MichaIng/DietPi/commit/2225fb6eee26177725dbd766424c24ce68d22cda

You should be able to test it, if you want, by overwriting yours. It does not depend on any other updated code, AFAIK: https://raw.githubusercontent.com/MichaIng/DietPi/dev/dietpi/func/dietpi-banner

@Fourdee 
Like it? 😃

I will also do some final tests and merge. Found that the 2 seconds curl timeout seems to be too short in some cases. WAN address sometimes does not show up. Sadly we can't use this on perhaps faster GitHub server.

closed 2019-03-23 by MichaIng

Hritik14, 2023-07-21:
I couldn't find (quickly) an obvious command to access the edit banner menu, so here it is for posterity: `dietpi-banner`

MichaIng, 2023-07-22:
Actually it was intended to exit directly with a meaningful error message when not executed as root (and when opening it in menu mode). Die do a bug this was not the case. Now it is 🙂: https://github.com/MichaIng/DietPi/commit/fd552c4

andrew-manger, 2023-09-13:
Is there a way we can install dietpi-banner onto a different OS flavor? I have a RasPi running DietPi as a WOL server and I LOVE the banner that pops on on login. Want to add it to my other Pis but I can't install DietPi on them.

MichaIng, 2023-09-30:
At least it uses a bunch of functions from `dietpi-globals`: https://github.com/MichaIng/DietPi/blob/master/dietpi/func/dietpi-globals

So if you copy that one as well to the expected location, at last on Debian and Ubuntu, it should work. But it is indeed not designed as a standalone script in the first place and does some DietPi-specific things like informing about available DietPi updates and such.

cybersholt, 2024-01-11:
@MichaIng very cool and useful script, though you might want to add the ability for the user to enter a location for wttr.in, their geolocation doesn't work the best for me.

MichaIng, 2024-01-12:
I just checked whether/how this is possible, and it is: https://github.com/chubin/wttr.in
There are even ways to check the whether at locations of interest. Quite nice, and more than trivial syntax. So we could just add an input box to type in any location reference, typical city name. Could you open a new issue for this request?