MichaIng/DietPi #2628: Broken link for weavedconnectd_1.3-07z1_armhf.deb

dietpi-user, 2019-03-10:
#### Details:
- Date           | Sun 10 Mar 16:33:33 CET 2019
- Bug report     | N/A
- DietPi version | v6.21.1 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B+ (armv7l) (index=3)
- Kernel version | #1200 SMP Tue Feb 12 20:27:48 GMT 2019
- Distro         | stretch (index=4)
- Command        | Connection test: https://github.com/weaved/installer/raw/master/Raspbian%20deb/1.3-07/weavedconnectd_1.3-07z1_armhf.deb
- Exit code      | 8
- Software title | DietPi-Software

#### Steps to reproduce:
Add the following line to dietpi.txt:

AUTO_SETUP_INSTALL_SOFTWARE_ID=68     # Remot3.it - (Weaved) access your device over the internet

#### Expected behaviour:
Installation of Remote3.it upon first boot.

#### Actual behaviour:
Endless installation loop due to broken link.

#### Extra details:
- ...

#### Additional logs:
```
Log file contents:
Spider mode enabled. Check if remote file exists.
--2019-03-10 16:33:30--  https://github.com/weaved/installer/raw/master/Raspbian%20deb/1.3-07/weavedconnectd_1.3-07z1_armhf.deb
Resolving github.com (github.com)... 192.30.253.112, 192.30.253.113
Connecting to github.com (github.com)|192.30.253.112|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
Remote file does not exist -- broken link!!!
```

MichaIng, 2019-03-10:
@dietpi-user 
Many thanks for your report.

Jep this is a known issue due to the old weaved repo has been dropped from GitHub. Fix for v6.22 is already done: https://github.com/MichaIng/DietPi/issues/2551

To install it now:
```
cd /tmp
wget https://github.com/remoteit/installer/releases/download/v2.2.5/connectd_2.2.5_armhf.deb
dpkg -i connectd_2.2.5_armhf.deb
rm connectd_2.2.5_armhf.deb
connectd_installer
```

dietpi-user, 2019-03-10:
Ok thanks, I didn't see that fix, many thanks for the quick reply!

closed 2019-03-10 by dietpi-user