MichaIng/DietPi #2630: NanoPi M3 | Image fails to boot?

MichaIng, 2019-03-10:
Ref: https://dietpi.com/phpbb/viewtopic.php?f=11&t=5600

MichaIng, 2020-01-01:
I mark this as closed. New Buster image will be done the next days.

closed 2020-01-01 by MichaIng

rozcietrzewiacz, 2024-08-30:
/reopen

Downloaded & flashed latest image - M3 doesn't boot. I'm no U-Boot expert, but it seems like it can't locate the sdcard device to load stuff from.
```
U-Boot 2016.01-armbian (Feb 03 2021 - 23:34:45 +0100)

DRAM:  1 GiB
MMC:   NEXELL DWMMC: 0, NEXELL DWMMC: 1
loaded from SD, getting env from MMC 1
*** Warning - bad CRC, using default environment

MIPI: display.0
HDMI: display.0, preset 0 (1280 *  720)
HDMI: phy ready...
LCD: [HDMI] dp.0.1 1220x680 16bpp FB:0x46000000
In:    serial
Out:   serial
Err:   serial
Hit any key to stop autoboot:  0
** File not found /boot.scr **
## Executing script at 40000000
Wrong image format for "source" command
1379 bytes read in 42 ms (31.3 KiB/s)
## Executing script at 40000000
233 bytes read in 54 ms (3.9 KiB/s)
** File not found /boot/dtb/nexell/ **
8672161 bytes read in 654 ms (12.6 MiB/s)
12853256 bytes read in 928 ms (13.2 MiB/s)
## Loading init Ramdisk from Legacy Image at 49000000 ...
   Image Name:   uInitrd
   Image Type:   AArch64 Linux RAMDisk Image (gzip compressed)
   Data Size:    8672097 Bytes = 8.3 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
ERROR: Did not find a cmdline Flattened Device Tree
Could not find a valid device tree
Card did not respond to voltage select!
** Bad device mmc 0 **
## Executing script at 40000000
Wrong image format for "source" command
** Bad device mmc 0 **
## Executing script at 40000000
Wrong image format for "source" command
nanoPi#
```

MichaIng, 2024-08-30:
I think it cannot find the device tree in particular. Let me check the image.

MichaIng, 2024-08-30:
Can you check in U-Boot console whether this variable is defined?
```sh
echo ${fdtfile}
```

rozcietrzewiacz, 2024-08-30:
Yeah, `echo ${fdtfile}` was returning nothing. And with `setenv fdtfile "s5p6818-nanopi-m3.dtb"` - it just booted now! 🎉 

One more thing to note: Before I saw your message I was in the middle of experimentation and - based on the observation mentioned earlier, I have also added  `setenv devnum 1` in `boot.cmd` (I did not recompile though). I'm not sure if this had any effect - I'll revert that and try

rozcietrzewiacz, 2024-08-30:
`devnum change` didn't matter :)

Thanks @MichaIng !

MichaIng, 2024-08-30:
Okay great. Looks like the last bootloader update broke it. I'll add `fdtfile` to our images now.

MichaIng, 2024-08-30:
Interesting, my NanoPi Fire3 boots fine without defining the device tree. Looks like the two bootloaders differ more than expected. However, I could not reliably detect whether it really was the Fire3 device tree, or the M3 one, both define a Broadcom WiFi device (which the Fire3 does not have).

I'll define the device tree in `dietpiEnv.txt` for both models/images now, does not hurt.

MichaIng, 2024-08-30:
Fixed with: https://github.com/MichaIng/DietPi/commit/a398cfc
New images are building: https://github.com/MichaIng/DietPi/actions/runs/10639011444, https://github.com/MichaIng/DietPi/actions/runs/10639012814

Another thing I recognised, is that the setup partition was not detected, hence `dietpi.txt` and `dietpiEnv.txt` were not copied from the FAT partition, and the root filesystem was not expended, as the setup FAT partition was still in place. Checking for the reason, `lsblk` seemed to have no returned the 2nd partition at all, but the root partition as last in the list. We had another such report, hence I added some debug output, also using `lsblk` to print what it sees, and there, the setup partition was there. So we seem to talk about fractions of a second where during early boot, when our script aims to import modified files from the setup partition, remove it, and expand the root partition + filesystem, the 2nd/setup partition is not visible yet. Weird is that the root filesystem is remounted R/W already at this stage (usually it is mounted R/O at first), so no idea how the kernel would not know about all partitions yet. Maybe `systemd-remount-fs.service` triggers some rechecking of the partitions, which clears the info and has a race condition with our script.

Not sure how to solve best. I am no fans or some arbitrary delay. But maybe we can run another `partprobe` and sleep for 0.5 seconds, like we do in other scripts after `partprobe` as minimal failsafe time for the kernel to really know/provide all partitions.

@rozcietrzewiacz 
Was the root filesystem expanded correct on your NanoPi M3?

rozcietrzewiacz, 2024-09-01:
```
# lsblk -f /dev/sda
NAME   FSTYPE FSVER LABEL       UUID                                 FSAVAIL FSUSE% MOUNTPOINTS
sda
├─sda1 ext4                     3998df48-4bda-490b-ab5e-635f4adf2572   11.5M    91% /mnt/sda1
└─sda2 vfat         DIETPISETUP D07C-922B
```
Does not look like it

rozcietrzewiacz, 2024-09-01:
If I tried editing `/boot/boot.cmd` to add `setenv fdtfile "s5p6818-nanopi-m3.dtb"`, but that doesn't get picked up - I guess this needs to be baked into the microsd disk on a lower level..? I did this stuff long time ago, I may be mixing up something  😅

rozcietrzewiacz, 2024-09-01:
Since there is still some concern about the expanding of partition, I didn't try the updated image yet (I don't want to wear the card too much). I'd rather try flashing just the updated U-boot, so I'm trying to do that from the (manually) booted M3. I can see that `dietpi-config` includes the option to update the bootloader, but I'm not sure if this would automatically use the `boot.cmd` content that I edited on `sda2`. The dialog message says just 
```
This will flash the U-Boot image of the installed linux-u-boot-* package to the MMC device /dev/mmcblk2.
```

rozcietrzewiacz, 2024-09-01:
@MichaIng - OK, I think I figured it out:
Booted on the M3, I did these based on the commands from build pipeline you provided:
```
# #(edited /boot/boot.cmd first)

root@DietPi:~# mkimage -C none -A arm64 -T script -d /boot/boot.cmd /boot/boot.scr
#(checked that the binary content of boot.scr contains the ftdfile command)

root@DietPi:~# /boot/dietpi/func/dietpi-set_hardware flash-u-boot-mmc /dev/mmcblk2

 DietPi-Set_hardware
─────────────────────────────────────────────────────
 Mode: flash-u-boot-mmc (/dev/mmcblk2)

764+1 records in
764+1 records out
[  OK  ] flash-u-boot-mmc /dev/mmcblk2 | Completed
root@DietPi:~#
root@DietPi:~# reboot
```

And it booted successfully 😃  🎉

MichaIng, 2024-09-01:
@rozcietrzewiacz 
Better/easier use `dietpiEnv.txt` to set `fdtfile` variable 🙂 . Otherwise, you do not need to reflash U-Boot, but that `mkimage` command is necessary. U-Boot reads `boot.scr` only, which has some sort of binary hash at its top, to prevent against executing corrupted scripts. So `boot.cmd` is only the source, but has not effect on its own.

Okay so the setup partition was not detected as well in your case. Can you show the related log?
```sh
cat /var/tmp/dietpi/logs/fs_partition_resize.log
```
You can manually run the script to get the partition removed and the root partition + filesystem expended:
```sh
/var/lib/dietpi/services/fs_partition_resize.sh
```

rozcietrzewiacz, 2024-09-01:
I think the first-boot script just didn't run - or I accidentally interrupted it while experimenting.
Anyway, here's the log:

```
root@DietPi:~# cat /var/tmp/dietpi/logs/fs_partition_resize.log
Removed /etc/systemd/system/local-fs.target.wants/dietpi-fs_partition_resize.service.
[ INFO ] Detected root drive /dev/mmcblk2 with root partition 1
[ INFO ] No DietPi setup partition found, last partition is: "ext4 "
NAME             LABEL        SIZE TYPE FSTYPE MOUNTPOINT
/dev/mmcblk2                  3.7G disk
├─/dev/mmcblk2p1              800M part ext4   /
└─/dev/mmcblk2p2 DIETPISETUP    1M part vfat
Disk /dev/mmcblk2: 3.71 GiB, 3980394496 bytes, 7774208 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x552ccc82

Old situation:

Device         Boot   Start     End Sectors  Size Id Type
/dev/mmcblk2p1 *       8192 1646591 1638400  800M 83 Linux
/dev/mmcblk2p2      1646592 1648639    2048    1M  c W95 FAT32 (LBA)

/dev/mmcblk2p1:
New situation:
Disklabel type: dos
Disk identifier: 0x552ccc82

Device         Boot   Start     End Sectors  Size Id Type
/dev/mmcblk2p1 *       8192 1646591 1638400  800M 83 Linux
/dev/mmcblk2p2      1646592 1648639    2048    1M  c W95 FAT32 (LBA)

The partition table has been altered.
resize2fs 1.46.2 (28-Feb-2021)
The filesystem is already 204800 (4k) blocks long.  Nothing to do!

```

It seems like in case of M3:
```
root@DietPi:~# lsblk -nrbo FSTYPE,LABEL /dev/mmcblk2

vfat DIETPISETUP
ext4
```

Which is weird, because both `fdisk -l` as well as `lsblk -nbo` present the partitions in correct order. I've no idea why `-r` messes up the order. But without it, the distance between `vfat` and `DIETPISETUP` increases:
```
root@DietPi:~# lsblk -nbo FSTYPE,LABEL /dev/mmcblk2

ext4
vfat   DIETPISETUP
```

So for the comparison on line `48` of `/var/lib/dietpi/services/fs_partition_resize.sh`, you can use either 
```
if [[ $LAST_PART == 'vfat   DIETPISETUP' ]]
```

or read the individual words first, e.g.:
```
read -r _fstype _label < <(lsblk -nbo FSTYPE,LABEL /dev/mmcblk2 | tail -1)
LAST_PART="$_fstype $_label"
```
...or just add `| sed 's/  */ /g'` after the `tail` 😁

MichaIng, 2024-10-08:
Great find. I would have never guessed that the "raw" flag changes the order. And it does so only on Bullseye, not from Bookworm on, which is why we did not detect it earlier.

I fixed it here, I think no need to check the filesystem type, as the label should be sufficiently unique: https://github.com/MichaIng/DietPi/commit/017698e