MichaIng/DietPi #2631: DietPi-Software | FFmpeg prerequisite: 7za command missing

dietpi-user, 2019-03-10:
#### Details:
- Date           | Sun 10 Mar 19:00:54 CET 2019
- Bug report     | N/A
- DietPi version | v6.21.1 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B+ (armv7l) (index=3)
- Kernel version | #1200 SMP Tue Feb 12 20:27:48 GMT 2019
- Distro         | stretch (index=4)
- Command        | 7za x -y 7.7z -offmpeg_rpi
- Exit code      | 127
- Software title | DietPi-Software

#### Steps to reproduce:
 Mode: Installing FFmpeg: audio & visual libary
[  OK  ] DietPi-Software | Connection test: https://dietpi.com/downloads/binaries/rpi/ffmpeg_rpi.7z
[  OK  ] DietPi-Software | wget https://dietpi.com/downloads/binaries/rpi/ffmpeg_rpi.7z -O 7.7z
[FAILED] DietPi-Software | 7za x -y 7.7z -offmpeg_rpi

tail -f /tmp/dietpi-firstrun-setup.log

#### Expected behaviour:
Installation of the ffmpeg library, with extraction of 7z file with 7za. However the command 7za is not available.

#### Actual behaviour:
Constant retries by dietpi-software on first login in the background to try and install the ffmpeg library (which fails everytime).

#### Extra details:
I've managed to work around this issue by logging into ssh and running the following command:

sudo apt-get install p7zip-full

(p7zip was installed already, but this does not contain 7za)

Then dietpi-software is able to finish successfully.

Also, I've added the following to dietpi.txt by the way (maybe ffmpeg is required by home assistant?):

AUTO_SETUP_AUTOMATED=1

AUTO_SETUP_INSTALL_SOFTWARE_ID=0      # Open SSH
AUTO_SETUP_INSTALL_SOFTWARE_ID=73     # Fail2Ban - prevents brute-force attacks with ip ban
AUTO_SETUP_INSTALL_SOFTWARE_ID=157    # Home Assistant - open-source home automation platform

#### Additional logs:
```
Log file contents:
/DietPi/dietpi/func/dietpi-globals: line 1373: 7za: command not found
```

--------------------------------------------------------------------

[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

MichaIng, 2019-03-10:
@dietpi-user 
Many thanks for your report.

Very strange, we did remove `7za` just with the 2 minutes ago released v6.22 Beta, but it is a core dependency of DietPi prior to that. (On v6.22 we use `7zr` as lightweight alternative instead).

Please verify this is installed:
`dpkg -l | grep '^p7zip'`

Otherwise install the related package:
`G_AGP p7zip-full`

Or apply the just now released beta, which does not use 7za anymore: https://github.com/MichaIng/DietPi/issues/2632

dietpi-user, 2019-03-10:
Okay thank you Michalng for your quick reply, no I don't think p7zip-full was present before I ran the apt-get install p7zip-full command. But just fyi, I downloaded the following image today, below the contents from the hash.txt file. With a previous image I didn't have these issues by the way (MD5               : aac641f4c233ba0d3c6987478602f960, Modified Time     : 31/01/2019 14:44:55), but then I didn't modify the dietpi.txt file neither.

Filename          : DietPi_RPi-ARMv6-Stretch.img
MD5               : c2ef66686da801c9634f457c89f7be7b
SHA1              : 2f6054d9e844d01f1ae987926e21cecc56862dca
CRC32             : fb6bf06b
SHA-256           : f68ae1507a9e31857ef10a6fdc32dee3a9087e8d1bd7e255e99f71a0dc8b4458
SHA-512           : e74436a34689635abd42a561c4141c5b88be15d79a13d7bdb8cf88e6d12d3cdc9f1c9839f97846ef2ad7026447e824f9b089c47741ea0291321f1c71c21dd181
SHA-384           : d2d9127f47517b4a9a161c40be2e87912ad52a0f2e2ee32f844d393c931e727d690ecb36898bce6a076519cc8dea3434
Full Path         : C:\Users\Micha\VirtualMachines\DietPi_RPi-ARMv6-Stretch.img
Modified Time     : 04.03.2019 20:44:12
Created Time      : 04.03.2019 20:44:12
Entry Modified Time: 04.03.2019 22:18:58
File Size         : 645.004.800
File Version      : 
Product Version   : 
Identical         : 
Extension         : img
File Attributes   : A

MichaIng, 2019-03-10:
@dietpi-user 
Omg now that you mention it, yeah I know how the issue occurred. I used the new DietPi-PREP script that only installs p7zip (but not p7zip-full) according to v6.22 changes, but the DietPi version is still v6.22.

Damn mistake, however easy to fix and v6.22 will be released soon which will resolve this issue with the image as well.

MichaIng, 2019-03-23:
v6.22 was just released, so the image should work now out of the box without above error.

closed 2019-03-23 by MichaIng