MichaIng/DietPi #2632: Open Beta v6.22 | Please help testing and hardening the upcoming release

MichaIng, 2019-03-10:
RC version | v6.20.0
---------- | -------
Changelog | https://github.com/Fourdee/DietPi/blob/beta/CHANGELOG.txt
Code changes | https://github.com/Fourdee/DietPi/compare/master...beta
How to apply | https://github.com/Fourdee/DietPi/blob/beta/BRANCH_SYSTEM.md

When testing, especially have an eye on features you regularly use anyway and that are mentioned in the change log above.

### Known issues
#### DietPi functionality
- DietPi-Config | Enabling WiFi + Ethernet adapters, both on different subnets, breaks WiFi connection in some cases: https://github.com/Fourdee/DietPi/issues/2103
  - Current workaround is to disable the adapter that is not in use (not for internet connection).
  - Otherwise a custom routing table is required.
- DietPi-Drive_Manager | encryptfs is currently not supported and will be removed from fstab, when executing drive manager: https://github.com/Fourdee/DietPi/issues/2078

#### SBC/device related
- RPi | LXAppearance (on LXDE desktop) hangs on dbus-launch: https://github.com/Fourdee/DietPi/issues/1791
- RPi | Enabling OpenGL/VC4 causes sound card selection to be invalid: https://github.com/Fourdee/DietPi/issues/2173
- Pine A64 | Bluetooth add-on module currently not supported: https://github.com/Fourdee/DietPi/issues/528
- Odroid C2 | HotSpot does not work with (some?) WiFi adapters: https://github.com/Fourdee/DietPi/issues/1955

#### Software title related
- Node-RED | Pre-installed modules cannot be updated via webUI: https://github.com/Fourdee/DietPi/issues/2073
- Raspimjpeg | With Lighttpd, streaming mjpeg does not work: https://github.com/Fourdee/DietPi/issues/1747

Joulinar, 2019-03-10:
Hi,

I was going to update to 6.22 but was getting below message

```
DietPi-Pre-patch
─────────────────────────────────────────────────────
 Mode: Applying critical pre-patches

[ WARN ] Pre-patch 3 | Storing global DietPi-Software password with enhanced security
[  OK  ] Successfully applied critical pre-patches

[  OK  ] DietPi-Update | APT update, please wait...
Get:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages [11.7 MB]
Hit:3 https://archive.raspberrypi.org/debian stretch InRelease
Get:4 https://cdn-aws.deb.debian.org/debian sid InRelease [242 kB]
Ign:4 https://cdn-aws.deb.debian.org/debian sid InRelease
Get:5 https://cdn-aws.deb.debian.org/debian sid/main armhf Packages [8,058 kB]
Fetched 20.0 MB in 20s (955 kB/s)
**Reading package lists...
W: GPG error: https://cdn-aws.deb.debian.org/debian sid InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 7638D0442B90D010 NO_PUBKEY 04EE7237B7D453EC
W: The repository 'https://deb.debian.org/debian sid InRelease' is not signed.**

[  OK  ] DietPi-Update | G_AGUP
[  OK  ] DietPi-Update | APT upgrade, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

[  OK  ] DietPi-Update | G_AGUG
```

And next one
```
[  OK  ] DietPi-Update | Unpack update archive
**rm: cannot remove 'DietPi-beta/dietpi/.??*': No such file or directory**
[  OK  ] DietPi-Update | Copy DietPi core files to RAMdisk
```

and one more
```
 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Completed

[ INFO ] DietPi-Update | Current version : v6.22.0
[ INFO ] DietPi-Update | Latest version  : v6.22.0
**/DietPi/dietpi/dietpi-update: line 528: local: can only be used in a function**
[  OK  ] DietPi-Update | systemctl restart dietpi-ramdisk
```

attached the full log

[update.txt](https://github.com/MichaIng/DietPi/files/2950084/update.txt)

MichaIng, 2019-03-10:
@Joulinar 
Jep all these are expected, fixed with the update itself and have no effect on success.

Because it is important that this was successful, could you paste:
`cat /etc/apt/preferences.d/dietpi-wireguard*`

And less important but check if `G_AGUP` now does not show the GPG error anymore?

Joulinar, 2019-03-10:
@MichaIng 
Just as info, Wireguard was already installed before updating to 6.22.

```
root@DietPi:~# cat /etc/apt/preferences.d/dietpi-wireguard*
Package: *
Pin: release n=sid
Pin-Priority: -1

Package: wireguard wireguard-dkms wireguard-tools
Pin: release n=sid
Pin-Priority: 99
root@DietPi:~#
```

If needed I can go for an empty installation before update and try to install WG once on 6.22

MichaIng, 2019-03-10:
@Joulinar 
> Just as info, Wireguard was already installed before updating to 6.22.

Jep I thought so. Okay the preferences for the sid repo were updated as desired and the Debian keyring is installed, so the GPG error has gone. Perfect, thanks for testing this 👍.

MichaIng, 2019-03-13:
Thanks to @WilburWalsh fixed an issue with the GMediaRender systemd unit: https://github.com/MichaIng/DietPi/issues/2576#issuecomment-472568788

MichaIng, 2019-03-16:
### Beta v6.22.1 has been released: https://github.com/MichaIng/DietPi/pull/2640

MichaIng, 2019-03-21:
### Beta v6.22.2 has been released: https://github.com/MichaIng/DietPi/pull/2643
- DietPi-Banner is now customizable, simply run: `dietpi-banner`

chaosx2019, 2019-03-21:
Updated to V6.22.2

dietpi-software doesnt work for me, 

> root@TNServer:~# dietpi-software
[  OK  ] DietPi-Software | Root access verified.
[  OK  ] DietPi-Software | RootFS R/W access verified.

[  OK  ] DietPi-Software | Initialized database
[  OK  ] DietPi-Software | Reading database
[  OK  ] DietPi-Software | Connection test: https://deb.debian.org/debian/
[  OK  ] DietPi-Run_NTPD | systemctl restart systemd-timesyncd
[ INFO ] DietPi-Run_NTPD | NTPD: Waiting for completion of systemd-timesyncd (1/60)
[ INFO ] DietPi-Run_NTPD | NTPD: Waiting for completion of systemd-timesyncd (2/60)

just cycles continueously

Joulinar, 2019-03-21:
I don't have issues running `dietpi-software` on DietPi v6.22.2 (beta)

```
root@DietPi:~# dietpi-software
[  OK  ] DietPi-Software | Root access verified.
[  OK  ] DietPi-Software | RootFS R/W access verified.

[  OK  ] DietPi-Software | Initialized database
[  OK  ] DietPi-Software | Reading database
[  OK  ] DietPi-Software | Connection test: http://raspbian.raspberrypi.org/raspbian
[  OK  ] NTPD: time sync | Completed
```

chaosx2019, 2019-03-22:
could it be platform related?, is anyone on x86_64 able to confirm that theirs works/ doesn't work?

Joulinar, 2019-03-22:
its working on my VM as well

```
DietPi v6.22.2 (beta) | 01:06 - Fr 22.03.2019
Device model | Virtual Machine (x86_64)
```

```
root@DietPi:~# dietpi-software
[  OK  ] DietPi-Software | Root access verified.
[  OK  ] DietPi-Software | RootFS R/W access verified.

[  OK  ] DietPi-Software | Initialized database
[  OK  ] DietPi-Software | Reading database
[  OK  ] DietPi-Software | Connection test: https://deb.debian.org/debian/
[  OK  ] NTPD: time sync | Completed
```

chaosx2019, 2019-03-22:
then its my build that has a problem. i switched to beta from master ran the update. something must have gone screwy. thanks for the confirm. will look into whats happened

Joulinar, 2019-03-22:
maybe do a restore to 6.21.1 and try to update to beta again.

chaosx2019, 2019-03-22:
override worked not entirely clean and is a bit risky but its fixed the problem its working normally now. very strange little bug that one

MichaIng, 2019-03-22:
@chaosx2019 
Your error is not related to DietPi-Software but it is the network time synchronization failing to connect/sync in reasonable time.

According to the prior connection check, network is generally there. The time actually as well looks good: `23:31 - Thu 21/03/19`
So yeah in this case overriding on the final error prompt (after 60th loop failed as well), is the best choice.

To investigate please paste:
```
journalctl -u systemd-timesyncd
```

What you can try:
`dietpi-config > Network Options: Misc > NTP Mirror >` and select preferred option here.
I would generally recommend to use "Gateway", which attempts syncs time with your router, which should be way faster then syncing to an external NTP server. In most cases the router supports this and syncs itself to a pre-configured external NTP server.

chaosx2019, 2019-03-22:
hi michael,

the journal shows no entries which is weird. also ive tried all the other suggestions and nothing is working.

Override works but its not remembered on reboot. any ideas?

MichaIng, 2019-03-22:
@chaosx2019 
Please run:
```
systemctl start systemd-timesyncd
sleep 2
systemctl status systemd-timesyncd
```

chaosx2019, 2019-03-22:
systemd-timesyncd.service - Network Time Synchronization
   Loaded: loaded (/lib/systemd/system/systemd-timesyncd.service; disabled; vendor pr
eset: enabled)
  Drop-In: /lib/systemd/system/systemd-timesyncd.service.d
           └─disable-with-time-daemon.conf
   Active: inactive (dead)
     Docs: man:systemd-timesyncd.service(8)

tried a couple of times

MichaIng, 2019-03-22:
@chaosx2019 
Ah, you have NTP installed. We really need to add an auto detection for this 🤔.

If you don't need super precise system time (high frequency sensor logging or stuff), or want to use your DietPi system as NTP server, please uninstall it:
`apt purge ntp`

If you really need to keep it, then select "Custom" as time sync mode:
```
dietpi-config > Advanced Options > Time sync mode > Custom
```

chaosx2019, 2019-03-22:
that fixed it. thanks i dont remember installing ntp, but i will keep an eye on that in the future

MichaIng, 2019-03-22:
### Beta v6.22.3 has been released: https://github.com/MichaIng/DietPi/pull/2648
- DietPi-Software | Allo GUI: Updated to version 13. Provides for view the USB status and sound card status on Dietpi Allo web GUI system status page. Many thanks to @sudeep!: https://github.com/sparky-sbc/sparky-test/tree/master/dietpi-gui-usbdebug
- DietPi-Banner | Added support for custom colour codes: #2649

#### This is expected to be the last beta. We will run final tests and release is expected to be tomorrow evening March 23rd.

MichaIng, 2019-03-23:
### v6.22 has been released. Many thanks to all testers ❤️ 🎉!!

closed 2019-03-23 by MichaIng