MichaIng/DietPi #2634: Funkwhale Cloud Media Player (Similar to Airsonic)

Millichrome, 2019-03-12:
### Creating a software request

#### Formal software information
- Software title | Funkwhale
- Short description | A cloud player. A modern alternative to airsonic/subsonic and Ampache. 
- Official URL | https://funkwhale.audio/
- GitLab URL | https://dev.funkwhale.audio/funkwhale/funkwhale
- Install documentation | https://docs.funkwhale.audio/installation/index.html

#### Are there similar/alternative software titles available with DietPi-Software?
Airsonic, Subsonic, Ampache, 

#### What makes your requested software better than the above solutions, if available?
Much better UI/UX. Active development.  

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?
Lot's of manual steps to install. I could not get the docker image installation to work. Users who are not pros at linux/docker would benefit from DietPi supporting Funkwhale. 

#### Can you provide the installation steps that you would suggest DietPi-Software to do?
You can either install Funkwhale via docker image or via "direct" install:
https://docs.funkwhale.audio/installation/index.html

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.
I am not a coder or linux pro, but I have a "non-production" 3B+ that I can use for testing.

MichaIng, 2019-03-12:
@Millichrome 
Many thanks for your request.

Please consider to add it to our FeatHub page: https://feathub.com/MichaIng/DietPi/

Millichrome, 2019-03-12:
Done! Cheers!

MichaIng, 2019-03-13:
For reference: https://feathub.com/MichaIng/DietPi/+59

MichaIng, 2022-01-02:
16 votes on FeatHub in ~3 years: https://feathub.com/MichaIng/DietPi/+59