MichaIng/DietPi #2637: Plex

JJGO, 2019-03-14:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?
- `dietpi-bugreport` ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

MichaIng, 2019-03-15:
@JJGO 
Thanks for your report. I will try to replicate.

Errors from your log:
```
Mar 14, 2019 20:25:42.665 [0x70dff400] ERROR - Error issuing curl_easy_perform(handle): 7
Mar 14, 2019 20:25:42.665 [0x70dff400] WARN - HTTP error requesting POST http://127.0.0.1:32600/devices/discover (0, No error) (Failed connect to 127.0.0.1:32600; Connection refused)
...
Mar 14, 2019 20:25:46.367 [0x745435d0] ERROR - Transcoder: Error cleaning old transcode sessions: boost::filesystem::directory_iterator::construct: No such file or directory: "/var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Cache/Transcode/Sessions"
...
Mar 14, 2019 20:25:46.816 [0x717fd400] ERROR - SQLITE3:(nil), 539, recovered 3 pages from /var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Cache/gn_lists.gdb-journal
...
Mar 14, 2019 20:25:48.371 [0x6dfff400] ERROR - Network Service: Error in browser handle read: 125 (Operation canceled) socket=-1
... repeatedly
```

With v6.22 we will switch to the new Plex v1.15 packages, although I am currently not yet sure if we should automatically update existing installs or leave them to not mess with running systems. The new installs on ARM do not use the 3rd party repo anymore, since with v1.15 official ARM binaries are available: https://github.com/MichaIng/DietPi/pull/2554

MichaIng, 2019-03-16:
Found an old thread: https://forums.plex.tv/t/error-error-issuing-curl_easy_perform-handle-28-on-an-sony-bravia-tv/93260/36
- Just what can be tried additionally is disabling all DNLA features from within the web UI (if possible).

WolfganP, 2019-03-16:
@JJGO stop Plex service and try running:
```
LD_PRELOAD=/usr/lib/arm-linux-gnueabihf/libarmmem-v6l.so; sudo rm /etc/ld.so.preload
```
Ref: https://forums.plex.tv/t/plex-server-suddenly-not-starting-on-raspberry-pi-after-crashed-update/387681/5

MichaIng, 2019-03-16:
Just some hours ago v1.15 has been pushed to the dev2day repo as well: https://forums.plex.tv/t/official-pms-for-armv7-what-will-happen-with-this-community-one/372290/21

So I suggest to update the package as well, before applying the above fix:
```
G_AGUP
G_AGI plexmediaserver-installer
```

The repo will be dropped ~1 month later.

JJGO, 2019-03-17:
Thanks for all the suggestions.

I updated the plexmediainstaller with G_AGI, reinstalled the plexmediaserver, and ran the coimmand @WolfganP mentioned. The behaviour is still the same. The Logs contain the following errors 

```
Mar 17, 2019 18:27:29.078 [0x6fcff450] ERROR - Error issuing curl_easy_perform(handle): 7
Mar 17, 2019 18:27:34.199 [0x73c4e540] ERROR - Transcoder: Error cleaning old transcode sessions: boost::filesystem::directory_iterator::construct: No such file or directory: "/var/lib/plexmediaserver/Library/Application Support/Plex Media Server/Cache/Transcode/Sessions"
Mar 17, 2019 18:27:35.841 [0x6dcff450] ERROR - Network Service: Error in browser handle read: 125 (Operation canceled) socket=-1
```

With the last error repeating every 100ms continuously. 

I will try disabling the DLNA part.

How can I install the new beta you mention?

WolfganP, 2019-03-17:
@JJGO it seems your issue is related to the file structure as some files/folders are missing.

If you want to install Plex manually, download the image from `https://www.plex.tv/media-server-downloads/`  and then execute:
`
sudo apt remove plexmediaserver-installer && sudo dpkg -i yourdownloadedimagefile.deb`

MichaIng, 2019-03-17:
@JJGO 
🤔 Lets see if disabling DLNA works.

Did you reboot after updating Plex and applying WolfganP suggesting?

How to apply the Beta: https://github.com/MichaIng/DietPi/blob/beta/BRANCH_SYSTEM.md#steps-to-use-the-dietpi-beta-branch
- Just note that this does not update your Plex or something, especially not since the repo already provides v1.15. So this along will not solve your issue most likely.
- You could try to purge Plex completely and install it freshly after applying the Beta. The installed package/binaries are different, although the same version, so there is a small chance that this will solve your issue.

Otherwise I suggest to report this to the Plex forum, respectively search for similar issues. Important to know that the issue occurred on v1.14 as well.

JJGO, 2019-03-17:
I believe I figured it out

The issue was that after the reinstall, Plex needed a connection from the local network to be configured. I was doing this through a OpenVPN connection since the RPi is offsite. The VPN allowed me to access the plex web client through 192.168.1.X:32400 but did not allow me to do the initial server setup. It did not give any error/warning either, I guess they could improve this.

I was able to get it working with SSH port forwarding using `-L 8888:localhost:32400`.  

My best guess is that my plex had the bug @WolfganP  was referencing and with the reinstall the user configuration was wiped causing this issue. 

Apparently all the errors in the logs were benign (had to google each one of them) and were related to missing folders and DNS with IPv6 configuration.

closed 2019-03-17 by JJGO