MichaIng/DietPi #2638: Nextcloud update from 15.04 to 05 leaves me with ocm ocs provider errors

minnux, 2019-03-15:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.21.1
- Distro version | stretch
- Kernel version | Linux DietPi 4.4.154-1124-rockchip-ayufan-ged3ce4d15ec1 #1 SMP Mon Oct 22 20:59:41 UTC 2018 aarch64 GNU/Linux
- SBC device | RockPro64 (aarch64)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | none

#### Additional Information (if applicable)
- Software title | Nextcloud 15.05
- updated
- Can this issue be replicated on a fresh installation of DietPi? | not sure
- `dietpi-bugreport` ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce

Upgrade from Nextcloud 15.04 to 15.05

#### Expected behaviour
no error messages
- ...

#### Actual behaviour
Nextcloud -> Settings -> Overview under "Security & setup warnings"

"Your web server is not properly set up to resolve "/ocm-provider/". This is most likely related to a web server configuration that was not updated to deliver this folder directly. Please compare your configuration against the shipped rewrite rules in ".htaccess" for Apache or the provided one in the documentation for Nginx at it's documentation page. On Nginx those are typically the lines starting with "location ~" that need an update."

"Your web server is not properly set up to resolve "/ocs-provider/". This is most likely related to a web server configuration that was not updated to deliver this folder directly. Please compare your configuration against the shipped rewrite rules in ".htaccess" for Apache or the provided one in the documentation for Nginx at it's documentation page. On Nginx those are typically the lines starting with "location ~" that need an update."

#### Extra details

I realize this is a Nextcloud upgrade issue and not a dietpi issue but the solutions I have found are for Apache/Nginx and/or on remote servers. See here: https://help.nextcloud.com/t/your-web-server-is-not-properly-set-up-to-resolve-ocm-provider/48530

I'm running the default dietpi Nextcloud installation that includes lighttpd webserver on a local server and I can't figure out how I would resolve this error message using my particular configuration and the solutions I've found on the Nextcloud forum. Any help as to what to change would be appreciated. Thank you.
[image]

MichaIng, 2019-03-15:
@minnux 
Many thanks for your report.

Yes I already recognized the same.

On my Apache server I solved it by adding:
```
Redirect permanent /ocm-provider /nextcloud/ocm-provider
Redirect permanent /ocs-provider /nextcloud/ocs-provider
```

On Lighttpd run:
```
echo '# Redirect OCM/OCS requests to Nextcloud endpoint:
url.redirect = (
	"^/ocm-provider"  => "/nextcloud/ocm-provider",
	"^/ocs-provider" => "/nextcloud/ocs-provider"
)' >> /etc/lighttpd/conf-available/99-dietpi-owncloud.conf
```

MichaIng, 2019-03-15:
PR up and ready to test, however solution is quite clear: https://github.com/MichaIng/DietPi/pull/2639

minnux, 2019-03-15:
That didn't resolve it. Should the last part be nextcloud.conf and not owncloud.conf?

MichaIng, 2019-03-15:
@minnux 
Whoopsie jep of course, copy&paste from wrong install 😅. Remove the owncloud.conf again 😉.

minnux, 2019-03-15:
Nextcloud web interface fails to load now. Here is my current 99-dietpi-nextcloud.conf

`$HTTP["url"] =~ "^/nextcloud($|/)" {
  setenv.add-environment += (
    "PHP_ADMIN_VALUE" => "opcache.memory_consumption=128",
  )
}

# Set "Referrer-Policy" = "no-referrer" security header
setenv.add-response-header = ( "Referrer-Policy" => "no-referrer" )
# Redirect OCM/OCS requests to Nextcloud endpoint:
url.redirect = (
"^/ocm-provider"  => "/nextcloud/ocm-provider",
"^/ocs-provider" => "/nextcloud/ocs-provider"
)
`

MichaIng, 2019-03-15:
@minnux 
Jep also ran into this issue. `url.redirect = (` fails if it is two times inside the configs.

Please change it to `url.redirect += (`, at best as well in: /etc/lighttpd/conf-enabled/99-dietpi-dav_redirect.conf

minnux, 2019-03-15:
It worked! Thanks for the feedback. No more error messages.

closed 2019-03-15 by minnux