MichaIng/DietPi #2641: Permissions on /mnt/dietpi_userdata reset after reboot

JJGO, 2019-03-18:
Every time I reboot the machine the permissions on `/mnt/dietpi_userdata/dokuwiki` are reset from `www-data:dokuwiki` to `dietpi:dietpi`. 

I need the folder to have user `www-data` for the web server to edit files and the group to include other users such as the syncthing daemon. 

I could automate the permission change on startup but my question is why is this happening and if it is expected behavior. 

Thank you

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
```
#!/bin/bash
G_DIETPI_VERSION_CORE=6G_DIETPI_VERSION_SUB=21
G_DIETPI_VERSION_RC=1G_GITBRANCH=master
G_GITOWNER=Fourdee
```
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
```
9.8
```
- Kernel version | `uname -a`
```
Linux Nal 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux
```
- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
```
RPi 3 Model B+ (armv7l)
```
- Power supply used | (EG: 5V 1A RAVpower)
5V 2.4A Official Raspberry Pi
- SDcard used | (EG: SanDisk ultra)
SanDisk  32GB

MichaIng, 2019-03-18:
@JJGO 
Many thanks for your report.

Nope permissions reset on user data is not expected on reboot, only when doing a `dietpi-software` install. But we are already thinking to skip that as well.

Could you please paste:
```
cat /DietPi/dietpi/.install_stage
cat /boot/dietpi/.install_stage
lsblk -f
```

JJGO, 2019-03-18:
I see, I correlated with the wrong thing then. Pretty much all my reboots were triggered from some install so it makes sense. 

The output is 

```
$ cat /DietPi/dietpi/.install_stage
2
$ cat /boot/dietpi/.install_stage
2
$ lsblk -f
NAME        FSTYPE LABEL  UUID                                 MOUNTPOINT
mmcblk0                                                        
├─mmcblk0p1 vfat   boot   5DB0-971B                            /boot
└─mmcblk0p2 ext4   rootfs 060b57a8-62bd-4d48-a471-0d28466d1fbb /
```

Is there a way to tell `dietpi-software` to skip some folders when reseting permissions on install?

MichaIng, 2019-03-18:
@JJGO 
> I see, I correlated with the wrong thing then. Pretty much all my reboots were triggered from some install so it makes sense. I tested and a single reboot does not change permissions.

Ah okay yes then this is expected. Currently there is no way to skip this, besides manually editing the script. But this would be reset on update etc. so I think currently your best way is to re-set permissions on boot.

However I will turn this issue into a feature request to skip setting permissions for the whole user data dir on each dietpi-software install, but instead set them individually only for the dirs/files required for the currently installed software title.

MichaIng, 2019-05-26:
Another vote for this by John: https://dietpi.com/phpbb/viewtopic.php?f=11&t=5866

MichaIng, 2019-08-06:
PR up: #3040

PR #3040 (2019-08-10): DietPi-Software | Skip setting file permissions globally after each install
**Status**: testing
- [x] DietPi-Set_software

Fixes: #2641

**Commit list/description**:
+ DietPi-Software | Skip "/DietPi/dietpi/func/dietpi-set_software setpermissions" to not overwrite user custom permissions and probably other software permission modes that are more secure. Instead apply required permissions directly during individual software config step.
+ DietPi-Software | BaiKal: Handle reinstalls properly, replace the old instance but keep the old database/configs dir. Install and enable required PHP modules  [... cut]
closed 2019-10-15 by MichaIng