MichaIng/DietPi #2642: unable to resolve host address 'deb.debian.org'

ngentil, 2019-03-19:
Log file contents:                                                                       
    │ Spider mode enabled. Check if remote file exists.                                        
    │ --2019-03-19 17:49:07--  https://deb.debian.org/debian/                                  
    │ Resolving deb.debian.org (deb.debian.org)... failed: Connection timed out.               
    │ wget: unable to resolve host address 'deb.debian.org'

MichaIng, 2019-03-19:
@ngentil 
Thanks for your report.

Does resolving other hostnames work?
`G_CHECK_URL dietpi.com`

In case is your DNS server configured correctly?
`cat /etc/resolv.conf`

Using DHCP the above content should be set automatically correctly as derived via DHCP. Otherwise you need to set this as static DNS via dietpi-config > Network Options: Adapters > Ethernet/WiFi > Change mode > STATIC and enter e.g. your local router as DNS server (if it support this) or any external DNS offered.

ngentil, 2019-03-21:
yes i  have set static ip and uses 8.8.8.8 , 8.8.4.4 as dns 
both dhcp and static dont work, i have corrected time zone also

ngentil, 2019-03-21:
this is on a odroid c2 - have tried armbian and it works fine...

MichaIng, 2019-03-21:
@ngentil 
And `G_CHECK_URL dietpi.com` fails with the same error?
Or using curl instead: `curl https://dietpi.com/downloads/testing/lighttpd.conf`

ngentil, 2019-03-24:
#### Details:
- Date           | Sun Mar 24 02:53:08 GMT 2019
- Bug report     | N/A
- DietPi version | v6.14.0 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | Meveric
- SBC device     | Odroid C2 (aarch64) (index=12)
- Kernel version | #1 SMP PREEMPT Sun Aug 19 15:27:56 CEST 2018
- Distro         | stretch (index=4)
- Command        | Connection test: dietpi.com
- Exit code      | 4
- Software title | 

====================================================

Spider mode enabled. Check if remote file exists.
--2019-03-24 02:52:54--  http://dietpi.com/
Resolving dietpi.com (dietpi.com)... failed: Connection timed out.
wget: unable to resolve host address 'dietpi.com'

=====================================================

Spider mode enabled. Check if remote file exists.                                        
--2019-03-24 03:02:59--  https://deb.debian.org/debian/                                  
Resolving deb.debian.org (deb.debian.org)... failed: Connection timed out.               
wget: unable to resolve host address ‘deb.debian.org’   
======================================================

DHCP is enabled by default...

MichaIng, 2019-03-24:
@ngentil 
Okay either your resolver is broken or internet access at all.

Please try connection to a plain IP: `G_CHECK_URL 149.20.4.15`
- This is one particular mirror deb.debian.org was resolved to on my machine.

Paste:
```
ip a
ip r
cat /etc/resolv.conf
cat /etc/network/interfaces
```

I would have asked for a bug report sent, but as connection fails, we need to investigate the outputs here 😉.

MichaIng, 2019-05-01:
We added the possibility to raise the timout and retry count for G_CHECK_URL which should resolve this issue. This could be further tweaked, e.g. we scrape the output for "timeout" and in case give more specific info and allow to change these settings directly (or open the submenu directly): https://github.com/MichaIng/DietPi/issues/2717#issuecomment-485405270

closed 2019-05-12 by MichaIng