MichaIng/DietPi #2644: Slow speed outside of LAN

minnux, 2019-03-20:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.21.1
- Distro version | stretch
- Kernel version | Linux DietPi 4.4.154-1124-rockchip-ayufan-ged3ce4d15ec1 #1 SMP Mon Oct 22 20:59:41 UTC 2018 aarch64 GNU/Linux
- SBC device | RockPro64 (aarch64)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | none

#### Additional Information (if applicable)
- Software title | Nextcloud 15.05
- updated
- Can this issue be replicated on a fresh installation of DietPi? | not sure
- `dietpi-bugreport` ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce

Download files outside of LAN

#### Expected behaviour
Should be speeds around 3 MB/s

#### Actual behaviour
Speed hovers between 300-400 KB/s

#### Extra details
This is not a new issue and has existed since I first installed NC via dietpi. I've done several tests via iperf and torrents from my Nextcloud server and they've all given me the speeds I should have. 

-When uploading and downloading files inside my LAN everything is fast. 

-When uploading files to my NC server from outside my LAN speeds are fast. 

-It's only when downloading files from my NC server outside my LAN that speeds are slow. Between 300-400 KB/s. 

I found this thread: https://help.nextcloud.com/t/slow-download-speed-on-wan/24254 and his issues seem to be almost identical to mine. But that was 2017 and it looks like by your comments that his issue was already resolved by now? And one of his links is broken as well. I'm using a LLMP stack where he was using Apache so not sure if my issue is related. What configuration changes should I be making to get the correct speeds? I can't help but think this is a PHP issue.

MichaIng, 2019-03-20:
@minnux 
Thanks for your report.

Are you sure that 300-400 KB/s are not expected due to upload cap? It is very common that upload bandwidth is way lower than download bandwidth in non fiber optic connections. This would then limit download speed from your NC accordingly.

So to compare e.g. with torrents you need to compare with the seeding/upload speed.

And as well note that transfer through Nextcloud is known to be not the fasted, at least when downloading via their clients or any other WebDAV client. Do to sort out any limitation of the Nextcloud transfer protocol, compare it with direct download of a file located within your webserver root.

minnux, 2019-03-20:
Yes I did the iperf test both ways, seeded a private torrent from my NC server to a friend as well. Both gave me upload speeds between 2.5-3 MB/s on their end. I used these tests to bypass going through NC. The current plan I'm on from my ISP gives me 400 Mb/s down and 20 Mb/s up with speed tests giving me approximately 486 Mb/s down and 26 Mb/s up. 

Downloading from the NC web interface or from the app/clients outside my network give me the aforementioned 300-400 KB/s speeds. When uploading a file remotely though the speeds are fine. I will run more tests and report back. Maybe FTP and SCP.

MichaIng, 2019-03-20:
@minnux 
Okay to sort out webserver limitations please run a static file download test from your webserver root.

minnux, 2019-03-20:
What protocol do you recommend I use? I tried SCP but the server is using dropbear so I'm guessing it didn't work because SCP isn't included in dropbear?

MichaIng, 2019-03-20:
@minnux 
Dropbear should provide SCP as well, however not 100% sure currently, OpenSSH definitely does.

But I don't meant this, but instead downloading via browser from your webroot directly. So place a large test file into` /var/www` and in your browser download via your.domain.org/testfile.

minnux, 2019-03-20:
Ah ok I see. Will do that then and report back

minnux, 2019-03-21:
Ok done. A 1GB file is downloading at ~ 300KB/s

minnux, 2019-03-21:
Also I forgot to add, it starts out downloading around 2 MB/s then quickly lowers to 300 KB/s

MichaIng, 2019-03-21:
@minnux 
Okay so it's not about Nextcloud but has to do with the webserver itself.

minnux, 2019-03-21:
Yes lighttpd with mariadb and php

minnux, 2019-03-21:
Ok here are some very strange details. 

I have being doing the remote tests on my mac at work since that's the remote network I have access to. Today I decided to bring my laptop running linux (solus 4) up here just for more tests. It's a cheap winbook with an intel cherry trail z8350 I believe and only has wifi. I connect to my work wifi and log into my nextcloud using firefox and am downloading my 1GB test file getting speeds between 2.8-3.0 MB/s. I go back to my work computer, a 2018 mac pro hardwired to the network and try to download the file again via firefox. 300 KB/s. I try Chrome on the mac pro. 300 KB/s. Safari. 300 KB/s. It doesn't make sense.

I go back to my laptop and download the test file from webroot and and getting about 1.6 MB/s, still better than 300 KB/s. I download the test file from nextcloud and it's the same, 1.6 MB/s. This could be wifi interference or something though as I do not have an ethernet port on this cheap laptop. I will have to try again and see if my USB ethernet adapter is compatible with my laptop.

In the meantime I would like to share a nextcloud link to the test file with you if you do not mind and if you can see what kind of speed you are getting.

Fourdee, 2019-03-22:
@minnux 

Might be worth installing NetData on the NC server, it can provide stats info and hopefully list any bottle necks during NC downloads:
http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=60#p1611

Also will trigger if any network issues occur on the server (eg: packet loss etc).

Also also, most ISP's will throttle upload/download speed when a hard set limit is reached. Might also be worth checking your ISP's bandwidth allocation/throttling policy to confirm if the cause.

MichaIng, 2019-03-22:
Okay so I sum up the important facts:
- It is not an issue of the NC server or Nextcloud in particular, since download speeds are slow as well when downloading directly via browser from webroot and with some browsers/systems download speed is expected fast.
- It is (as of current tests) limited to downloads from the webserver.
- It is limited to non-local network downloads, so either NAT or ISP related.
- It is limited to uploads "from" the server, since uploads "to" the server have expected speed.

I guess WiFi is not relevant here, since local network downloads are fast and also WiFi speed, even if unstable connection, usually outweigh www bandwidth even in case of "g".
But still a good idea to test with Ethernet.

Yeah NetData is indeed the best idea to monitor transfer speeds and issues!

minnux, 2019-03-22:
Ok netdata has been installed via dietpi-software. There's a lot of data there to go over so it will take me a little while to interpret.

MichaIng, 2019-05-26:
@minnux 
Any news on this? I mark the issue as closed for now. Feel free to reopen if required.

closed 2019-05-26 by MichaIng