MichaIng/DietPi #2646: Nextcloud Cron-Job Problems

Borotes, 2019-03-20:
For quite some time now, I have the problem that my Nextcloud is suddenly no longer reachable and the sync isn't working.

Luckily, I am still able to access it via Putty and restart it. However, lately restarting doesn't help. Only turning off the Raspberry Pi by "poweroff" completely and turning it on again helps. 

If I then, I sign in with my admin account I am getting the following error message saying that there is a problem with the cron service.

[image]

Is there a solution to this? When this happened in the past I reinstalled it, but the problem seems to reappear.

Nextcloud version: 15.0.5
DietPi: v6.20.

I would appreciate any help 😊

MichaIng, 2019-03-20:
@Borotes 
Thanks for your report.

Could you try to update to current Beta? It resolves some issues with failing services on startup: https://github.com/MichaIng/DietPi/blob/beta/BRANCH_SYSTEM.md

Then we check service states:
`dietpi-services status`

Borotes, 2019-03-24:
Thank you, for your quick reply! I just updated and keep you posted.

Borotes, 2019-03-24:
It is looking like that now. If it happens again, I will paste the new log in here 😊

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: status

[  OK  ] DietPi-Services | fail2bantactive (running) since Sun 2019-03-24 11:33:                                                                                                                                                             24 CET; 1min 11s ago
[  OK  ] DietPi-Services | redis-servertactive (running) since Sun 2019-03-24 11                                                                                                                                                             :33:25 CET; 1min 10s ago
[  OK  ] DietPi-Services | mariadbtactive (running) since Sun 2019-03-24 11:33:2                                                                                                                                                             9 CET; 1min 7s ago
[  OK  ] DietPi-Services | php7.0-fpmtactive (running) since Sun 2019-03-24 11:3                                                                                                                                                             3:29 CET; 1min 6s ago
[  OK  ] DietPi-Services | lighttpdtactive (running) since Sun 2019-03-24 11:33:                                                                                                                                                             30 CET; 1min 6s ago
[  OK  ] DietPi-Services | crontactive (running) since Sun 2019-03-24 11:33:30 C                                                                                                                                                             ET; 1min 6s ago
[  OK  ] DietPi-Services | dropbeartactive (running) since Sun 2019-03-24 11:33:                                                                                                                                                             22 CET; 1min 14s ago
[  OK  ] DietPi-Services | dietpi-ramdisktactive (exited) since Sun 2019-03-24 1                                                                                                                                                             1:33:11 CET; 1min 25s ago
[  OK  ] DietPi-Services | dietpi-ramlogtactive (exited) since Sun 2019-03-24 11                                                                                                                                                             :33:10 CET; 1min 25s ago
[  OK  ] DietPi-Services | dietpi-preboottactive (exited) since Sun 2019-03-24 1                                                                                                                                                             1:33:12 CET; 1min 24s ago
[  OK  ] DietPi-Services | dietpi-boottactive (exited) since Sun 2019-03-24 11:3                                                                                                                                                             3:22 CET; 1min 14s ago
[  OK  ] DietPi-Services | dietpi-postboottactive (exited) since Sun 2019-03-24                                                                                                                                                              11:33:22 CET; 1min 14s ago
[ INFO ] DietPi-Services | dietpi-wifi-monitortinactive (dead)

MichaIng, 2019-03-24:
@Borotes 
Great, looks good for now. Is the `t` between the service name and "`active`" just a copy paste error or is it really there (instead of free space) on the terminal?

Borotes, 2019-03-24:
https://i.imgur.com/eq0QOli.png

Here is a screenshot, I don't think it's a copy paste error 😲

MichaIng, 2019-03-24:
@Borotes 
Jep see above. Syntax issue, luckily visual only.

MichaIng, 2019-03-24:
Okay I mark this as closed. However feel free to reopen if the issue reappears.

closed 2019-03-24 by MichaIng