MichaIng/DietPi #2647: diet pi installing hassio, just using the dietpi-software

zjxlsgh, 2019-03-21:
#### Details:
- Date           | Thu 21 Mar 12:14:56 AWST 2019
- Bug report     | N/A
- DietPi version | v6.21.1 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B+ (armv7l) (index=3)
- Kernel version | #1200 SMP Tue Feb 12 20:27:48 GMT 2019
- Distro         | stretch (index=4)
- Command        | 7za x -y 7.7z -offmpeg_rpi
- Exit code      | 127
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
/DietPi/dietpi/func/dietpi-globals: line 1373: 7za: command not found
```

MichaIng, 2019-03-21:
@zjxlsgh 
Many thanks for your report.

Jep known issue with current RPi image until v6.22 release: https://github.com/MichaIng/DietPi/issues/2631

To resolve now run: `G_AGI p7zip-full`

zjxlsgh, 2019-03-21:
that was quick, i just found out i need 7zip to make it work, very new to linux here. xD thanks. pls delete the post now

MichaIng, 2019-03-23:
v6.22 was just released, so the image should work now out of the box without above error.

closed 2019-03-23 by MichaIng