MichaIng/DietPi #2652: USBridge won't update

mikepinkerton, 2019-03-22:
#### Details:
- Date           | Thu  3 Nov 17:22:48 GMT 2016
- Bug report     | N/A
- DietPi version | v6.13
- Img creator    | n/a
- Pre-image      | n/a
- SBC device     | Sparky SBC (armv7l) (index=70)
- Kernel version | #22 SMP PREEMPT Fri Dec 22 10:45:41 IST 2017
- Distro         | stretch (index=4)
- Command        | G_AGUP
- Exit code      | 100
- Software title | DietPi-Update

#### Steps to reproduce:
- dietpi-update from command line

#### Expected behaviour:
Software updates

#### Actual behaviour:
Errors, see below. 

#### Additional logs:
```
Log file contents:
Ign:1 http://ftp.debian.org/debian stretch InRelease
Hit:2 http://ftp.debian.org/debian stretch-updates InRelease
Hit:3 http://ftp.debian.org/debian stretch-backports InRelease
Hit:4 http://ftp.debian.org/debian stretch Release
Ign:5 https://deb.debian.org/debian-security stretch/updates InRelease
Err:6 https://deb.debian.org/debian-security stretch/updates Release
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Reading package lists...
E: The repository 'https://deb.debian.org/debian-security stretch/updates Release' does no longer have a Release file.
```

MichaIng, 2019-03-22:
@mikepinkerton 
Many thanks for your report.

~Confirmed~: ... öhm now it works. Hard to replicate since deb.debian.org is a mirror director so it connects to different near mirrors each time. Just one minute ago it did not work in my tests as well, but now it works:
- HTTP: http://cdn-aws.deb.debian.org/debian-security/
- HTTPS: https://cdn-aws.deb.debian.org/debian-security/

Okay I guess it is a temporary issue with some or one certain mirror behind the mirror director. Simply retry now or tomorrow. You can work around by switching to another explicit mirror via `dietpi-config > Network Options: Misc > APT Mirror`. E.g. check:
- https://www.debian.org/mirror/list
- And use one that is close to your geo location.

mikepinkerton, 2019-03-23:
I verified that switching the APT mirror does not fix the problem.

MichaIng, 2019-03-23:
@mikepinkerton 
Ah sorry I've overseen to more important part of your log paste:
`server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none`

Please do:
`apt install --reinstall ca-certificates apt-transport-https`

mikepinkerton, 2019-03-23:
I tried that, here is its output:

root@USBridge:~# apt install --reinstall ca-certificates apt-transport-https
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following package was automatically installed and is no longer required:
  libopts25
Use 'apt autoremove' to remove it.
The following packages will be upgraded:
  apt-transport-https
1 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 85 not upgraded.
Need to get 349 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 http://ftp.uk.debian.org/debian stretch/main armhf apt-transport-https armhf 1.4.9 [167 kB]
Get:2 http://ftp.uk.debian.org/debian stretch/main armhf ca-certificates all 20161130+nmu1+deb9u1 [182 kB]
Fetched 349 kB in 0s (361 kB/s)      
Preconfiguring packages ...
(Reading database ... 30811 files and directories currently installed.)
Preparing to unpack .../apt-transport-https_1.4.9_armhf.deb ...
Unpacking apt-transport-https (1.4.9) over (1.4.8) ...
Preparing to unpack .../ca-certificates_20161130+nmu1+deb9u1_all.deb ...
Unpacking ca-certificates (20161130+nmu1+deb9u1) over (20161130+nmu1+deb9u1) ...
Setting up apt-transport-https (1.4.9) ...
Setting up ca-certificates (20161130+nmu1+deb9u1) ...
Updating certificates in /etc/ssl/certs...
0 added, 0 removed; done.
Processing triggers for ca-certificates (20161130+nmu1+deb9u1) ...
Updating certificates in /etc/ssl/certs...
0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...
done.

Now when i do dietpi-update, it still fails in the same place.

Log file contents:
Ign:1 http://ftp.uk.debian.org/debian stretch InRelease
Hit:2 http://ftp.uk.debian.org/debian stretch-updates InRelease
Hit:3 http://ftp.uk.debian.org/debian stretch-backports InRelease
Hit:4 http://ftp.uk.debian.org/debian stretch Release
Ign:5 https://deb.debian.org/debian-security stretch/updates InRelease
Err:6 https://deb.debian.org/debian-security stretch/updates Release
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Reading package lists...
E: The repository 'https://deb.debian.org/debian-security stretch/updates Release' does no longer have a Release file.
```
(note trying a different mirror, the UK one).

MichaIng, 2019-03-24:
@mikepinkerton 
Okay if the error still persists, please try the alternative security repo URL:
```
sed -i '/debian-security/d' /etc/apt/sources.list
echo 'deb http://security.debian.org/debian-security stretch/updates main contrib non-free/' >> /etc/apt/sources.list
```

mikepinkerton, 2019-03-27:
That got farther, then I got this error:

```
Log file contents:
Spider mode enabled. Check if remote file exists.
--2016-11-08 01:06:03--  https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.248.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.248.133|:443... connected.
ERROR: The certificate of ‘raw.githubusercontent.com’ is not trusted.
ERROR: The certificate of ‘raw.githubusercontent.com’ is not yet activated.
The certificate has not yet been activated
```

I'd file a separate bug, but as it still seems cert-related, I figured I'd keep it here.

MichaIng, 2019-03-27:
@mikepinkerton 
My fault, I didn't see that your system clock is out of sync: `Date | Thu 3 Nov 17:22:48 GMT 2016`
This is the reason for all SSL related issue. The certificates are "not yet" valid in your systems opinion 🤣.

Run: `/DietPi/dietpi/run_ntpd 1`
Check if the system clock is correct then: `date`

mikepinkerton, 2019-03-27:
Thanks for noticing the date issue. I turned NTP sync back on (it was required to disable in order to fix other known issues with prior upgrades) and restarted and update is now complete. 

Marking closed. It's probably worth calling out when NTP is disabled when running dietpi-update because it's almost guaranteed to fail at some point because of clock drift with this hardware.

closed 2019-03-27 by mikepinkerton

MichaIng, 2019-03-27:
@mikepinkerton 
`NTP` can stay disabled/removed. But then you need to enable `boot + daily` or `daemon` time sync mode via dietpi-config > Advanced Options to enable time sync via systemd-timesyncd.
If you use NTP (as another time sync daemon), then you need to choose "custom" to disable the then obsolete systemd-timesyncd runs.

Actually `NTP` is not required in very most cases, only if you need very exact times when doing e.g. high precision sensor logging and such. But usually the lightweight systemd internal method is accurate enough. Just verify that you chose a valid mirror in dietpi-config > Network Options Misc > NTP Mirror. It's a bid confusing that we call it NTP there as well, since this is the abbreviation of the underlying protocol, but the NTP package is just one option to use this protocol, and on that is not too leightweight (according to resource usage).

mikepinkerton, 2019-03-27:
To be clear I had to re-enable time sync in dietpi-config, which i had to disable to work around other dietpi updating bugs in the past.

MichaIng, 2019-03-27:
@mikepinkerton 
Ah okay, sorry for misunderstanding. Jep that is true, I hope our time sync implementation does not produce issues anymore now 😉.