MichaIng/DietPi #2659: RaspberryPi: booting from USB-Drive

Gill-Bates, 2019-03-25:
Hello,

I still haven't managed to boot the Raspberry Pi without an SD card, exclusively with a USB hard disk (SSD 2.5"). I write the image (Dietpi/Raspian) on the disk, connect it to the Raspberry Pi and boot the system. But it doesn't start.

Of course I set the boot bit. Moving the filesystem to the external disc works curiously. However, the SD card must always remain in the device.

Am I trying to do something here that isn't officially supported? I would like to do without the SD card completely.

MichaIng, 2019-03-25:
@Gill-Bates
Indeed it is a missing feature of the RPi itself, AFAIK. The boot partition itself needs to stay on the SDcard while the root partition can be moved by changing the cmdline.txt entry accordingly. This is what we do when moving the root partition via dietpi-drive_manager.

Not sure currently but I read something about USB boot support. Will investigate. Perhaps we can offer an USB image then.

MichaIng, 2019-03-26:
Ah lol we already have the feature added to `dietpi-config`: https://github.com/MichaIng/DietPi/blob/dev/dietpi/dietpi-config#L1317-L1325
- However since RPi3+ does not require the boot bit to be set, we can enhance this slightly.
- Also since above the warn somehow that this cannot be undone, would be interesting if SDcard boot is still possible with this enabled, because otherwise a second warning prompt would be good.

So yeah, definitely not possible on RPi1/2.

MichaIng, 2019-06-17:
Already implemented and just fixed, as this option was shown on RPi 3+ instead of RPi 3 non-plus only: https://github.com/MichaIng/DietPi/commit/b21976e58880b262f11b1c596b2d24d481397155

closed 2019-06-17 by MichaIng

netwrkx, 2019-11-09:
@MichaIng I can't get my RP3b+ to boot from hdd, whenever I boot from sdcard, my RP3B+ can detect my hdd and access it.

Any idea why?

                                                                          ```
DietPi-Update
─────────────────────────────────────────────────────
 Mode: Checking for DietPi updates

[ INFO ] DietPi-Update | Checking mirror: https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/server_version-6
[  OK  ] DietPi-Update | Using update server: https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/server_version-6
[  OK  ] DietPi-Update | No update required, your DietPi installation is already up to date:
[ INFO ] DietPi-Update | Current version : v6.26.3                       [ INFO ] DietPi-Update | Latest version  : v6.26.3
```

MichaIng, 2019-11-11:
Ref: #3223