MichaIng/DietPi #2660: Update /etc/host when using static IP

Gill-Bates, 2019-03-25:
It would be cool if /etc/hosts gets an entry with the hostname of the device as soon as DHCP is switched to Static IP.

MichaIng, 2019-03-25:
@Gill-Bates 
`/etc/hostname` and `/etc/hosts` is already updated when changing the host name via `dietpi-config`: https://github.com/MichaIng/DietPi/blob/master/dietpi/func/change_hostname#L35-L36

You want the local IP to be applied as host name when switching to static IP? Not sure if this is the desired behaviour. If your device is reachable from www, you should use the FQDN (e.g. `host.domain.org`) as host name, optionally with some leading local name if you offer web services from multiple devices on your local network, e.g. `cloud.domain.org` and `streaming.domain.org` or such.

Okay it doesn't matter actually. It is only the question if the system listens and loops these internal FQDN/host names or not. But using `localhost` or `127.0.0.1` on the device itself and the local IP from within the local network works always anyway.

I was digging a bid and found this old bug report that is the reason why the 127.0.1.1 line actually exists: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=316099
- So 127.0.0.1 should only be associated with localhost and optionally localhost.localdomain only.
- The FQDN/host name otherwise should be associated with either the external IP that points to this host directly or (AFAIK in all NAT situations) 127.0.1.1. Otherwise asking the system for the FQDN (e.g. `hostname -f`) would show the local domain instead of the wanted FQDN.

So yeah, finally I think we handle this already as intended. If you open your device to www then change the host name into the FQDN that points to this server. If you indeed have a static external IP, then replace 127.0.1.1 with it. But since we cannot know your external domain nor if your WAN IP is static or not, we cannot automate this change.

borpin, 2019-04-01:
I have had a similar issue. I set an FQDN via `dietpi-config` (no A record externally), so I could generate an SSL certificate by DNS challenge and navigate via https to the FQDN (LAN only).  Creating the hostname in this way, I believe, set the hosts file to;
```
127.0.0.1 localhost
127.0.1.1 <your_FQDN_hostname>
```
Unfortunately this was not available on my LAN except via the static ip.  To make it accessible internally I had to add a line;
```
192.168.x.x   <your_FQDN_hostname>
```
Adding this line then enabled me to navigate to https://<your_FQDN_hostname> on my LAN (obviously once I had setup the webserver correctly).

This may be what is being requested.

MichaIng, 2019-04-02:
@borpin 
Okay this somehow makes sense when accessing via LAN only.
The DNS challenge works aside from any /etc/hosts entries, right?

Actually it makes sense to always use the local IP there (if static). Otherwise the machine (or even local LAN) would resolve FQDN to the external IP which is an unnecessary overhead.

Only issue:
- If we automate this but e.g. user switches between LAN and WLAN, the entry can be wrong.
- We would need to identify reliable the current main network interface, at best every time any network state changes, check if it has a static IP assigned and in case adjust the /etc/hosts.
- To be consistent, the IPv6 entries should then be adjusted as well (?).
- The big problem now, if anything goes wrong or the IP changes for what any other reason, local software won't be able to resolve the hostname anymore. They should use "localhost" or "127.0.0.1" anyway but the whole idea behind the "127.0.1.1" entry (Debian-only) was to allow local software resolving the FQDN without having it only an alias of "localhost" or "localhost.localdomain".

So actually I am afraid of touching this, especially since for very most users it never matters. At best we can add a prompt to inform user adjusting this manually or ask if we should do this, but it should be not touched without enough understanding what it is for and in which cases it CAN be changed.

borpin, 2019-04-02:
@MichaIng 
> The DNS challenge works aside from any /etc/hosts entries, right?

Yes

As I am running this on a VM with a very static setup I would not find (or consider - so congrats for doing so) these issues.

Perhaps an option to write it to /etc/hosts would be useful?  This is probably an edge case.  Perhaps some warning text with the hostname update that the hosts file may need to be updated?

> To be consistent, the IPv6 entries should then be adjusted as well (?)

I'm currently avoiding IPV6 like the plague for just these sorts of reasons!

MichaIng, 2019-04-02:
Okay so when changing the hostname via dietpi-config, check /DietPi/dietpi/.network for the main network adapter (this is estimated via default route), check whether it's IP is static via /etc/network/interfaces and in case offer to add this IP to /etc/hosts via input box where one can adjust it and with related info that this should only be used if adapters are never  switched.

Since IPv6-only does anyway not work and I doubt that there is anyone not using IPv4 in LAN, IPv6 entries can be skipped now. I am as well not sure what is resolved if the FQDN is associated with local IPv4+IPv6 IPs.

borpin, 2019-04-03:
Yes I think so :smile: 

> add this IP to /etc/hosts via input box

Presume you mean add IP and hostname?  Does this only work if it is an FQDN?  Perhaps list as an 'Advanced' setting?

MichaIng, 2019-04-04:
@borpin 
> Presume you mean add IP and hostname?

Yep I mean instead of adding `127.0.1.1 <hostname>` to /etc/hosts, use `<local IP> <hostname>` instead.

> Does this only work if it is an FQDN?

AFAIK it works with any hostname, for sure at least device internally. If also throughout the local LAN the associated hostname is resolved should be a matter of the router. E.g. my router shows the hostnames for all local LAN devices. When I change a hostname I need to refresh the DHCP lease to make the router receiving the new hostname from the device. But anyway this should only work if the router is as well used as local DNS server (which is in my case default when connection via DHCP), otherwise not sure if it overrides DNS requests to local LAN automatically... Since this topic is about doing this for static IP (not DHCP), did you manually choose the router as DNS server or an external one (e.g. 8.8.8.8)?

When accessing externally, it is anyway only a matter of port forwarding through NAT.

And finally it makes pretty much sense to use the FQDN then. E.g. you do not want to enter a different URL on your browser when accessing from LAN or externally. You want to use the same every time so you can configure e.g. some software clients to be able to connect to the server from wherever you are. The FQDN will always work, but from local LAN you want this to be resolved to the local IP directly to avoid having the request loop through WAN. Hence the /etc/hosts entry to override DNS on local machine and hopefully as well throughout local LAN **<= which needs to be tested when no DHCP and external DNS server is used!**

MichaIng, 2019-04-04:
Did some further tests in my particular LAN/router setup:
- On first access the router takes the hostname from `/etc/hostname` regardless of `/etc/hosts` entry.
- But to allow local resolving via this hostname, it was required to connect at least one time via DHCP client. Otherwise it lists the device with hostname in the network overview but does not resolve the hostname locally regardless. If it's a FQDN it is resolved with the external IP instead of course. This was also regardless of /etc/hosts entry.
- As fast as I did a DHCP lease with the device, it's local DNS resolved correctly throughout local LAN, as well regardless of /etc/hosts entry.
- In my case even with /etc/hostname + /etc/hosts containing desired hostname (associated with local IP) and adding the device with matching hostname manually to the router clients list is not enough to resolve the name correctly in LAN. Even then I need to connect with DHCP client one time, then immediately name resolving works.

So if you make different experiences I guess it's up to the router/DHCP server and it's DNS capabilities in how all this info is stored and provided to clients. So then it indeed looks like /etc/hosts by itself is only used by the device itself and thus should hold the local IP and never the external IP to prevent loopback request overhead.
If in your case /etc/hosts counts, then this is special to your router/DHCP/DNS server. The result is the same that local IP should be used.

borpin, 2019-04-05:
@MichaIng 

> Yep I mean instead of adding 127.0.1.1 <hostname> to /etc/hosts, use <local IP> <hostname> instead.

I have both entries.

The OP may have a different view, but I think that this should be closed - no change - too much of an edge case 😄.

Cheers

MichaIng, 2019-04-05:
@borpin 
> I have both entries.

Hmm, you have your hostname/FQDN associated with two IPs (local and 127.0.1.1)? That wouldn't make much sense. I guess the local system then just resolves with the first associated IP while the second entry has no effect :thinking:.

> I think that this should be closed - no change - too much of an edge case smile.

Perhaps. 127.0.1.1 works for all "correctly"/intended programmed software. But still, the more I was reading, the more I am sure that adding the local IP instead (if it is static) indeed is the intended way. So adding as above the static IP check and a prompt should not harm. It should be made clear, that in question user should not apply it, but if the device indeed is always connected to the same network with single network adapter (and static IP) then it makes sense to apply it to /etc/hosts.

MichaIng, 2021-10-01:
I'm going to mark this as closed. The `/etc/hosts` file is meant to be used by the local host only, not by any external one. It makes sense that the local host uses the shortest way to connect to itself, when its own hostname is used, and the loopback IP range (127.*) is assured to be the shortest way. Using the physical network adapters IP instead in theory includes additional routing or filter rules being applied, and in any case it makes it more difficult to apply filter rules to or watch loopback requests/packets.

closed 2021-10-01 by MichaIng