MichaIng/DietPi #2661: FeatureRequest: https for PiHole

Gill-Bates, 2019-03-25:
Hello,

It would be cool if lighthttpd automatically gets an HTTPS certificate when Pihole is installed. Currently I have to create it manually and then edit the lighthttpd.conf or the external.conf manually.

Cheers.

MichaIng, 2019-03-25:
@Gill-Bates 
Many thanks for your request.

You can apply HTTPS to your webserver via `dietpi-letsencrypt` (after installing CertBot via `dietpi-software`) already. Pi-hole has not much to do with this and whether one needs HTTPS on it or not depends on how Pi-hole is used. So I think we should not automate this but leave it optional as above.

What we could do is:
When selecting any software title that uses a webserver, offer to select CertBot for HTTPS as well and start `dietpi-letsencrypt` directly from within the install script. Of course only if HTTPS is not yet applied for the chosen webserver.

Gill-Bates, 2019-03-28:
hmmm. does not work with local domains ...

https://letsencrypt.org/docs/certificates-for-localhost/

MichaIng, 2019-03-29:
@Gill-Bates 
If you use it within local network only, then HTTPS is somehow obsolete? You can create a self-signed certificate and apply it to the webserver, but since you are not trusted certificate authority, browsers will throw a warning, some will even deny to connect.

We had the idea to extend the feature set of `dietpi-letsencrypt`, respectively create a new extended script that allows creating and applying self-signed certificates as well, but this might take a while 😉: https://github.com/MichaIng/DietPi/issues/2363#issuecomment-449649984

MichaIng, 2020-01-01:
I mark this as closed, optional self-signed HTTPS is part of the issue linked above.

closed 2020-01-01 by MichaIng