MichaIng/DietPi #2662: FeatureRequest: Ramlog #2 vs Full Logrotate

Gill-Bates, 2019-03-25:
Hello,

I understand I can choose three different Ramlogs:

1.Log only in RAM
2. Log in RAM, write to disk hourly
3. Complete logging incl. logrotate

Why don't I have a logrotate at No. 2? This variant is apparently ideal to increase the lifetime of SSD/SD cards. I understand that No. 2 has the same amount of logs as full logging. So logrotate would make sense, wouldn't it?

Cheers.

MichaIng, 2019-03-25:
@Gill-Bates 
Many thanks for your request.

Note that the amount of logs between 2. and 3. is not the same:
- 3. installs `rsyslog` which creates syslog, daemon.log, cron.log and some other files inside /var/log. These can actually already be watched via `journalctl` (which also allows to make them boot-persistant by simply creating `mkdir /var/log/journal`), so there is not much point in having this IMO. Only benefit is that you can easier read those logs from other systems, although Windows does not support ext4 and other Linux (+systemd) OS can read the journald log file as well.
  - So actually IMO we should promote `systemd-journald`/`journalctl` to read system logs, optionally make them boot-persistant, and not use `rsyslog` anymore as being only an overhead on modern Debian.
- 2. does not contain system logs (besides DPKG, APT and of course software logs). But you are right that currently the log files on the disk grow unlimited. So it makes sense to clear/cut them by times. Using `logrotate` for this would be too complicated. We would need to add entries for every single file. But we could add an own additional log clear loop that e.g. cuts entries daily or weekly, based on e.g. user choice. I already have an idea by writing an identifier line to the files every chosen amount of time and removing all content until+including any existing line first. So when choosing 1 day persistent logs, the files fill up on day 1 and at the end the cleaner adds an identifier line. At the end of day 2 the cleaner finds this line and removes all logs until this, then adds a new identifier to be clean to on day 3. I hope you get the idea? 🤣
- 1. of course does not required any change since the logs (should) stay small when being cleaned every hour.

Gill-Bates, 2019-03-25:
Gotcha! Thanks for sharing your ideas with us

stormbard, 2024-01-02:
It should probably be noted in the documentation that filling the disk is a possibility for ramlog#2, I didn't see it there and just hit this issue causing some pretty gnarly issues.