MichaIng/DietPi #2663: FeatureRequest: Sending E-Mail for System Events

Gill-Bates, 2019-03-25:
Hello,
it would be cool if Dietpi sent critical system events by e-mail. Is there already a function for this? I would be happy to design the HTML mail templates for it.

UserStory: I store SMTP crendentials and Dietpi sends e-mails for the following events, for example:

- Disk corrupt
- reboot
- CPU hot
- RAM full
- etc...

MichaIng, 2019-03-25:
@Gill-Bates 
Many thanks for your request again.

Hmm this would require a new tool, e.g. `dietpi-system_monitor` which regularly checks these values and either logs them (optionally only in case of critical value) or send by email. However then we need an email client or MTA. We already though about such some time ago and found msmtp some time ago: https://packages.debian.org/de/stretch/msmtp-mta
Or even without fill MTA: https://packages.debian.org/de/stretch/msmtp

Just btw NetData also support such: https://docs.netdata.cloud/health/notifications/email/
Of course it might be an overkill and especially monitors many network related stats as well, but I think you could configure it to only show the desired data.

Gill-Bates, 2019-03-26:
Wow. Netdata would be an overkill. 
[Dietpi Template.zip](https://github.com/MichaIng/DietPi/files/3006798/Dietpi.Template.zip)

MichaIng, 2019-03-26:
@Gill-Bates 
Looks beautiful, thanks for this.

Lets see when I can do a start on the monitor script. However to raise priority, feel free to add this request to our FeatHub page: https://feathub.com/MichaIng/DietPi/

muibusan, 2019-05-09:
Here's the link to the said feature request:
[https://feathub.com/MichaIng/DietPi/+65](url)

MichaIng, 2021-03-08:
[ssmtp](https://packages.debian.org/ssmtp) seems to be the absolut minimal solution btw, but I'm not sure if it can be configured (easily) to send mails via modern mail providers/SMTP servers.

DietPi and available APT package updates could trigger an email as well. It should be somehow combined with the banner. E.g. the line which currently informs about DietPi or APT updates could be replaced with the events of a daily check which includes temperature/voltage/corruption warnings.

Gill-Bates, 2021-03-08:
> [ssmtp](https://packages.debian.org/ssmtp) seems to be the absolut minimal solution btw, but I'm not sure if it can be configured (easily) to send mails via modern mail providers/SMTP servers.

Should be possible: https://support.cloud.engineyard.com/hc/en-us/articles/205407478-Set-Up-SSMTP-for-Mail-Relay-to-AuthSMTP

dpantel, 2023-05-04:
@MichaIng FYI, ssmpt is unmaintained and your first instinct of using msmtp was correct. cheers.