MichaIng/DietPi #2664: Can't auto login from SSH client

UNiXMIT, 2019-03-25:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.22.3
- Distro version | 9.8
- Kernel version | Linux DietPi 3.10.105-bsp-1.2-ayufan-118 #1 SMP PREEMPT Sun Oct 15 21:40:57 UTC 2017 aarch64 GNU/Linux
- SBC device | Pine A64 (aarch64)

#### Additional Information (if applicable)
- Software title | DietPi
- Updated from 6.21 to 6.22.3

#### Steps to reproduce
I have reproduced this on various SSH clients but try Termius. You can save the username and password for the connection. Set the username and password and try to connect to DietPi. You will still be forced to input your password.

#### Expected behaviour
I expect to be logged in with the password saved in Termius like it was in 6.21

#### Actual behaviour
Instead of logging in i am prompted for a password.

#### Additional Info
Client - https://www.termius.com/

MichaIng, 2019-03-25:
@UNiXMIT 
Many thanks for your report.

Hmm that is strange, as any DietPi/login scripts do not affect the SSH login prompt but only apply after login has been done. So this should have something to do with the SSH server (Dropbear in question?).

I just tried with PuTTY and it works well here on testing machine (from Windows client cmd terminal):
`putty.exe -l root -pw dietpi 192.168.178.29`

My goodness Terminus is a monster with > 100M download size. I am currently not keen do download/install it 😅. With which other (lightweight/standalone) SSH client did you try it?

MichaIng, 2019-03-25:
@UNiXMIT 
Hehe I think I found the issue, respectively @xsak did: https://github.com/MichaIng/DietPi/issues/2667

Are you logging in as user that requires to enter a password when calling `sudo`?

Then run:
```
sed -i "/until sudo systemctl/c\		while pgrep -f '/DietPi/dietpi/postboot' &> /dev/null" /DietPi/dietpi/dietpi-login
```

UNiXMIT, 2019-03-25:
That’s right. I just ran the command and it’s fixed it. Good find.

closed 2019-03-25 by UNiXMIT

MichaIng, 2019-03-25:
@UNiXMIT 
Great, thanks for testing. ~I close this issue then in favour of the other one.~