MichaIng/DietPi #2665: Error while updating from v6.21.1 to v6.22.3 on Odroid C2

BerndKohl, 2019-03-25:
# dietpi-update
[  OK  ] DietPi-Update | Root access verified.
[  OK  ] DietPi-Update | RootFS R/W access verified.

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Checking for DietPi updates

[ INFO ] DietPi-Update | Checking mirror: https://raw.githubusercontent.com/Fourdee/DietPi/master/dietpi/server_version-6
[  OK  ] DietPi-Update | Using update server: https://raw.githubusercontent.com/Fourdee/DietPi/master/dietpi/server_version-6

[  OK  ] DietPi-Update | Update available
[ INFO ] DietPi-Update | Current version : v6.21.1
[ INFO ] DietPi-Update | Latest version  : v6.22.3
[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=61998 MB | required=100 MB
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : syncthing
[  OK  ] DietPi-Services | stop : plexmediaserver
[  OK  ] DietPi-Services | stop : plexpy
[  OK  ] DietPi-Services | stop : nginx
[  OK  ] DietPi-Update | wget https://raw.githubusercontent.com/Fourdee/DietPi/master/dietpi/pre-patch_file -O pre-patch_file

 DietPi-Pre-patch
─────────────────────────────────────────────────────
 Mode: Applying critical pre-patches

[ WARN ] Pre-patch 3 | Storing global DietPi-Software password with enhanced security
[  OK  ] Successfully applied critical pre-patches

[  OK  ] DietPi-Update | APT update, please wait...
Hit http://dev2day.de jessie InRelease
Ign http://ftp.debian.org jessie InRelease
Get:1 http://ftp.debian.org jessie-updates InRelease [145 kB]
Get:2 http://fuzon.co.uk all InRelease [16.6 kB]
Get:3 http://fuzon.co.uk jessie InRelease [5,630 B]
Hit http://dev2day.de jessie/main armhf Packages
Get:4 http://ftp.debian.org jessie-backports InRelease [135 kB]
Get:5 http://fuzon.co.uk all/c2 arm64 Packages [4,216 B]
Hit http://ftp.debian.org jessie Release.gpg
Hit http://fuzon.co.uk all/main arm64 Packages
Hit http://ftp.debian.org jessie Release
Hit http://fuzon.co.uk all/testing arm64 Packages
Hit http://fuzon.co.uk all/main armhf Packages
Hit http://fuzon.co.uk all/testing armhf Packages
Get:6 http://ftp.debian.org jessie-updates/main arm64 Packages/DiffIndex [12.3 kB]
Ign http://dev2day.de jessie/main Translation-en_GB
Ign http://dev2day.de jessie/main Translation-en
Hit http://ftp.debian.org jessie-updates/contrib arm64 Packages
Get:7 http://ftp.debian.org jessie-updates/non-free arm64 Packages/DiffIndex [1,228 B]
Get:8 http://ftp.debian.org jessie-updates/main armhf Packages/DiffIndex [12.3 kB]
Hit http://ftp.debian.org jessie-updates/contrib armhf Packages
Hit http://fuzon.co.uk jessie/backports arm64 Packages
Get:9 http://ftp.debian.org jessie-updates/non-free armhf Packages/DiffIndex [1,228 B]
Hit http://fuzon.co.uk jessie/main arm64 Packages
Hit http://ftp.debian.org jessie-updates/contrib Translation-en
Get:10 http://fuzon.co.uk jessie/backports armhf Packages [85.9 kB]
Get:11 http://ftp.debian.org jessie-updates/main Translation-en/DiffIndex [4,180 B]
Get:12 http://ftp.debian.org jessie-updates/non-free Translation-en/DiffIndex [1,228 B]
Get:13 http://fuzon.co.uk jessie/main armhf Packages [79.5 kB]
Get:14 http://ftp.debian.org jessie-updates/main arm64 2019-03-24-1411.39.pdiff [28 B]
Get:15 http://ftp.debian.org jessie-backports/main arm64 Packages [32 B]
Get:16 http://ftp.debian.org jessie-updates/main arm64 2019-03-24-1411.39.pdiff [28 B]
Get:17 http://ftp.debian.org jessie-backports/contrib arm64 Packages [32 B]
Ign http://fuzon.co.uk all/c2 Translation-en_GB
Ign http://fuzon.co.uk all/c2 Translation-en
Ign http://fuzon.co.uk all/main Translation-en_GB
Ign http://fuzon.co.uk all/main Translation-en
Ign http://fuzon.co.uk all/testing Translation-en_GB
[... 187 lines cut ...]
Err http://ftp.debian.org jessie/contrib arm64 Packages
  404  Not Found
Ign http://fuzon.co.uk all/testing Translation-en_GB
Err http://ftp.debian.org jessie/non-free arm64 Packages
  404  Not Found
Ign http://fuzon.co.uk all/testing Translation-en
Ign http://fuzon.co.uk jessie/backports Translation-en_GB
Ign http://fuzon.co.uk jessie/backports Translation-en
Ign http://fuzon.co.uk jessie/main Translation-en_GB
Ign http://fuzon.co.uk jessie/main Translation-en
Fetched 32.5 kB in 5s (6,461 B/s)
W: Failed to fetch http://ftp.debian.org/debian/dists/jessie/main/binary-arm64/Packages  404  Not Found

W: Failed to fetch http://ftp.debian.org/debian/dists/jessie/contrib/binary-arm64/Packages  404  Not Found

W: Failed to fetch http://ftp.debian.org/debian/dists/jessie/non-free/binary-arm64/Packages  404  Not Found

E: Some index files failed to download. They have been ignored, or old ones used instead.
```

--------------------------------------------------------------------

[FAILED] DietPi-Update | Unable to continue, DietPi-Update will now terminate.

root@four:~# D

MichaIng, 2019-03-25:
@BerndKohl 
Many thanks for your report.

Little hint:
- Please always wrap your terminal pasts into `code` format, so it is smaller, better readable, avoids line breaks and unwanted markdown formats and such 😉.

My goodness indeed the `binary-arm64` directory has been gone on the Jessie repo 👀: http://ftp.debian.org/debian/dists/jessie/main/
Compare with the content of the Stretch repo: http://ftp.debian.org/debian/dists/stretch/main/

This was already the case with the security repo some time ago, which is while we removed this on arm64 Jessie systems: http://security-cdn.debian.org/debian-security/dists/jessie/updates/main/
- See: https://github.com/MichaIng/DietPi/issues/1915
But here it's about the main repo...

Just checking the packages list website: https://packages.debian.org/de/jessie/apt
- Even for system base packages there are no listed arm64 variants anymore....

I am not sure if this is a current/temporary issue while syncing around with Buster release and such of if indeed Debian drops support for Jessie and arm64 completely now which would be veery strange. Yesterday I was actually trying to install a fresh Jessie test image via Jessie installer on VM and couldn't find it: https://cdn-aws.deb.debian.org/debian/dists/jessie/main/debian-installer/binary-amd64/ Needed to use the Stretch installer instead...

The download links for Debian Jessie arm64 images are dead as well: https://www.debian.org/releases/jessie/debian-installer/

Okay I couldn't find any new about this, similar to missing news about dropped Jessue arm64 security repo.

Could you please try the following?
```
sed -i '/ftp.debian.org/d' /etc/apt/sources.list
cat << _EOF_ >> /etc/apt/sources.list
deb [arch=all] http://deb.debian.org/debian/ jessie main contrib non-free
deb http://deb.debian.org/debian/ jessie-updates main contrib non-free
deb [arch=all] http://deb.debian.org/debian-security/ jessie/updates main contrib non-free
deb http://deb.debian.org/debian/ jessie-backports main contrib non-free
_EOF_
```
This adds the repo the the list in a way that hopefully only the `debian-all` sub dir is searched for packages. Also for testing the security repo is re-added this way. The jessie-updates jessie-backports still has an debian-arm64 sub dir.

MichaIng, 2019-03-25:
Just as a general note. I would recommend that you install a fresh DietPi Stretch image by times, since Jessie support (not only our side) is fading. The Kodi/GPU acceleration issues have been solved meanwhile, if this was an issue.

BerndKohl, 2019-03-25:
`# sed -i '/ftp.debian.org/d' /etc/apt/sources.list
cat << _EOF_ >> /etc/apt/sources.list
> deb [arch=all] http://deb.debian.org/debian/ jessie main contrib non-free
> deb http://deb.debian.org/debian/ jessie-updates main contrib non-free
> deb [arch=all] http://deb.debian.org/debian-security/ [...]
root@four:~# l
drwxr-xr-x   - root 17 Jun  2018 exa
.rwxr-xr-x 327 root 20 Sep  2018 start_network.sh
drwxr-xr-x   - root 10 Feb 19:07 Sync
root@four:~# dietpi-update 
[  OK  ] DietPi-Update | Root access verified.
[  OK  ] DietPi-Update | RootFS R/W access verified.

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Checking for DietPi updates

[ INFO ] DietPi-Update | Checking mirror: https://raw.githubusercontent.com/Fourdee/DietPi/master/dietpi/server_version-6
[  OK  ] DietPi-Update | Using update server: https://raw.githubusercontent.com/Fourdee/DietPi/master/dietpi/server_version-6

[  OK  ] DietPi-Update | Update available
[ INFO ] DietPi-Update | Current version : v6.21.1
[ INFO ] DietPi-Update | Latest version  : v6.22.3
[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=62015 MB | required=100 MB
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : syncthing
[  OK  ] DietPi-Services | stop : plexmediaserver
[  OK  ] DietPi-Services | stop : plexpy
[  OK  ] DietPi-Services | stop : nginx
[  OK  ] DietPi-Update | wget https://raw.githubusercontent.com/Fourdee/DietPi/master/dietpi/pre-patch_file -O pre-patch_file

 DietPi-Pre-patch
─────────────────────────────────────────────────────
 Mode: Applying critical pre-patches

[  OK  ] Successfully applied critical pre-patches

[  OK  ] DietPi-Update | APT update, please wait...
Hit http://dev2day.de jessie InRelease
Hit http://fuzon.co.uk all InRelease
Hit http://fuzon.co.uk jessie InRelease
Ign http://deb.debian.org jessie InRelease
Get:1 http://deb.debian.org jessie-updates InRelease [145 kB]
Hit http://dev2day.de jessie/main armhf Packages
Get:2 http://deb.debian.org jessie/updates InRelease [44.9 kB]
Ign http://deb.debian.org jessie-backports InRelease
Get:3 http://deb.debian.org jessie Release.gpg [2,420 B]
Ign http://deb.debian.org jessie-backports Release.gpg
Get:4 http://deb.debian.org jessie Release [148 kB]
Ign http://dev2day.de jessie/main Translation-en_GB
Ign http://dev2day.de jessie/main Translation-en
Ign http://deb.debian.org jessie-backports Release
Hit http://fuzon.co.uk all/c2 arm64 Packages
Hit http://fuzon.co.uk all/main arm64 Packages
Hit http://fuzon.co.uk all/testing arm64 Packages
Hit http://fuzon.co.uk all/main armhf Packages
Hit http://fuzon.co.uk all/testing armhf Packages
Get:5 http://deb.debian.org jessie-updates/contrib arm64 Packages [32 B]
Hit http://fuzon.co.uk jessie/backports arm64 Packages
Hit http://fuzon.co.uk jessie/main arm64 Packages
Hit http://fuzon.co.uk jessie/backports armhf Packages
Hit http://fuzon.co.uk jessie/main armhf Packages
Get:6 http://deb.debian.org jessie-updates/contrib armhf Packages [32 B]
Get:7 http://deb.debian.org jessie-updates/contrib Translation-en [14 B]
Get:8 http://deb.debian.org jessie-updates/main Translation-en [14 B]
Get:9 http://deb.debian.org jessie-updates/non-free Translation-en [14 B]
Get:10 http://deb.debian.org jessie/updates/main all Packages [290 kB]
Get:11 http://deb.debian.org jessie/updates/contrib all Packages [1,138 B]
[... 133 lines cut ...]
Ign http://deb.debian.org jessie-backports/contrib Translation-en
Ign http://deb.debian.org jessie-backports/main Translation-en_GB
Ign http://deb.debian.org jessie-backports/main Translation-en
Ign http://deb.debian.org jessie-backports/non-free Translation-en_GB
Ign http://deb.debian.org jessie-backports/non-free Translation-en
Fetched 8,785 kB in 9s (901 kB/s)
W: Failed to fetch http://deb.debian.org/debian/dists/jessie-backports/main/binary-arm64/Packages  404  Not Found

W: Failed to fetch http://deb.debian.org/debian/dists/jessie-backports/contrib/binary-arm64/Packages  404  Not Found

W: Failed to fetch http://deb.debian.org/debian/dists/jessie-backports/non-free/binary-arm64/Packages  404  Not Found

W: Failed to fetch http://deb.debian.org/debian/dists/jessie-backports/main/binary-armhf/Packages  404  Not Found

W: Failed to fetch http://deb.debian.org/debian/dists/jessie-backports/contrib/binary-armhf/Packages  404  Not Found

W: Failed to fetch http://deb.debian.org/debian/dists/jessie-backports/non-free/binary-armhf/Packages  404  Not Found

E: Some index files failed to download. They have been ignored, or old ones used instead.
```

--------------------------------------------------------------------

[FAILED] DietPi-Update | Unable to continue, DietPi-Update will now terminate.
`

BerndKohl, 2019-03-25:
> Just as a general note. I would recommend that you install a fresh DietPi Stretch image by times, since Jessie support (not only our side) is fading. The Kodi/GPU acceleration issues have been solved meanwhile, if this was an issue.

Hm, I would give it a try but I am nut sure how to preserve/transfer my Plex media data...

MichaIng, 2019-03-25:
Lol indeed the jessie-backports dir not missing but entering it fails:
- https://cdn-aws.deb.debian.org/debian/dists/ `jessie-backports` exists
- But: https://cdn-aws.deb.debian.org/debian/dists/jessie-backports/ file not found...

My goodness what is Debian doing there.... Especially since I think I just checked the arm64 sub dir inside just 5 minutes ago...

Ääämm lol and the ftp.debian.org mirror now does not have this directory anymore: http://ftp.debian.org/debian/dists/
It looks that while we are writing here, several Jessie APT parts are removed from the repo...

However to to remove the backports repo for now:
`sed -i '/jessie-backports/d' /etc/apt/sources.list`

BerndKohl, 2019-03-25:
So I think we got e few steps further...

`dietpi-update 
[  OK  ] DietPi-Update | Root access verified.
[  OK  ] DietPi-Update | RootFS R/W access verified.

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Checking for DietPi updates

[ INFO ] DietPi-Update | Checking mirror: https://raw.githubusercontent.com/Fourdee/DietPi/master/dietpi/server_version-6
[  OK  ] DietPi-Update | Using update server: https://raw.githubusercontent.com/Fourdee/DietPi/master/dietpi/server_version-6

[  OK  ] DietPi-Update | Update available
[ INFO ] DietPi-Update | Current version : v6.21.1
[ INFO ] DietPi-Update | Latest version  : v6.22.3
[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=61967 MB | required=100 MB
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : syncthing
[  OK  ] DietPi-Services | stop : plexmediaserver
[  OK  ] DietPi-Services | stop : plexpy
[  OK  ] DietPi-Services | stop : nginx
[  OK  ] DietPi-Update | wget https://raw.githubusercontent.com/Fourdee/DietPi/master/dietpi/pre-patch_file -O pre-patch_file

 DietPi-Pre-patch
─────────────────────────────────────────────────────
 Mode: Applying critical pre-patches

[  OK  ] Successfully applied critical pre-patches

[  OK  ] DietPi-Update | APT update, please wait...
Hit http://dev2day.de jessie InRelease
Hit http://fuzon.co.uk all InRelease
Hit http://fuzon.co.uk jessie InRelease
Ign http://deb.debian.org jessie InRelease
Hit http://deb.debian.org jessie-updates InRelease
Hit http://dev2day.de jessie/main armhf Packages
Hit http://deb.debian.org jessie/updates InRelease
Hit http://deb.debian.org jessie Release.gpg
Hit http://deb.debian.org jessie Release
Ign http://dev2day.de jessie/main Translation-en_GB
Ign http://dev2day.de jessie/main Translation-en
Get:1 http://deb.debian.org jessie-updates/main arm64 Packages/DiffIndex [12.3 kB]
Get:2 http://deb.debian.org jessie-updates/contrib arm64 Packages [32 B]
Get:3 http://deb.debian.org jessie-updates/non-free arm64 Packages/DiffIndex [1,228 B]
Get:4 http://deb.debian.org jessie-updates/main armhf Packages/DiffIndex [12.3 kB]
Get:5 http://deb.debian.org jessie-updates/contrib armhf Packages [32 B]
Get:6 http://deb.debian.org jessie-updates/non-free armhf Packages/DiffIndex [1,228 B]
Get:7 http://deb.debian.org jessie-updates/contrib Translation-en [14 B]
Get:8 http://deb.debian.org jessie-updates/main Translation-en/DiffIndex [4,180 B]
Get:9 http://deb.debian.org jessie-updates/non-free Translation-en/DiffIndex [1,228 B]
Get:10 http://deb.debian.org jessie/updates/main all Packages [290 kB]
Hit http://fuzon.co.uk all/c2 arm64 Packages
Hit http://fuzon.co.uk all/main arm64 Packages
Get:11 http://deb.debian.org jessie/updates/contrib all Packages [1,138 B]
Hit http://fuzon.co.uk all/testing arm64 Packages
Get:12 http://deb.debian.org jessie/updates/non-free all Packages [4,260 B]
Hit http://fuzon.co.uk all/main armhf Packages
Get:13 http://deb.debian.org jessie/updates/contrib Translation-en [1,211 B]
Hit http://fuzon.co.uk all/testing armhf Packages
Get:14 http://deb.debian.org jessie/updates/main Translation-en [334 kB]
Get:15 http://deb.debian.org jessie/updates/non-free Translation-en [11.8 kB]
Hit http://fuzon.co.uk jessie/backports arm64 Packages
Hit http://fuzon.co.uk jessie/main arm64 Packages
Hit http://fuzon.co.uk jessie/backports armhf Packages
Get:16 http://deb.debian.org jessie/main all Packages [3,039 kB]
Hit http://fuzon.co.uk jessie/main armhf Packages
Get:17 http://deb.debian.org jessie/contrib all Packages [23.8 kB]
[... 87 lines cut ...]
100   120    0   120    0     0    167      0 --:--:-- --:--:-- --:--:--   167
100 2502k  100 2502k    0     0  1056k      0  0:00:02  0:00:02 --:--:-- 2690k
[  OK  ] DietPi-Update | Unpack update archive
rm: cannot remove ‘DietPi-master/dietpi/.??*’: No such file or directory
[  OK  ] DietPi-Update | Copy DietPi core files to RAMdisk
[  OK  ] DietPi-Update | Copy DietPi rootfs files in place
[  OK  ] DietPi-Update | Set execute permissions for DietPi scripts
[  OK  ] DietPi-Set_software | Connection test: https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt
[  OK  ] DietPi-Set_software | Patching dietpi.txt
[  OK  ] DietPi-Set_software | Downloading current dietpi.txt
[  OK  ] DietPi-Set_software | dietpi.txt verification completed
[  OK  ] verify_dietpi.txt  | Completed

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Running incremental patching

[ INFO ] DietPi-Update | Current version : v6.21.1
[ INFO ] DietPi-Update | Latest version  : v6.22.3
[ INFO ] DietPi-Update | Patching 6.21.1 to 6.22
[  OK  ] DietPi-Patchfile | Added setting for i in /etc/bashrc.d/*.sh /etc/bashrc.d/*.bash; do [ -r "$i" ] && . $i; done to end of file /etc/bash.bashrc
[ INFO ] DietPi-Patchfile | APT installation for: p7zip, please wait...
E: Package 'p7zip' has no installation candidate
[FAILED] DietPi-Patchfile | G_AGI p7zip
`

MichaIng, 2019-03-25:
@BerndKohl 
Yeah damn, the `p7zip` is not available in the remaining repos...

Hmm that is bad. You cannot even skip the p7zip install since on Jessie (different than on Stretch+) the already installed p7zip-full package does not include p7zip as dependency. I just had a look if a manual install is possible, but it fails due to one dependency library that is too old on Jessie `libstdc++6 (>= 5)`: https://packages.debian.org/de/stretch/p7zip

And most likely if some reinstall is triggered e.g. then there will be many more affected packages.

You know what, I would really install Stretch now, if you find some time. It looks like attempting to go on with Jessie will only result in further issues even on simply tasks if you want to install any APT package.
Do the following:
- Do/update a full system backup onto an external drive/USB dongle via: `dietpi-backup`
- Install our Odroid C2 image: https://dietpi.com/downloads/images/DietPi_OdroidC2-ARMv8-Stretch.7z
- Skip installing additional software at first.
- Copy the `mnt/dietpi_userdata` and `var/lib/plexmediaserver/` from the backup onto your system to have e.g. media files and Plex config files back again.
- Copy any other settings files or software data that you manually configured from the backup as well. Simply keep the backup for a while so if anything is missing you can always recover it.

You could also wait a day and see if perhaps the repo reappears, and/or if that is too much hassle for now 😉.

Alternative would be to do an in-situ distro upgrade, but that might produce follow up errors and create a greater overall effort over time...

vielmetti, 2019-03-26:
Looks like the announcement was here:

https://twitter.com/debian/status/1109080168318926851

https://lists.debian.org/debian-devel-announce/2019/03/msg00006.html

```
Hello,

as Wheezy and Jessie have been integrated into the archive.debian.org
structure recently, we are now removing all of Wheezy and all non-LTS
architectures of Jessie from the mirror network starting today.
That is, only Jessie i386, amd64, armel and armhf will continue to be
hosted on the normal mirrors.

The data is, of course, not lost - the whole of it is synced
to archive.debian.org, so if you still need it you will be able to get
it from there.[1]

[1] Keep in mind that they have no security support at all!

--
bye, Joerg
The sun? That’s the hottest place on Earth.
```

MichaIng, 2019-03-26:
@vielmetti 
Perfect, many thanks for this info!

@BerndKohl 
So this means the following `sources.list` will still work:
```
cat << _EOF_ > /etc/apt/sources.list
deb http://archive.debian.org/debian/ jessie main contrib non-free
deb http://deb.debian.org/debian/ jessie-updates main contrib non-free
deb [arch=all] http://deb.debian.org/debian-security/ jessie/updates main contrib non-free
deb http://archive.debian.org/debian/ jessie-backports main contrib non-free
_EOF_
```
- Security repo only for `[arch=all]` packages, `jessie-updates` are still on main repo, but might be removed soon as well.

The recommendation to switch to Stretch when you can stays correct 😉.

BerndKohl, 2019-03-26:
Sorry, but what about the last line complaining in an error about 7Zip?

` DietPi-Update
─────────────────────────────────────────────────────
 Mode: Running incremental patching

[ INFO ] DietPi-Update | Current version : v6.21.1
[ INFO ] DietPi-Update | Latest version  : v6.22.3
[ INFO ] DietPi-Update | Patching 6.21.1 to 6.22
[  OK  ] DietPi-Patchfile | Added setting for i in /etc/bashrc.d/*.sh /etc/bashrc.d/*.bash; do [ -r "$i" ] && . $i; done to end of file /etc/bash.bashrc
[ INFO ] DietPi-Patchfile | APT installation for: p7zip, please wait...
E: Package 'p7zip' has no installation candidate
[FAILED] DietPi-Patchfile | G_AGI p7zip`

vielmetti, 2019-03-26:
p7zip is not available for jessie, per https://packages.debian.org/search?arch=arm64&keywords=p7zip-full

A relevant bug is https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=824160 corresponding to CVE-2016-2334 CVE-2016-2335.

MichaIng, 2019-03-26:
@vielmetti 
As you filtered by arch=arm64 no package is available for the main repo anymore. But generally it is available for Jessie: https://packages.debian.org/jessie/p7zip

Will check the archive repo, which should have it actually:
```
Package: p7zip
Version: 9.20.1~dfsg.1-4.1+deb8u3
Installed-Size: 883
Maintainer: Mohammed Adnène Trojette <adn+deb@diwi.org>
Architecture: arm64
Depends: libc6 (>= 2.17), libgcc1 (>= 1:4.7), libstdc++6 (>= 4.1.1)
Suggests: p7zip-full
Description: 7z file archiver with high compression ratio
Homepage: http://p7zip.sourceforge.net/
Description-md5: 8f7261a1031bb31dfcb3cba829894130
Tag: admin::file-distribution, implemented-in::c++, interface::commandline,
 role::program, scope::utility, use::compressing, works-with::archive,
 works-with::file
Section: utils
Priority: optional
Filename: pool/main/p/p7zip/p7zip_9.20.1~dfsg.1-4.1+deb8u3_arm64.deb
Size: 289888
MD5sum: 760b073f5b9cb85d60bdecdf83b3e250
SHA1: 3bccf1037fb0236df89da35a1921f8af6a326d23
SHA256: 3105f811674e374436a341041ba0745fb8900a44c19f0e876b06afd3c4b30b60
```
- From: http://archive.debian.org/debian/dists/jessie/main/binary-arm64/

Manual download works as well: http://archive.debian.org/debian/pool/main/p/p7zip/p7zip_9.20.1~dfsg.1-4.1+deb8u3_arm64.deb

@BerndKohl 
Could you verify:
```
cat /etc/apt/sources.list
apt update
apt -y install p7zip
```

BerndKohl, 2019-03-27:
Guys, I think I made it.

Had to update `sources.list` as you proposed and use `--force-yes` when installing p7zip but now it looks like everything is fine.

Anyway I will try to move to a fresh install using the latest image on the website (hoping to get Stretch).

Thank you all for your support.

MichaIng, 2019-03-27:
@BerndKohl 
Hmm if `--force-yes` was required, then perhaps the archive repo has by default a lower priority. Although strange since I thought the priorities are just about the disto, which is "jessie"/"stable" here as well.

Could you please paste:
`apt-cache policy p7zip-full`

I will reopen the issue since we need to do these repo changes for all Jessie ARMv8 devices with next DietPi update.

BerndKohl, 2019-03-27:
`apt-cache policy p7zip-full`

`p7zip-full:`
`  Installed: 9.20.1~dfsg.1-4.1+deb8u3`
`  Candidate: 9.20.1~dfsg.1-4.1+deb8u3`
`  Version table:`
` *** 9.20.1~dfsg.1-4.1+deb8u3 0`
`        500 http://archive.debian.org/debian/ jessie/main arm64 Packages`
`        100 /var/lib/dpkg/status`

MichaIng, 2019-03-27:
@BerndKohl 
Ah you have `p7zip-full` still installed? At least DietPi scripts do not require it anymore but yeah I remember we left the choice to user to keep it or not.

I think the priority is forced to 500 if a package is already installed.
However I just tested on VM and also for non-installed packages from archive repo it is 500 🤔:
```
root@VM-Jessie:~# apt-cache policy python-dev
python-dev:
  Installed: (none)
  Candidate: 2.7.9-1
  Version table:
     2.7.9-1 0
        500 http://archive.debian.org/debian/ jessie/main amd64 Packages
```

I just found another issue when attempting to update from `jessie-backports` branch. APT denies to use it since it is not "valid" anymore: http://archive.debian.org/debian/dists/jessie-backports/Release
> Valid-Until: Tue, 19 Feb 2019 20:25:43 UTC

For the security repo it's the same soon: http://security.debian.org/debian-security/dists/jessie/updates/Release
> Valid-Until: Sat, 06 Apr 2019 16:32:28 UTC

`jessie-updates` have been removed completely:
- https://deb.debian.org/debian/dists/
- http://archive.debian.org/debian/dists/

So finally Jessie systems can only use the single entry:
`deb http://deb.debian.org/debian/ jessie main contrib non-free`
And Jessie ARMv8 systems only:
`deb http://archive.debian.org/debian/ jessie main contrib non-free`

But aside from that, installing from archive repo didn't require `--force-yes` in my case 🤔.

BerndKohl, 2019-03-30:
Guys, I "migrated" to Stretch and everything is fine so far.
Just to let you know...

MichaIng, 2019-03-30:
@BerndKohl 
Great, thanks for reporting.

I reopen the issue as I still need to implement the sources.list adjustment on Jessie systems when updating to v6.23 😉.

MichaIng, 2019-04-22:
PR up to apply APT list changes: https://github.com/MichaIng/DietPi/pull/2728

closed 2019-05-12 by MichaIng