MichaIng/DietPi #2666: Dietpi Update Failure

kalexakis, 2019-03-25:
#### Details:
- Date           | Mon 25 Mar 22:31:52 EET 2019
- Bug report     | N/A
- DietPi version | v6.19.7 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Kernel version | #1159 SMP Sun Nov 4 17:50:20 GMT 2018
- Distro         | stretch (index=4)
- Command        | chmod -R +x /DietPi /var/lib/dietpi/services /etc/cron.daily/dietpi /etc/cron.hourly/dietpi /etc/profile.d/dietpi-*.sh /etc/bashrc.d/dietpi-bash_completion.sh /etc/bashrc.d/dietpi-login.sh /etc/bashrc.d/dietpi-posix-set-locale.sh /etc/bashrc.d/dietpi-ps1_kill_animation.sh /etc/bashrc.d/dietpi-unsupported_terminal.sh
- Exit code      | 1
- Software title | DietPi-Update

#### Steps to reproduce:

1. dietpi-update
2. ...

#### Expected behaviour:

- Update from v6.19.7 to current v6.22.3

#### Actual behaviour:

- Error message

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
chmod: cannot access '/etc/profile.d/dietpi-*.sh': No such file or directory
```

MichaIng, 2019-03-25:
@kalexakis 
Many thanks for your report.

Jep this is a known issue if you once attempted unsuccessfully to update to v6.20.

Please do the following:
```
> /etc/profile.d/dietpi-unsupported_terminal.sh
> /DietPi/.update_stage
echo 1 > /DietPi/.install_stage
dietpi-update 1
```

kalexakis, 2019-03-26:
Thank you for your help. Issue is partially resolved. I am now being asked to set new password for root and dietpi user and reinstall packages as if this was a clean installation.

MichaIng, 2019-03-26:
@kalexakis 
Hum, with `> /DietPi/dietpi/.update_stage` (creating this file) it should have been actually handled correctly. If this is found by the updater, the install stage should be updated to match the new system (value 2).

Could you check if it still exists somehow?
```
ls -l /DietPi/dietpi/.update_stage
```

You could simply exit dietpi-software prompt after login and override the install stage with the correct value:
```
echo 2 > /DietPi/dietpi/.install_stage
```

But I would prefer to redo the steps to assure all flags were set correctly:
```
> /etc/profile.d/dietpi-unsupported_terminal.sh
> /DietPi/dietpi/.update_stage
# Verify the above file really exists
echo 1 > /DietPi/dietpi/.install_stage
# Re-run the last update. It should go through quickly. Do not reboot when asked.
dietpi-update -1
```
- When update has finished (and you skipped reboot, if asked), verify the new install stage has been applied:
```
cat /DietPi/dietpi/.install_stage # should show "2"
ls -l /DietPi/dietpi/.update_stage # should not exist anymore
```

kalexakis, 2019-03-27:
Issue resolved, following your latest instructions!
Much appreciated.

closed 2019-03-27 by kalexakis