MichaIng/DietPi #2667: dietpi-postboot asks for password (sudo) after update to 6.22.3

xsak, 2019-03-25:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.22.3
- Distro version | stretch
- Kernel version | Linux raspi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux
- SBC device | RPi 2 Model B (armv7l)
- Power supply used | 5V 1A RAVpower
- SDcard used | SanDisk

#### Additional Information (if applicable)
- Software title | dietpi-postboot
- The system was just updated to 6.22.3 (from 6.21.x)
- Can this issue be replicated on a fresh installation of DietPi?
- `dietpi-bugreport` ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce

1. update to 6.22.3
2. login with user via ssh with passwordless keys method
3. sudo password asked to executes command: `sudo systemctl -ln 0 --no-pager status dietpi-postboot` and user should enter their own password...

#### Expected behaviour
- Login without asking for sudo password

#### Actual behaviour
- Asks for sudo password

#### Extra details
- I log with my user to the freshly updated dietpi v6.22.3
- It asks for password, but I do not enter it, meanwhile...
- On another ssh terminal I login to the machine (it asks for sudo password and I enter it here)
- Checking the suspect program with `ps -ef | grep sudo`
- The result is: "sudo systemctl -ln 0 --no-pager status dietpi-postboot"
- Workaround: Edit /etc/sudoers with command `visudo`:
- Add line: "%sudo   ALL=(ALL:ALL) NOPASSWD: /bin/systemctl -ln 0 --no-pager status dietpi-postboot"

MichaIng, 2019-03-25:
@xsak 
Many thanks for your report.

So you didn't add your user to the sudoers list in a way that it can run sudo without password, right?

Hmm yeah indeed we should allow login even as non-sudo user without such prompt and/or error. `sudo ALL=(ALL:ALL) NOPASSWD:` requires at least sudo permissions. I think `pgrep 'dietpi-postboot'` should work instead without root/sudo permissions.

Just as background:
- We want to hold our login script until dietpi-postboot has finished which is the boot systemd unit that starts user services (webservers, media servers, databases and such) at the end of the boot process.
- This is since the login script in case starts Kodi, Chromium, a desktop or such, if chosen as autostart option. And in some cases related background services need to be started for this to succeed.
- We added `systemctl status dietpi-postboot` to check whether the service has existed already or not yet. But `systemctl` requires root permissions.
- Yeah but adding `sudo` is no great solution as login should work without prompt/error with any kind of user. I think replacing that with `pgrep 'dietpi-postboot'` should be the even quicker and simpler check.

MichaIng, 2019-03-25:
This would even cause an endless loop on non-sudoers logins (until cancelled), since the check returns "false" if the command fails due to missing permissions.

closed 2019-05-12 by MichaIng