MichaIng/DietPi #2669: Failed to Update v6.22.3

jmtramos, 2019-03-26:
# Hello when I try to update it gives me s bunch of errors.

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Mon Mar 25 21:46:12 2019 from 192.168.1.89-bash: /usr/bin/nproc: cannot execute binary file: Exec format error
/DietPi/dietpi/func/dietpi-globals: line 105: /usr/bin/nproc: cannot execute binary file: Exec format error
/DietPi/dietpi/func/dietpi-globals: line 105: /usr/bin/nproc: cannot execute binary file: Exec format error
/DietPi/dietpi/func/dietpi-globals: line 105: /usr/bin/nproc: cannot execute binary file: Exec format error
 ────────────────────────────────────────────────
 DietPi | Update available
 ────────────────────────────────────────────────
 v6.21.1 | RPi 3 Model B (armv7l)
 ────────────────────────────────────────────────
 eth0 | 192.168.1.65
 ────────────────────────────────────────────────
 DietPi Team     : Daniel Knight (founder), MichaIng, K-Plan Image           : DietPi Core Team (pre-image: Raspbian Lite)
 Web             : https://DietPi.com | https://twitter.com/dietpi_ Patreon Legends : PINE64 community
 Donate          : https://DietPi.com/#donate
 DietPi Hosting  : Powered by https://MyVirtualServer.com
DietPi-Update | RPi 3 Model B (armv7l) | IP: 192.168.1.65                                                ┌─────────────────────┤ DietPi-Update ├─────────────────────┐      │                                                           │      │                                                           │      │ ------------------------Notice------------------------    │      │ - A benefit of DietPi is: We use standard Linux (Debian)  │      │ configurations and commands.                              │      │ - A potential downside is: We can't possibly accommodate  │      │ or predict all modification to Linux configurations files │      │ by the end user, outside of DietPi programs, during       │      │ updates.                                                  │      │                                                           │      │ Although we test the updates thoroughly, if you have made │      │ any custom changes to Linux configuration files outside   │      │ of the DietPi programs, an update may trigger a potential │      │ issue.                                                    │      │ ------------------------------------------------------    │      │                                                           │      │ Do you wish to continue and update DietPi to v6.22.3?     │      │                                                           │      │               <Ok>                   <Cancel>             │      │                                                           │      └───────────────────────────────────────────────────────────┘                                                  /DietPi/dietpi/func/dietpi-globals: line 105: /usr/bin/nproc: cannot execute binary file: Exec format error
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : couchpotato
[  OK  ] DietPi-Services | stop : medusa
[  OK  ] DietPi-Services | stop : qbittorrent
[  OK  ] DietPi-Services | stop : avahi-daemon
[  OK  ] DietPi-Update | wget https://raw.githubusercontent.com/Fourdee/DietPi/master/dietpi/pre-patch_file -O pre-patch_file

 DietPi-Pre-patch
─────────────────────────────────────────────────────
 Mode: Applying critical pre-patches                  
[  OK  ] Successfully applied critical pre-patches

[  OK  ] DietPi-Update | APT update, please wait...
Get:1 http://mirror.ox.ac.uk/sites/archive.raspbian.org/archive/raspbian stretch InRelease [15.0 kB]        Hit:2 https://archive.raspberrypi.org/debian stretch InRelease
Fetched 15.0 kB in 1s (8,398 B/s)                     Reading package lists...                              [  OK  ] DietPi-Update | G_AGUP                       [  OK  ] DietPi-Update | APT upgrade, please wait...  Reading package lists...                              Building dependency tree...
Reading state information...
Calculating upgrade...
Unquoted string "remote" may clash with future reserved word at /usr/share/perl/5.24/Symbol.pm line 1.
debconf: Perl may be unconfigured (Bad name after n' at /usr/share/perl/5.24/Symbol.pm line 1.
Compilation failed in require at /usr/lib/arm-linux-gnueabihf/perl/5.24/IO/File.pm line 9.
BEGIN failed--compilation aborted at /usr/lib/arm-linux-gnueabihf/perl/5.24/IO/File.pm line 9.
Compilation failed in require at /usr/share/perl/5.24/FileHandle.pm line 9.
Compilation failed in require at (eval 1) line 3.
BEGIN failed--compilation aborted at (eval 1) line 3.

──────┤ DietPi Error Handler: ├───────────────────────
 problems prevent configuration of nfs-common:^M                             nds on rpcbind; however:^M     
d is not configured yet.^M                            
                       ──────┤ DietPi Error Handler: ├───────────────────────
mpilation aborted at /usr/sbin/update-rc.d line 73.^M 
essing package rpcbind (--configure):^M               ──────┤ DietPi Error Handler: ├───────────────────────mpilation aborted at /usr/sbin/update-rc.d line 73.^M essing package rpcbind (--configure):^M               alled post-installation script returned error exit sta problems prevent configuration of nfs-common:^M      nds on rpcbind; however:^M                            d is not configured yet.^M                            essing package nfs-common (--configure):^M            lems - leaving unconfigured^M                         untered while processing:^M
──────┤ DietPi Error Handler: ├───────────────────────mpilation aborted at /usr/sbin/update-rc.d line 73.^M essing package rpcbind (--configure):^M               alled post-installation script returned error exit sta problems prevent configuration of nfs-common:^M      nds on rpcbind; however:^M                            d is not configured yet.^M                            essing package nfs-common (--configure):^M            lems - leaving unconfigured^M                         untered while processing:^M                           usr/bin/dpkg returned an error code (1)                    Retry               : Re-run the last co              DietPi-Config       : Edit network, APT/                <Ok>                            <Exit>         ──────────────────────────────────────────────────────
--------------------------------------------------------------------
- DietPi has encounted an error, and, is unable to continue        -
- Please create a ticket: https://github.com/Fourdee/DietPi/issues -
- Copy and paste the BLUE lines below, into the ticket             -
--------------------------------------------------------------------

#### Details:
- Date           | Tue 26 Mar 09:58:19 GMT 2019
- Bug report     | N/A
- DietPi version | v6.21.1 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (index=3)
[... 215 lines cut ...]
/DietPi/dietpi/func/dietpi-globals: line 105: /usr/bin/nproc: cannot execute binary file: Exec format error
[  OK  ] DietPi-Update | Root access verified.
[  OK  ] DietPi-Update | RootFS R/W access verified.

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Checking for DietPi updates

[ INFO ] DietPi-Update | Checking mirror: https://raw.githubusercontent.com/Fourdee/DietPi/master/dietpi/server_version-6
[  OK  ] DietPi-Update | Using update server: https://raw.githubusercontent.com/Fourdee/DietPi/master/dietpi/server_version-6

[  OK  ] DietPi-Update | Update available
[ INFO ] DietPi-Update | Current version : v6.21.1
[ INFO ] DietPi-Update | Latest version  : v6.22.3
[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=18689 MB | required=100 MB
[  OK  ] DietPi-Update | Update is being applied, please wait...
/DietPi/dietpi/func/dietpi-globals: line 105: /usr/bin/nproc: cannot execute binary file: Exec format error
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : couchpotato
[  OK  ] DietPi-Services | stop : medusa
[  OK  ] DietPi-Services | stop : qbittorrent
[  OK  ] DietPi-Services | stop : avahi-daemon
[ .... ] DietPi-Update | wget https://raw.

MichaIng, 2019-03-26:
@jmtramos
Many thanks for your report.
Little hint: please always wrap your terminal pastes into code format to make out better readable, like the additional logs part of the error handler output.

I will come back to you when I am at home. Basically we need to go through the single APT packages that failed to install.

jmtramos, 2019-03-26:
> @jmtramos
> Many thanks for your report.
> Little hint: please always wrap your terminal pastes into code format to make out better readable, like the additional logs part of the error handler output. [...]

Hi MichaIng,

Thanks for the reply, sorry about the messy post (noob)! I will post correctly next time.

MichaIng, 2019-03-26:
@jmtramos 
> Thanks for the reply, sorry about the messy post (noob)!

No worries, I cleaned it a bid. Actually the middle part between the blue lines from the terminal are usually enough to paste here.

#### However interesting is as well:
```
/DietPi/dietpi/func/dietpi-globals: line 105: /usr/bin/nproc: cannot execute binary file: Exec format error
```
- This `Exec format error` usually indicates a wrong architecture, however most properly as of the other errors it is due to SDcard corruption.

#### Then it looks like the RPi repo is missing:
```
Get:1 http://mirror.ox.ac.uk/sites/archive.raspbian.org/archive/raspbian stretch InRelease [15.0 kB]        Hit:2 https://archive.raspberrypi.org/debian stretch InRelease
Fetched 15.0 kB in 1s (8,398 B/s)                     Reading package lists...   
```
- Raspbian repo is there but the RPi specific firmware and such repo not.

#### And most importantly APT + DPKG fails due to an perl/debconf error, which is the reason for all other package installs/configurations as well:
```
Calculating upgrade...
Unquoted string "remote" may clash with future reserved word at /usr/share/perl/5.24/Symbol.pm line 1.
debconf: Perl may be unconfigured (Bad name after n' at /usr/share/perl/5.24/Symbol.pm line 1.
Compilation failed in require at /usr/lib/arm-linux-gnueabihf/perl/5.24/IO/File.pm line 9.
BEGIN failed--compilation aborted at /usr/lib/arm-linux-gnueabihf/perl/5.24/IO/File.pm line 9.
Compilation failed in require at /usr/share/perl/5.24/FileHandle.pm line 9.
Compilation failed in require at (eval 1) line 3.
BEGIN failed--compilation aborted at (eval 1) line 3.
) -- aborting
```

#### Try the following to resolve:
- Disable debconf package pre-configuration to work around the perl error:
```
mv  /etc/apt/apt.conf.d/70debconf /etc/apt/70debconf_bak
```
- Re-add the RPi repo:
```
echo 'deb https://archive.raspberrypi.org/debian/ stretch main ui' > /etc/apt/sources.list.d/raspi.list
```
- Update the package lists:
```
apt update
```
- Reinstall `perl-base`:
```
apt -y install --reinstall perl-base
```
- Re-enable debconf pre-configuration:
```
mv /etc/apt/70debconf_bak /etc/apt/apt.conf.d/70debconf
```
- Reinstall the failed packages:
```
apt -y install --reinstall debconf coreutils rpcbind nfs-common rsync
```

MichaIng, 2019-04-18:
@jmtramos 
Could you resolve this issue meanwhile?

MichaIng, 2019-05-01:
I mark this issue as closed due to missing user reply. Feel free to reopen if required.

closed 2019-05-01 by MichaIng