MichaIng/DietPi #2670: Include "Please use spacebar to select option" note in dietpi-banner

Millichrome, 2019-03-27:
### Creating a bug report/issue

After initially running the dietpi-banner command, I couldn't figure out how to select any of the options. Then I remembered having the same issue when initially running dietpi-software, so I ran it and noticed the note about using spacebar. 

Not a big deal at all. I am a casual linux user, so maybe it's just me not being accustomed to linux/cli conventions.        

#### Required Information
- DietPi version | 6.22.3
- Distro version | stretch
- Kernel version | 4.4.71+
- SBC device | ASUS Tinker Board

MichaIng, 2019-03-27:
@Millichrome 
Many thanks for your request.

Yeah indeed I remember when using Whiptail the first time I also struggled. Also how to go to <Ok> <Cancel> buttons via <TAB>/<left>/<right>.

Perhaps our programs main menus should contain the basic Whiptail navigation hints, or we add some Whiptail intro when logging in the first time?

However adding the info from dietpi-software is easy for now.

Millichrome, 2019-03-27:
A tiny text prompt would be perfect. I am sure I am not the only user with this issue. :) 

I agree that it would be too much to add it to every menu. IMO it shouldn't really clutter the "dietpi-banner" (at least as it is right now). Just my take though.

MichaIng, 2019-03-27:
@Millichrome 
The menu now shows:
`Please (de)select options via spacebar to be shown in the DietPi-Banner:`

Millichrome, 2019-03-28:
Awesome! Hope this helps other people too. :)

closed 2019-04-18 by MichaIng