MichaIng/DietPi #2671: Wireguard service fails to start after update from 6.21 to 6.22

swrobel, 2019-03-27:
#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=22
G_DIETPI_VERSION_RC=3
G_GITBRANCH=master
G_GITOWNER=Fourdee
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
stretch
- Kernel version | `uname -a`
Linux DietPi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux
- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
RPi 3 Model B+ (armv7l)
- Power supply used | (EG: 5V 1A RAVpower)
5V 2.1A iPad charger
- SDcard used | (EG: SanDisk ultra)
SanDisk

#### Additional Information (if applicable)
- Software title | Wireguard
- Was the software title installed freshly or updated/migrated? Updated
- Can this issue be replicated on a fresh installation of DietPi? Most likely not
- `dietpi-bugreport` ID | bfa7faf1-0e3d-404b-b1dc-fa8da172aa5b

#### Steps to reproduce
```
root@DietPi:~# systemctl start wg-quick@wg0
Job for wg-quick@wg0.service failed because the control process exited with error code.
See "systemctl status wg-quick@wg0.service" and "journalctl -xe" for details.
root@DietPi:~# systemctl status wg-quick@wg0
● wg-quick@wg0.service - WireGuard via wg-quick(8) for wg0
   Loaded: loaded (/lib/systemd/system/wg-quick@.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2019-03-27 22:01:05 GMT; 5s ago
     Docs: man:wg-quick(8)
           man:wg(8)
           https://www.wireguard.com/
           https://www.wireguard.com/quickstart/
           https://git.zx2c4.com/WireGuard/about/src/tools/man/wg-quick.8
           https://git.zx2c4.com/WireGuard/about/src/tools/man/wg.8
  Process: 2191 ExecStart=/usr/bin/wg-quick up wg0 (code=exited, status=1/FAILURE)
 Main PID: 2191 (code=exited, status=1/FAILURE)

Mar 27 22:01:05 DietPi systemd[1]: Starting WireGuard via wg-quick(8) for wg0...
Mar 27 22:01:05 DietPi wg-quick[2191]: [#] ip link add wg0 type wireguard
Mar 27 22:01:05 DietPi wg-quick[2191]: RTNETLINK answers: Operation not supported
Mar 27 22:01:05 DietPi wg-quick[2191]: Unable to access interface: Protocol not supported
Mar 27 22:01:05 DietPi wg-quick[2191]: [#] ip link delete dev wg0
Mar 27 22:01:05 DietPi wg-quick[2191]: Cannot find device "wg0"
Mar 27 22:01:05 DietPi systemd[1]: wg-quick@wg0.service: Main process exited, code=exited, status=1/FAILURE
Mar 27 22:01:05 DietPi systemd[1]: Failed to start WireGuard via wg-quick(8) for wg0.
Mar 27 22:01:05 DietPi systemd[1]: wg-quick@wg0.service: Unit entered failed state.
Mar 27 22:01:05 DietPi systemd[1]: wg-quick@wg0.service: Failed with result 'exit-code'.
```

#### Expected behaviour
Wireguard should start as it did before

#### Actual behaviour
Wireguard will not start, even after a reboot

#### Extra details
It does not seem the wireguard kmod is loaded:

```
root@DietPi:~# lsmod
Module                  Size  Used by
sg                     28672  0
uas                    24576  0
uio_pdrv_genirq        16384  0
uio                    20480  1 uio_pdrv_genirq
fixed                  16384  0
ip_tables              24576  0
x_tables               32768  1 ip_tables
ipv6                  425984  27
```
However I believe all necessary packages are installed:

```
root@DietPi:~# dpkg -l | grep wireguard
ii  wireguard                     0.0.20181218-1               all          fast, modern, secure kernel VPN tunnel (metapackage)
ii  wireguard-dkms                0.0.20181218-1               all          fast, modern, secure kernel VPN tunnel (DKMS version)
ii  wireguard-tools               0.0.20181218-1               armhf        fast, modern, secure kernel VPN tunnel (userland utilities)
```

MichaIng, 2019-03-27:
@swrobel 
Many thanks for your report.

Most likely the kernel was upgraded but not the WireGuard kernel module. I also see that it's version is not the up-to-date: https://packages.debian.org/sid/wireguard

Looks like it was not auto-upgraded.

Could you try:
`apt upgrade`
Check if the WireGuard packages are upgraded as well. If not run:
`apt full-upgrade`
Check if the WireGuard packages are upgraded as well. If not run:
`apt install wireguard wireguard-dkms wireguard-tools`
The last should force installing the packages regardless of repo priorities.

Please report back which step did the WireGuard upgrade. It will include a rebuild of the module against the current kernel version.

swrobel, 2019-03-28:
Thanks, that worked!

MichaIng, 2019-04-14:
Fix is handled here: https://github.com/MichaIng/DietPi/pull/2700

closed 2019-05-12 by MichaIng