MichaIng/DietPi #2673: hostapd service fails to start

DigitalLifestyleCreations, 2019-03-29:
### Creating a bug report/issue

#### Details:
- Date           | Fri 29 Mar 07:40:05 GMT 2019
- Bug report     | N/A
- DietPi version | v6.22.3 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B+ (armv7l) (index=3)
- Kernel version | #1200 SMP Tue Feb 12 20:27:48 GMT 2019
- Distro         | stretch (index=4)
- Command        | systemctl start hostapd
- Exit code      | 1
- Software title | DietPi-Config

#### Steps to reproduce:

1. systemctl start hostapd
2. OR go to dietpi-config, Network Options: Adapters, Change Wireless Network setttings, go to State - [Disabled] | Select to turn on hotspot -- highlight and select. Error will occur.

#### Expected behaviour:

- hostapd should start and hotspot should be turned on

#### Actual behaviour:

- hostapd fails to start, hotspot remains disabled

#### Extra details:

Output of systemctl status hostapd.service:
● hostapd.service - Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator
   Loaded: loaded (/lib/systemd/system/hostapd.service; disabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Fri 2019-03-29 07:52:13 GMT; 1s 
ago
  Process: 2663 ExecStart=/usr/sbin/hostapd -P /run/hostapd.pid -B $DAEMON_OPTS ${DAEMON_CONF} 
(code=exited, status=1/FAILURE)
      CPU: 14ms

Mar 29 07:52:13 MicroSrv-Pi3 systemd[1]: Failed to start Advanced IEEE 802.11 AP and IE
EE 802.1X/WPA/WPA2/EAP Authenticator.
Mar 29 07:52:13 MicroSrv-Pi3 systemd[1]: hostapd.service: Unit entered failed state.
Mar 29 07:52:13 MicroSrv-Pi3 systemd[1]: hostapd.service: Failed with result 'exit-code
'.

Output of journalctl -xe
Mar 29 07:50:46 MicroSrv-Pi3 systemd[1]: Failed to start Advanced IEEE 802.11 AP and IE
EE 802.1X/WPA/WPA2/EAP Authenticator.
-- Subject: Unit hostapd.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit hostapd.service has failed.
-- 
-- The result is failed.
Mar 29 07:50:46 MicroSrv-Pi3 systemd[1]: hostapd.service: Unit entered failed state.
Mar 29 07:50:46 MicroSrv-Pi3 systemd[1]: hostapd.service: Failed with result 'exit-code
'.
Mar 29 07:50:48 MicroSrv-Pi3 systemd[1]: hostapd.service: Service hold-off time over, schedulin
g restart.
Mar 29 07:50:48 MicroSrv-Pi3 systemd[1]: Stopped Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WP
A2/EAP Authenticator.
-- Subject: Unit hostapd.service has finished shutting down
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit hostapd.service has finished shutting down.
Mar 29 07:50:48 MicroSrv-Pi3 systemd[1]: Starting Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/W
PA2/EAP Authenticator...
-- Subject: Unit hostapd.service has begun start-up
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit hostapd.service has begun starting up.
Mar 29 07:50:48 MicroSrv-Pi3 hostapd[2307]: Configuration file:
Mar 29 07:50:48 MicroSrv-Pi3 hostapd[2307]: Could not open configuration file '' for reading.
Mar 29 07:50:48 MicroSrv-Pi3 hostapd[2307]: Failed to set up interface with
Mar 29 07:50:48 MicroSrv-Pi3 hostapd[2307]: Failed to initialize interface
Mar 29 07:50:48 MicroSrv-Pi3 systemd[1]: hostapd.service: Control process exited, code=
exited status=1

#### Additional logs:
```
Log file contents:
Job for hostapd.service failed because the control process exited with error code.
See "systemctl status hostapd.service" and "journalctl -xe" for details.
```

MichaIng, 2019-03-29:
@DigitalLifestyleCreations 
Many thanks for your report.

Will check. First quick view:
`Could not open configuration file '' for reading.`
- Looks like the config file is somehow missing. Will check code and run local tests.

schnuckz, 2019-04-12:
Hi,

same issue on my rpi3, found the following fix works for me
```
wget http://dietpi.com/downloads/binaries/all/libssl1.0.0_1.0.1t-1+deb8u7_armhf.deb -O package.deb
dpkg -i package.deb
dietpi-services restart

systemctl unmask hostapd
systemctl enable hostapd
systemctl start hostapd
```

MichaIng, 2019-04-17:
@schnuckz 
Many thanks for this. So my assumption:
- Our custom binary is build on Jessie with libssl1.0.0/1
- On Stretch libssl1.0.2 is available and installed, thus `hostapd` binary denies to start.
- So yeah we either need to install as you did our libssl1.0.0 binary on Stretch + above or revert to Debian shipped hostapd package (older version 2.4) on Stretch.
- Problem is the custom build required for RTL8188C* WiFi devices: https://github.com/MichaIng/DietPi/issues/2531

And I wonder since in many cases on Stretch WiFi Hotspot works well 🤔. I have to run tests...
How did you actually find this solution? Any hints from logs or such that point into the direction of libssl1.0.0 package requirement?

schnuckz, 2019-04-18:
@MichaIng 

after fresh install:

```
root@xxxxxx:~# service hostapd restart
Failed to restart hostapd.service: Unit hostapd.service is masked.
```

```
root@xxxxxx:~# systemctl unmask hostapd
Removed /etc/systemd/system/hostapd.service.
```

```
root@xxxxxx:~# service hostapd restart
Job for hostapd.service failed because the control process exited with error code.
See "systemctl status hostapd.service" and "journalctl -xe" for details.
```

```
root@xxxxxx:~# systemctl status hostapd.service
● hostapd.service - Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authent
icator
   Loaded: loaded (/lib/systemd/system/hostapd.service; disabled; vendor preset:
 enabled)
   Active: activating (auto-restart) (Result: exit-code) since Thu 2019-04-18 05
:19:15 CEST; 580ms ago
  Process: 1588 ExecStart=/usr/sbin/hostapd -P /run/hostapd.pid -B $DAEMON_OPTS 
${DAEMON_CONF} (code=exited, status=127)

Apr 18 05:19:14 xxxxxx systemd[1]: Failed to start Advanced IEEE 802
.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
Apr 18 05:19:15 xxxxxx systemd[1]: hostapd.service: Unit entered fai
led state.
Apr 18 05:19:15 xxxxxx systemd[1]: hostapd.service: Failed with resu
lt 'exit-code'.
```

```
root@xxxxxx:~# systemctl enable hostapd
Synchronizing state of hostapd.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable hostapd
insserv: warning: current start runlevel(s) (empty) of script `hostapd' overrides LSB defaults (2 3 4 5).
insserv: warning: current stop runlevel(s) (0 1 2 3 4 5 6) of script `hostapd' overrides LSB defaults (0 1 6).
```

```
root@xxxxxx:~# systemctl start hostapd
Job for hostapd.service failed because the control process exited with error code.
See "systemctl status hostapd.service" and "journalctl -xe" for details.
```

```
root@xxxxxx:~# systemctl status hostapd.service
● hostapd.service - Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authent
icator
   Loaded: loaded (/lib/systemd/system/hostapd.service; enabled; vendor preset: 
enabled)
   Active: activating (auto-restart) (Result: exit-code) since Thu 2019-04-18 05
:20:45 CEST; 1s ago
  Process: 1724 ExecStart=/usr/sbin/hostapd -P /run/hostapd.pid -B $DAEMON_OPTS 
${DAEMON_CONF} (code=exited, status=127)

Apr 18 05:20:44 xxxxxx systemd[1]: Failed to start Advanced IEEE 802
.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
Apr 18 05:20:45 xxxxxx systemd[1]: hostapd.service: Unit entered fai
led state.
Apr 18 05:20:45 xxxxxx systemd[1]: hostapd.service: Failed with resu
lt 'exit-code'.
```

```
root@xxxxxx:~# /usr/sbin/hostapd -P /run/hostapd.pid
/usr/sbin/hostapd: error while loading shared libraries: libssl.so.1.0.0: cannot open shared object file: No such file or directory
```

after I tried to start hostapd, I use "google" :) and found this
https://github.com/MichaIng/DietPi/issues/1299#issuecomment-351623366

MichaIng, 2019-04-18:
@schnuckz 
Ahh I see. Now I also found where to fix has gone:
- Mark for install: https://github.com/MichaIng/DietPi/blob/dev/dietpi/dietpi-software#L2556-L2566
- Install: https://github.com/MichaIng/DietPi/blob/dev/dietpi/dietpi-software#L6303-L6335

So the libssl1.0.0 package should be already installed. Will verify this is the case.

Also strange that the service is masked. Couldn't find any code where we mask this. We enable/disable/start/stop within DietPi-Config when toggle Hostspot there, but never mask it.

MichaIng, 2019-04-18:
Testing on notebook:
🈯️ libssl1.0.0 is installed, and is present after install ran. So no need to manually install it.
🈺 isc-dhcp-server is installed. The service start triggered by APT install fails with: `dhcpcheck syslog for diagnostics. ... failed!` I guess since `rsyslog` is not installed and I read that by default is logs to `/var/log/syslog` as well (besides `journald`). However we create a configuration with `log-facility local7;` which means `journald` (only), and then finally after install finished the service starts fine.
🈺 I recognized some info about deprecated `INTERFACES` variable which should be now `INTERFACESv4` for IPv4 and is migrated automatically. Not urgent but I fixed this: https://github.com/MichaIng/DietPi/commit/4cf97648c2e793a8f62cccc8878724ae69eef0ee
🈯️ Aside from that, Hotspot works perfectly here. libssl1.0.0 installed and no masked service.

When doing a fresh install, could you paste the whole `dietpi-software` output and check:
```
dpkg -l libssl1.0.0
journalctl -u hostapd
```

schnuckz, 2019-04-19:
dietpi-software output:
```
─────────────────────────────────────────────────────
 DietPi v6.22.3 : 09:03 - Fri 19/04/19
 ─────────────────────────────────────────────────────
 - Device model : RPi 3 Model B (armv7l)
 - CPU temp : 67'C : 152'F (Running hot, not recommended)
 - LAN IP : 192.168.131.6 (eth0)
 ─────────────────────────────────────────────────────
[  OK  ] DietPi-Software | Free space check: path=/ | available=26050 MiB | required=500 MiB
[  OK  ] DietPi-Software | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Software | Connection test: http://raspbian.raspberrypi.org/raspbian
[  OK  ] NTPD: time sync | Completed
[ SUB1 ] DietPi-Services > unmask
[  OK  ] DietPi-Services | unmask all: proftpd
[  OK  ] DietPi-Services | unmask all: cron
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : proftpd

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Update & upgrade APT

[ INFO ] DietPi-Software | APT update, please wait...
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:2 https://archive.raspberrypi.org/debian stretch InRelease
Reading package lists...
[  OK  ] DietPi-Software | G_AGUP
[ INFO ] DietPi-Software | Running apt simulation to check for errors, please wait...
[ INFO ] DietPi-Software | APT installation for: bash-doc -s, please wait...
Inst bash-doc (4.4-5 Raspbian:stable [all])
Conf bash-doc (4.4-5 Raspbian:stable [all])
[  OK  ] DietPi-Software | G_AGI bash-doc -s
[ INFO ] DietPi-Software | APT upgrade, please wait...
[  OK  ] DietPi-Software | G_AGUG

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Checking for prerequisite software

[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : proftpd
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : proftpd

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installing WiFi Hotspot: turn your device into a wifi hotspot

[  OK  ] DietPi-Software | Connection test: https://dietpi.com/downloads/binaries/all/hostapd_2.5_all.zip
[ INFO ] DietPi-Software | G_THREAD_START_0 | wget https://dietpi.com/downloads/binaries/all/hostapd_2.5_all.zip -O 60.zip
[ INFO ] DietPi-Software | APT installation for: hostapd isc-dhcp-server iptables libnl-3-200, please wait...
Preconfiguring packages ...
Selecting previously unselected package libnfnetlink0:armhf.
(Reading database ... 37611 files and directories currently installed.)
Preparing to unpack .../00-libnfnetlink0_1.0.1-3_armhf.deb ...
Unpacking libnfnetlink0:armhf (1.0.1-3) ...
Selecting previously unselected package libip6tc0:armhf.
Preparing to unpack .../01-libip6tc0_1.6.0+snapshot20161117-6_armhf.deb ...
Unpacking libip6tc0:armhf (1.6.0+snapshot20161117-6) ...
Selecting previously unselected package libiptc0:armhf.
Preparing to unpack .../02-libiptc0_1.6.0+snapshot20161117-6_armhf.deb ...
Unpacking libiptc0:armhf (1.6.0+snapshot20161117-6) ...
Selecting previously unselected package libxtables12:armhf.
Preparing to unpack .../03-libxtables12_1.6.0+snapshot20161117-6_armhf.deb ...
Unpacking libxtables12:armhf (1.6.0+snapshot20161117-6) ...
Selecting previously unselected package libnetfilter-conntrack3:armhf.
[... 115 lines cut ...]
[  OK  ] DietPi-Software | Added setting ieee80211n=1 to end of file /etc/hostapd/hostapd.conf

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Configuring Tor Hotspot: optional: route hotspot traffic through tor

[ SUB1 ] DietPi-Services > dietpi_controlled
[  OK  ] DietPi-Services | dietpi_controlled : isc-dhcp-server
[  OK  ] DietPi-Services | dietpi_controlled : proftpd
[  OK  ] DietPi-Services | dietpi_controlled : hostapd
[  OK  ] DietPi-Services | dietpi_controlled : cron
[ .... ] DietPi-Survey | (1/3) Testing connection to ssh.dietpi.com, please wait[  OK  ] DietPi-Survey | Connection test: ssh.dietpi.com

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installation completed
```

```
root@xxx-rpi02:~# dpkg -l libssl1.0.0
dpkg-query: no packages found matching libssl1.0.0
root@xxx-rpi02:~# journalctl -u hostapd
-- No entries --
root@xxx-rpi02:~# 
```

MichaIng, 2019-04-19:
@schnuckz 
Ah found the reason why libssl1.0.0 is not installed in your case. When installing Tor Hotspot we mark WiFi Hotspot base install while install looks are already running, AFTER pre-requirements are marked. Will switch that.

It works after installing libssl1.0.0 manually, right?

schnuckz, 2019-04-21:
@MichaIng 
Yes, after manual install it works.

```
root@xxxx-rpi02:~# wget http://dietpi.com/downloads/binaries/all/libssl1.0.0_1.0.1t-1+deb8u7_armhf.deb -O package.deb
--2019-04-21 13:24:07--  http://dietpi.com/downloads/binaries/all/libssl1.0.0_1.0.1t-1+deb8u7_armhf.deb
Resolving dietpi.com (dietpi.com)... 104.27.179.199, 104.27.178.199, 2606:4700:30::681b:b2c7, ...
Connecting to dietpi.com (dietpi.com)|104.27.179.199|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://dietpi.com/downloads/binaries/all/libssl1.0.0_1.0.1t-1+deb8u7_armhf.deb [following]
--2019-04-21 13:24:07--  https://dietpi.com/downloads/binaries/all/libssl1.0.0_1.0.1t-1+deb8u7_armhf.deb
Connecting to dietpi.com (dietpi.com)|104.27.179.199|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 867176 (847K) [application/x-debian-package]
Saving to: ‘package.deb’

package.deb         100%[===================>] 846.85K  4.27MB/s    in 0.2s    

2019-04-21 13:24:08 (4.27 MB/s) - ‘package.deb’ saved [867176/867176]

root@xxxx-rpi02:~# dpkg -i package.deb
Selecting previously unselected package libssl1.0.0:armhf.
(Reading database ... 37903 files and directories currently installed.)
Preparing to unpack package.deb ...
Unpacking libssl1.0.0:armhf (1.0.1t-1+deb8u7) ...
Setting up libssl1.0.0:armhf (1.0.1t-1+deb8u7) ...
root@xxxx-rpi02:~# dietpi-services restart
[  OK  ] DietPi-Services | Root access verified.

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: restart

[  OK  ] DietPi-Services | restart : isc-dhcp-server
[  OK  ] DietPi-Services | restart : proftpd
[FAILED] DietPi-Services | restart : hostapd
[  OK  ] DietPi-Services | restart : cron
[ SUB1 ] DietPi-Process_tool > Apply
[  OK  ] DietPi-Process_tool | Cron (5803) : Nice      0
[  OK  ] DietPi-Process_tool | Cron (5803) : Affinity  0-3
[  OK  ] DietPi-Process_tool | Cron (5803) : Scheduler SCHED_OTHER 0
[  OK  ] DietPi-Process_tool | DHCP Server (5739) : Nice      0
[  OK  ] DietPi-Process_tool | DHCP Server (5739) : Affinity  0-3
[  OK  ] DietPi-Process_tool | DHCP Server (5739) : Scheduler SCHED_OTHER 0
[  OK  ] DietPi-Process_tool | Dropbear (765) : Nice      0
[  OK  ] DietPi-Process_tool | Dropbear (765) : Affinity  0-3
[  OK  ] DietPi-Process_tool | Dropbear (765) : Scheduler SCHED_OTHER 0
[  OK  ] DietPi-Process_tool | Dropbear (5472) : Nice      0
[  OK  ] DietPi-Process_tool | Dropbear (5472) : Affinity  0-3
[  OK  ] DietPi-Process_tool | Dropbear (5472) : Scheduler SCHED_OTHER 0
[  OK  ] DietPi-Process_tool | ProFTP (5784) : Nice      0
[  OK  ] DietPi-Process_tool | ProFTP (5784) : Affinity  0-3
[  OK  ] DietPi-Process_tool | ProFTP (5784) : Scheduler SCHED_OTHER 0
[  OK  ] DietPi-Process_tool | Completed
root@xxxx-rpi02:~# systemctl start hostapd
Failed to start hostapd.service: Unit hostapd.service is masked.
root@xxxx-rpi02:~# systemctl unmask hostapd
Removed /etc/systemd/system/hostapd.service.
root@xxxx-rpi02:~# systemctl enable hostapd
Synchronizing state of hostapd.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable hostapd
insserv: warning: current start runlevel(s) (empty) of script `hostapd' overrides LSB defaults (2 3 4 5).
insserv: warning: current stop runlevel(s) (0 1 2 3 4 5 6) of script `hostapd' overrides LSB defaults (0 1 6).
root@xxxx-rpi02:~# systemctl start hostapd
root@xxxx-rpi02:~# 
```

MichaIng, 2019-04-21:
@schnuckz 
Okay perfect, then this issue is fixed.

closed 2019-05-12 by MichaIng