MichaIng/DietPi #2674: Setup doesn't continue after first reboot when using PREP_SYSTEM_FOR_DIETPI.sh

banbenny, 2019-03-29:
Happened on a few boards including Orange Pi PC/BananaPi M2+ and NanoPi NEO2 boards. I install a self compiled Armbian image with all the right kernel and kernel modules and such, previously after the the script is done it would continue with setup after a reboot but lately it reboots and doesn't continue it is also missing any of the diet-scripts even no /DietPi/dietpi/.version exists. It was working fine for months until I think last week. Tried to sort it out myself but I can't find anything wrong.

Easy to repeat, use the prep script and it happens every single time.

Here's a console log of the full install process: https://pastebin.com/raw/JdAAUZ6m

Expected behaviour: Setup to continue after first reboot

MichaIng, 2019-03-29:
@banbenny 
Indeed strange, your log is missing preparation steps `0` to `3`, directly after the initial APT run + locale setting it starts with step 4, thus skips the important DietPi-RAMdisk activation that is then missing on reboot.

Which version of the script and how did you download?

banbenny, 2019-03-29:
I've downloaded: https://raw.githubusercontent.com/Fourdee/DietPi/master/PREP_SYSTEM_FOR_DIETPI.sh

Couldn't copy all the steps properly cuz I'm using the serial interface to connect to the boards. If there's any logs I can collect I'd gladly do so. But for now I'm happy running the update script fixes it 😌

Been a very happy user of DietPi for a long time. I like it a lot, this is the first time I ever ran into anything I couldn't sort out by searching the forum or internet.

MichaIng, 2019-03-29:
@banbenny 
You were able to run `dietpi-update`?
So now it runs as expected?

If you try again, please try with:
Please retry with:
```
bash -c "$(curl -sSL https://raw.githubusercontent.com/MichaIng/DietPi/dev/PREP_SYSTEM_FOR_DIETPI.sh)"
```

After it has finished, check: `systemctl status dietpi-ramdisk`
- This should show "enabled", if not please report back and enable manually with: `systemctl enable dietpi-ramdisk`
______
I see the issue about serial console. Yeah next ...

.... ah see your edit, so the RAMdisk was actually up. Ahh actually I see the issue now. Indeed there was some change not applied to the master version of the PREP script with v6.22. I already added it to v6.23 but was not aware that this results in the missing first run login running scripts. Indeed it makes sense that the update fixes it as it contains applying the change as wel. I will merge the fix/change into PREP script master. It's okay to do this outside of regular release since it does not affect running DietPi instances/code.

MichaIng, 2019-03-29:
**Done: https://github.com/MichaIng/DietPi/pull/2675**

Many thanks again for reporting. I mark this as closed.

closed 2019-03-29 by MichaIng