MichaIng/DietPi #2676: Error loading sudoers policy

shlatchz, 2019-03-29:
### Creating a bug report/issue

#### Required Information
- DietPi version:

> #!/bin/bash
> G_DIETPI_VERSION_CORE=6
> G_DIETPI_VERSION_SUB=22
> G_DIETPI_VERSION_RC=3
> G_GITBRANCH=master
> G_GITOWNER=Fourdee

- Distro version:

> stretch
> 9.8

- Kernel version:

> Linux DietPi 4.9.0-8-amd64 #1 SMP Debian 4.9.144-3.1 (2019-02-19) x86_64 GNU/Linux

- SBC device:

> Native PC (x86_64)

#### Additional Information (if applicable)
Bug Report ID:  3b0cd5c5-8000-40c6-ac83-14972cc6877b

#### Steps to reproduce
I installed the latest version of dietpi

#### Actual behaviour

After restart, I receive this error:

> sh: 1: /usr/bin/xauth: not found
> sudo: error in /etc/sudo.conf, line 0 while loading plugin "sudoers_policy"
> sudo: /usr/lib/sudo/sudoers.so must be only be writable by owner
> sudo: fatal error, unable to load plugins
> sudo: error in /etc/sudo.conf, line 0 while loading plugin "sudoers_policy"0)
> sudo: /usr/lib/sudo/sudoers.so must be only be writable by owner
> sudo: fatal error, unable to load plugins
> sudo: error in /etc/sudo.conf, line 0 while loading plugin "sudoers_policy"1)
> sudo: /usr/lib/sudo/sudoers.so must be only be writable by owner
> sudo: fatal error, unable to load plugins
> sudo: error in /etc/sudo.conf, line 0 while loading plugin "sudoers_policy"2)
> sudo: /usr/lib/sudo/sudoers.so must be only be writable by owner
> sudo: fatal error, unable to load plugins
> sudo: error in /etc/sudo.conf, line 0 while loading plugin "sudoers_policy"3)
> sudo: /usr/lib/sudo/sudoers.so must be only be writable by owner
> sudo: fatal error, unable to load plugins
> sudo: error in /etc/sudo.conf, line 0 while loading plugin "sudoers_policy"4)
> sudo: /usr/lib/sudo/sudoers.so must be only be writable by owner
> sudo: fatal error, unable to load plugins
> sudo: error in /etc/sudo.conf, line 0 while loading plugin "sudoers_policy"5)
> sudo: /usr/lib/sudo/sudoers.so must be only be writable by owner
> sudo: fatal error, unable to load plugins
> sudo: error in /etc/sudo.conf, line 0 while loading plugin "sudoers_policy"6)
> sudo: /usr/lib/sudo/sudoers.so must be only be writable by owner
> sudo: fatal error, unable to load plugins
> sudo: error in /etc/sudo.conf, line 0 while loading plugin "sudoers_policy"7)
> sudo: /usr/lib/sudo/sudoers.so must be only be writable by owner
> sudo: fatal error, unable to load plugins
> sudo: error in /etc/sudo.conf, line 0 while loading plugin "sudoers_policy"8)
> sudo: /usr/lib/sudo/sudoers.so must be only be writable by owner
> sudo: fatal error, unable to load plugins
> sudo: error in /etc/sudo.conf, line 0 while loading plugin "sudoers_policy"9)
> sudo: /usr/lib/sudo/sudoers.so must be only be writable by owner
> sudo: fatal error, unable to load plugins
> sudo: error in /etc/sudo.conf, line 0 while loading plugin "sudoers_policy"10)
[... 113 lines cut ...]
> sudo: error in /etc/sudo.conf, line 0 while loading plugin "sudoers_policy"48)
> sudo: /usr/lib/sudo/sudoers.so must be only be writable by owner
> sudo: fatal error, unable to load plugins
> sudo: error in /etc/sudo.conf, line 0 while loading plugin "sudoers_policy"49)
> sudo: /usr/lib/sudo/sudoers.so must be only be writable by owner
> sudo: fatal error, unable to load plugins
> sudo: error in /etc/sudo.conf, line 0 while loading plugin "sudoers_policy"50)
> sudo: /usr/lib/sudo/sudoers.so must be only be writable by owner
> sudo: fatal error, unable to load plugins
> sudo: error in /etc/sudo.conf, line 0 while loading plugin "sudoers_policy"51)
> sudo: /usr/lib/sudo/sudoers.so must be only be writable by owner
> sudo: fatal error, unable to load plugins
> sudo: error in /etc/sudo.conf, line 0 while loading plugin "sudoers_policy"52)
> sudo: /usr/lib/sudo/sudoers.so must be only be writable by owner
> sudo: fatal error, unable to load plugins
> sudo: error in /etc/sudo.conf, line 0 while loading plugin "sudoers_policy"53)
> sudo: /usr/lib/sudo/sudoers.so must be only be writable by owner
> sudo: fatal error, unable to load plugins
> sudo: error in /etc/sudo.conf, line 0 while loading plugin "sudoers_policy"54)
> sudo: /usr/lib/sudo/sudoers.so must be only be writable by owner
> sudo: fatal error, unable to load plugins
> sudo: error in /etc/sudo.conf, line 0 while loading plugin "sudoers_policy"55)
> sudo: /usr/lib/sudo/sudoers.so must be only be writable by owner
> sudo: fatal error, unable to load plugins
> [ INFO ] Waiting for DietPi-Postboot to finish... (Press CTRL+C to abort) (56)

MichaIng, 2019-03-29:
@shlatchz 
I guess you attempt to login as user with no sudo permissions?

Please do:
```
sed -i "/until sudo systemctl/c\		while pgrep -f '/DietPi/dietpi/postboot' &> /dev/null" /DietPi/dietpi/dietpi-login
```

Ref: https://github.com/MichaIng/DietPi/issues/2667

shlatchz, 2019-03-29:
@MichaIng No more error after reboot. thanks.

MichaIng, 2019-03-29:
@shlatchz 
Thanks for quick test. I mark this as closed as this fix has been implemented for v6.23.

closed 2019-03-29 by MichaIng

dimlev, 2019-04-26:
Now every time I login as a non root user I immediately get a prompt for my password. My first thought was that it was an upgrade issue because pressing CTRL+C shows the following:

`
Waiting for Postboot to finish... (Press CTRL+C to abort)
`

If there is no other way to check for the status of postboot, can I suggest adding a text explaining why I am being asked for the password.

MichaIng, 2019-04-26:
@dimlev 
Jep known issue and fixed with v6.23. See here how to fix now: https://github.com/MichaIng/DietPi/issues/2667