MichaIng/DietPi #2679: Pi-Hole: Blocking Page not shown

Gill-Bates, 2019-03-31:
Instead of showing the blocking page, Pi-Hole delivers:

[ERROR]: Unable to parse results from queryads.php: Unhandled error message ()

Looks like a known issue which be can solved by tweaking on some settings:

https://github.com/pi-hole/pi-hole/issues/2195

MichaIng, 2019-03-31:
@Gill-Bates 
Many thanks for your report.

Which webserver do you use?

Generally we should add webserver configs to our Pi-hole installer as of:
- Nginx: https://github.com/pi-hole/docs/blob/master/docs/guides/nginx-configuration.md
- Lighttpd: `setenv.add-environment = ("fqdn" => "true")` + https://github.com/pi-hole/pi-hole/blob/master/advanced/lighttpd.conf.debian

Gill-Bates, 2019-04-01:
I am using the built-in lighthttpd server

MichaIng, 2019-04-01:
@Gill-Bates
Could you try to add
```
setenv.add-environment = ("fqdn" => "true")
```

to your `/etc/lighttpd/lighttpd.conf`?

In case `"mod_setenv",` needs to be added to the upper part as well with contains the modules to lead. Note that there needs to be a comma after all the mod_ lines besides the last one.

I will run same that's later as well.

MichaIng, 2019-04-18:
Just tested here and works very well without any adjustments.

@Gill-Bates 
Just to verify:
- You did install Pi-hole via `dietpi-software`?
- You are on Raspberry Pi Stretch, right?
- Any other software installed that uses the webserver (and might have added/changed settings)?

Fourdee, 2019-05-13:
@MichaIng 

Unable to replicate any issues on ARMv8 with v6.23. I will mark this as closed for now due to pending additional info from @Gill-Bates and no response.

@Gill-Bates If you can provide required info (https://github.com/MichaIng/DietPi/issues/2679#issuecomment-484707368) and/or the issue is still occurring, please reopen the ticket.

MichaIng, 2019-05-13:
@Fourdee 
It is still a known issue on Pi-hole side: https://github.com/pi-hole/pi-hole/issues/2195
- As you can see it's about the usage of a variable `SERVER_ADDR` vs `HTTP_HOST` to allow/deny access to Pi-hole pages.
- But I still didn't understand in which situations (network setups?) the current code (`HTTP_HOST`) fails, so am unable to replicate/trigger the error.
- However not much we can do our side. I am not keen to change the value manually in our installer as long as it's not clear what possible side effects are. And every `pihole -up` would anyway overwrite/revert.

MichaIng, 2020-03-13:
PR up: https://github.com/pi-hole/pi-hole/pull/3067

mschoettle, 2020-04-13:
> PR up: [pi-hole/pi-hole#3067](https://github.com/pi-hole/pi-hole/pull/3067)

This PR was replaced with pi-hole/pi-hole#3263 so that the fix will be included in version 5 of Pi-Hole.

closed 2020-09-10 by MichaIng