MichaIng/DietPi #2681: Restore IP Address to Boot Banner

borpin, 2019-04-01:
Please restore the LAN IP Address to the Boot Banner by default please.  Really useful and extremely irritating it has gone 😄. Note this is the **Boot** banner, not the **Login** Banner.

Reason - first boot in VM or SBC, video connected, you used to be able to identify the allocated IP without any further effort.  Now you **have** to login via the console to get the IP or use a different tool.

MichaIng, 2019-04-01:
@borpin 
Now that you say it, you are absolutely right.
The local IP and perhaps hostname as well should be shown in the upper part of the banner as identifier which is shown before login. E.g. as well if you want to SSH directly while leaving the login prompt one some LCD alone.

We will change that for v6.23 and I post instructions later on how to achieve that on your system.

borpin, 2019-04-01:
Hi @MichaIng, I can live without it for now - just good to know it will reappear. Thanks for all your efforts.

Fourdee, 2019-04-05:
@borpin  :+1: Agree.

@MichaIng Quick fix for this, function IP print:
https://github.com/MichaIng/DietPi/commit/69a5762fc98f10d2ff2dce9cebc4847b56d2c4e1

Fourdee, 2019-04-05:
:u6307: Testing fine, completed.

closed 2019-04-05 by Fourdee

MichaIng, 2019-04-05:
@Fourdee 
Perfect, thanks :slightly_smiling_face:!