MichaIng/DietPi #2682: Neo2 squeezebox not starting / system stuck on dietpi config

avworldme, 2019-04-01:
### Creating a bug report/issue
I have installed the DietPi_v6.14_NanoPiNEO2-ARMv8-Stretch on my Neo2 LTS, however the system is stuck on Diet-Pi Configuration, I cannot go to command prompt.  When I choose exit, it just loops back to the configuration men.

I also cannot get squeezebox to run.  I don't have any troubleshooting logs as i am no able to get command prompt.

I am using the Samsung Evo+ SD card.
any guidance would be appreciated.

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?
- `dietpi-bugreport` ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

MichaIng, 2019-04-01:
@avworldme 
Thanks for your report.

Do not choose "Exit" but "Install" to finish first run setup. Perhaps we should change the wording to make this clear if it indeed was the mistake? Or as/if you did this already for installing Squeezebox, does the DietPi-Software prompt re-appear after reboot?

nicolasbuch, 2019-04-01:
Just encountered the same issue with DietPi v6.14 Odroid XU4. Screen gets stuck in the first dietpi config. Choosing exit or install just loops back to same config. 

I noticed that it was a few versions behind, so i fixed the issue by updating to the newest version before running the initial dietpi-config. I did the following:

When booting (just before the automated script throws you into the dietpi-config screen) hit CTRL + C to cancel. Run dietpi-update. Next time it boots finish the config and enjoy :)

avworldme, 2019-04-01:
Thank you nicolasbuch, that worked for me with a bit of modification.  I don't have a video output on the Neo2, so my system booted too quick to hit ctrl-C.  So the trick for me was to open a new terminal window which gave me the message:

[ INFO ] DietPi-Login | DietPi-Software is already running on another terminal/screen. Please complete or exit, before continuing.

that was enough time form me to break, then do the update.  then i removed lms and reinstalled it again and boom there it is.

Now I have a new issue, i'll open another ticket.

Perhaps it's time for a new image for the Neo2.

MichaIng, 2019-04-04:
@avworldme @nicolasbuch 
I believe it is resolved with: https://github.com/MichaIng/DietPi/pull/2684
This also fits to your solutions to manually run dietpi-update before doing first run setup.

avworldme, 2019-04-04:
Is it possible to update the DietPi Neo 2 image to the latest?  I believe the one that's available now is 6.14.  So it will always have this issue.

MichaIng, 2019-04-04:
@avworldme 
> So it will always have this issue.

Nope, the issue was within the online part of the updater only, not the offline part which is shipped with the image.

If the image is still on the device you can test via:
```
rm -R /tmp/DietPi*
> /DietPi/dietpi/.update_stage
echo 0 > /DietPi/dietpi/install_stage
/DietPi/dietpi/dietpi-update 1
```

avworldme, 2019-04-04:
Ok, I just tried a fresh install and I indeed didn't have any more loops.  Thank you.

MichaIng, 2019-04-04:
@avworldme 
Thanks for testing. The change to the update system with v6.20 really caused a lot of trouble. I hope now the last bit is resolved.

closed 2019-05-12 by MichaIng